BiancoRoyal/node-red-contrib-modbus #1: Feature request - endianness

thebladgeek, 2017-01-20:
Thanks for your Nodes, using them to read data from a RS485 power meter.
I have two issues; 
1. The address is off by one. This is a common challenge with ModbusRTU and not a major problem to work around (I just put the starting address for the frequency as the last byte of the voltage).
2. Endianness. The data is arranged as LSB,MSB. Having an option to swap the byte order would be a major advantage when reading many registers at once.

biancode, 2017-01-20:
Hi, can you explain the 1. a bit, please?

biancode, 2017-01-20:
2. you can do that with the buffer of the second output - read multiple and switch all buffers a 16Bit

thebladgeek, 2017-01-20:
1. Some manufacturers start counting their address registers from 4000, some from 40001 (as per the Modbus spec). Hence, you can often be 'offset of one' or "using the offset" (another common term to explain it).

2. Please elaborate on this hidden feature. The 'info' tab in Node-RED for your node simply states 'Output 2: modbus response'. There is no option to switch the buffers that I can find.

Thanks for the quick reply, we are trying to sort the output in a function block, if there is a better way, it could save a lot of time.

biancode, 2017-01-20:
1. OK
2. Oh you didn't watch the videos :D 

[Modbus Response](https://youtu.be/Bq4BNaGyQ3E?list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH)

see full [Playlist](https://www.youtube.com/playlist?list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH)

thebladgeek, 2017-01-20:
Yes, watched the videos, I need to use the data in my node, not your buffer
node, that is why I can not follow the video (as you say in the audio)
exactly as shown.

Thanks for answering my questions.

Great work on the nodes, appreciate your work.

biancode, 2017-01-20:
you got the raw data in your flow

[image]

[{"id":"e0f58d2d.cc965","type":"modbus-read","z":"849ecf81.3ac91","name":"Automatic Read FC3","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"0","quantity":"10","rate":"10","rateUnit":"s","server":"","x":290,"y":840,"wires":[[],["11029b37.b15935","4f5abae6.37cda4"]]},{"id":"11029b37.b15935","type":"debug","z":"849ecf81.3ac91","name":"","active":false,"console":"false","complete":"true","x":470,"y":840,"wires":[]},{"id":"4f5abae6.37cda4","type":"function","z":"849ecf81.3ac91","name":"extract buffer","func":"return { payload: msg.payload.buffer };","outputs":1,"noerr":0,"x":490,"y":880,"wires":[["3a55fdc5.810522"]]},{"id":"3a55fdc5.810522","type":"debug","z":"849ecf81.3ac91","name":"","active":false,"console":"false","complete":"false","x":670,"y":880,"wires":[]}]

biancode, 2017-01-20:
I'll check that to have a option in Feb.

thebladgeek, 2017-01-20:
Yes, we have the raw data, but the endianness still needs to be swapped.

Thanks for letting us know you will have that option in the future.
It will make using your node a lot more flexible.

biancode, 2017-01-21:
[{"id":"34f2ef3a.17b59","type":"function","z":"849ecf81.3ac91","name":"extract buffer","func":"return [{ payload: msg.payload.buffer.readInt16LE(0,2)},\n{ payload: msg.payload.buffer.readInt16BE(0,2) }];","outputs":"2","noerr":0,"x":410,"y":1700,"wires":[["5cd727ab.845578"],["5cd727ab.845578"]]}]

biancode, 2017-01-21:
https://github.com/biancode/node-red-contrib-modbus/wiki/Buffer-and-endianness---32Bit---64Bit

thebladgeek, 2017-01-24:
Thanks for the flow, still working on getting it to swap correctly.
One thing that is making it hard is the very noisy console.
This Node is the only one I have seen make so many log entries.....
Am I doing something wrong?

24 Jan 00:36:08 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"read msg","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4,"queueUnitId":4},"unitid":4,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:08 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"queue response activate sending","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4,"queueUnitId":4}} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:08 - [info] [modbus-client:ttyUSBzero] Client -> {"state":"QUEUEING","message":"dequeue command simpleser","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:08 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"queue check is not a unit","unitid":null} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"push to Queue by Unit-Id","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4},"unit":4} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"queue read msg","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4},"state":"EMPTY","queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"state":"QUEUEING","message":"dequeue command simpleser","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"queue check","unitid":4,"sendAllowed":true,"queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"serial sending and wait","unitid":4,"commandData":{"msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4}},"queueLength":0,"sendAllowedForNext":false,"delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"read msg","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4,"queueUnitId":4},"unitid":4,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"queue response activate sending","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4,"queueUnitId":4}} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"state":"QUEUEING","message":"dequeue command simpleser","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:10 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"queue check is not a unit","unitid":null} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"push to Queue by Unit-Id","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4},"unit":4} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"queue read msg","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4},"state":"EMPTY","queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"state":"QUEUEING","message":"dequeue command simpleser","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"queue check","unitid":4,"sendAllowed":true,"queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"serial sending and wait","unitid":4,"commandData":{"msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4}},"queueLength":0,"sendAllowedForNext":false,"delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"read msg","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4,"queueUnitId":4},"unitid":4,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"info":"queue response activate sending","msg":{"topic":"polling","from":"wattNode","payload":{"unitid":"4","fc":3,"address":"1032","quantity":"2"},"queueNumber":0,"queueUnit":4,"queueUnitId":4}} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"state":"QUEUEING","message":"dequeue command simpleser","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4
24 Jan 00:36:12 - [info] [modbus-client:ttyUSBzero] Client -> {"type":"queue check is not a unit","unitid":null} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 4

biancode, 2017-01-24:
Start without verbose mode -v option

thebaldgeek, 2017-02-07:
After testing this workaround for many days, we can not use the -v option.
It mutes (suppresses) too much important diagnostic information from other nodes.

Also, in one case, we are using Node-RED in an embedded device and do not have control of the Node-RED start command.

Please consider an option to make this node less chatty in the console log. Thanks.