BiancoRoyal/node-red-contrib-modbus #2: Modbus Read

mtoko, 2017-01-21:
Thank you for the useful software I am using this in interesting applications with many possibilities

The Modbus Read node configured in RTU mode works only at a poll rate of 5 seconds no other poll rates appear to work?

Your assistance is appreciated

Do you think there is a possibility of a similar node for Bacnet ?

biancode, 2017-01-22:
a) the poll rate on my tests is working from 0.01 sec to 10 sec - what is the problem? Please post your flow to check - do you work with the queue or without? May you need RTU buffered not RTU.

b) BACnet is possible

biancode, 2017-01-22:
[{"id":"d5ef2643.487918","type":"modbus-read","z":"8aa1c6a2.dcf3c8","name":"","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"0","quantity":"10","rate":"1","rateUnit":"s","server":"7c85d392.df18bc","x":220,"y":200,"wires":[["198520f9.269a7f"],["32e03cb2.1a1d74"]]},{"id":"fd48ceea.8c149","type":"modbus-server","z":"8aa1c6a2.dcf3c8","name":"","logEnabled":false,"serverPort":10502,"responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":240,"y":80,"wires":[[],[],[]]},{"id":"32e03cb2.1a1d74","type":"modbus-response","z":"8aa1c6a2.dcf3c8","name":"","registerShowMax":20,"x":440,"y":180,"wires":[]},{"id":"198520f9.269a7f","type":"debug","z":"8aa1c6a2.dcf3c8","name":"","active":false,"console":"false","complete":"false","x":420,"y":140,"wires":[]},{"id":"7c85d392.df18bc","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000}]

mtoko, 2017-01-22:
I have posted a project on Upwork about Bacnet if you are interested

https://www.upwork.com/jobs/~0173529dfffe536d3c

The flow is simple and rates work from 1 to 5 seconds but not above 5 second. 

[{"id":"1093b10c.13a84f","type":"modbus-read","z":"1c53ebdf.30ba44","name":"Analogue2","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"11","quantity":"1","rate":"5","rateUnit":"s","server":"1e48779.8b02e88","x":244,"y":72,"wires":[["82b29648.ba1698"],["75f7f267.06462c"]]},{"id":"75f7f267.06462c","type":"modbus-response","z":"1c53ebdf.30ba44","name":"","registerShowMax":20,"x":502,"y":127,"wires":[]},{"id":"82b29648.ba1698","type":"debug","z":"1c53ebdf.30ba44","name":"","active":true,"console":"false","complete":"false","x":510,"y":60,"wires":[]},{"id":"1e48779.8b02e88","type":"modbus-client","z":"","name":"","clienttype":"serial","bufferCommands":true,"tcpHost":"127.0.0.1","tcpPort":"502","serialPort":"/dev/ttyUSB0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"2","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"100","clientTimeout":1000,"reconnectTimeout":5000}]

Thank you

biancode, 2017-03-03:
please test with node-red-contrib-modbus@1.0.6

closed 2017-03-03 by biancode