BiancoRoyal/node-red-contrib-modbus #3: Feature/Hint request - Modbus request byte array

Killerloop2, 2017-01-21:
Hi, first and foremost thank you for these great collection of nodes.

Is there a way to get the actual bytes of the request being sent? I would like to get something like this in my debug tab: "11 06 00 01 00 03 9A 9B"

where:

11: The Slave Address (11 hex = address17 )
06: The Function Code 6 (Preset Single Register)
0001: The Data Address of the register.
0003: The value to write 
9A9B: The CRC (cyclic redundancy check) for error checking.

even if the target device does not answer.

biancode, 2017-01-22:
no - not for now - that is a feature request to [modbus-serial](https://github.com/yaacov/node-modbus-serial) to return it in response

biancode, 2017-03-23:
please try v1.0.8 with `DEBUG=node_red_contrib_modbus_client,node_red_contrib_modbus,modbus-serial node-red -v`

closed 2017-03-23 by biancode