BiancoRoyal/node-red-contrib-modbus #5: CRC Error

yonas124, 2017-01-30:
Hi biancode,

I got "CRC Error" when I try to read more than 1 register over Modbus serial RS485 (sparkfun shield). If I read just 1, it works flawlessly.

[image]

biancode, 2017-01-30:
Hi, please I need more information about your device or a manual link. CRC is Big Endian by default - some implementations on devices are wrong in that case. If it is to configure check that CRC configuration at your device, please.

yonas124, 2017-01-31:
The device is called PM800 by Schneider Electric. You can find the modbus register list [here](https://www.google.co.id/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwiAm62yn-vRAhUCTo8KHR5mBssQFggZMAA&url=http%3A%2F%2Fwww.schneider-electric.com%2Flibrary%2FSCHNEIDER_ELECTRIC%2FSE_LOCAL%2FAPS%2F206807_1E86%2FPublic__PM800_Register_List_v12.2xx.pdf&usg=AFQjCNGTPPgX7d2eQKzgKt9RMDiH7WmvoA&sig2=gdeynyIVQTycDYk1tk_Y3A&bvm=bv.145822982,d.c2I)

In the previous version 0.8.2, I can select the endian (big or little) but the option is not there anymore since I updated to version 1.0.1.

I cannot change device's CRC configuration.
"Manufacturing Data CRC" value is 48251. I don't know what this means.

biancode, 2017-02-01:
please, post your flow

yonas124, 2017-02-02:
[image]

[{"id":"32a26606.f81a2a","type":"modbus-read","z":"6bedd68d.a19a68","name":"R1","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"1699","quantity":"2","rate":"10","rateUnit":"s","server":"a833778d.035c68","x":152.00003051757812,"y":554.6000213623047,"wires":[["1b0d63dc.53368c"],[]]},{"id":"1b0d63dc.53368c","type":"debug","z":"6bedd68d.a19a68","name":"","active":false,"console":"false","complete":"payload","x":357.0000915527344,"y":548.7999877929688,"wires":[]},{"id":"a833778d.035c68","type":"modbus-client","z":"","name":"PM800","clienttype":"serial","bufferCommands":false,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","serialPort":"/dev/serial0","serialType":"RTU","serialBaudrate":"38400","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"200","unit_id":"1","commandDelay":"","clientTimeout":"10000","reconnectTimeout":"2000"}]

biancode, 2017-02-02:
please try:
[{"id":"8d9cebcf.0f4f48","type":"modbus-read","z":"c751892a.2202a8","name":"R1","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"1699","quantity":"2","rate":"1","rateUnit":"s","server":"5cfbae9.bc3595","x":350,"y":320,"wires":[["4202368e.b6b098"],["771964f5.0ef35c"]]},{"id":"4202368e.b6b098","type":"debug","z":"c751892a.2202a8","name":"","active":false,"console":"false","complete":"payload","x":550,"y":300,"wires":[]},{"id":"771964f5.0ef35c","type":"modbus-response","z":"c751892a.2202a8","name":"","registerShowMax":20,"x":570,"y":340,"wires":[]},{"id":"5cfbae9.bc3595","type":"modbus-client","z":"c751892a.2202a8","name":"PM800","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","serialPort":"/dev/serial0","serialType":"RTU-BUFFERD","serialBaudrate":"38400","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"200","unit_id":"1","commandDelay":"","clientTimeout":"3000","reconnectTimeout":"5000"}]

biancode, 2017-02-02:
please take a look at the [Wiki-Pages ](https://github.com/biancode/node-red-contrib-modbus/wiki/Queue-hints)

biancode, 2017-02-02:
@yonas124 try https://www.npmjs.com/package/node-red-contrib-isa - you can map modbus flat data to structured OPC UA - it is not only isa - see [examples](https://github.com/biancode/node-red-contrib-isa/tree/master/examples)

yonas124, 2017-02-02:
Still not connecting, with or without queue
[image]

I can connect normally using [this one](https://www.npmjs.com/package/node-red-contrib-serial-modbus)

biancode, 2017-02-02:
send your flow from the working package, please

biancode, 2017-02-02:
That package is exactly the same but uses alway RTU buffered

yonas124, 2017-02-02:
This is the flow from the working modbus serial

`[{"id":"11914137.f913cf","type":"function","z":"6bedd68d.a19a68","name":"DecodeBuffer1","func":"msg.payload = ((msg.payload.readUInt16BE(6)*1000000000000+msg.payload.readUInt16BE(4)*100000000+msg.payload.readUInt16BE(2)*10000+msg.payload.readUInt16BE(0))/1000);\nreturn msg;","outputs":1,"noerr":0,"x":327,"y":539,"wires":[["5d45c760.41b238"]]},{"id":"d63d1491.f2d0e8","type":"function","z":"6bedd68d.a19a68","name":"DecodeBuffer2","func":"msg.payload = ((msg.payload.readUInt16BE(14)*1000000000000+msg.payload.readUInt16BE(12)*100000000+msg.payload.readUInt16BE(10)*10000+msg.payload.readUInt16BE(8))/1000);\nreturn msg;","outputs":1,"noerr":0,"x":329,"y":592,"wires":[["13b8c221.bc339e"]]},{"id":"8135f409.775f78","type":"modbusSerial in","z":"6bedd68d.a19a68","port":"7d3b04a0.33890c","slaves":"1","start":"1699","count":"8","dtype":"holding","period":"2","topic":"","name":"","x":107,"y":540,"wires":[["11914137.f913cf","d63d1491.f2d0e8"]]},{"id":"7d3b04a0.33890c","type":"modbusSerialConfig","z":"","port":"/dev/serial0","baud":"38400","data":"8","parity":"none","stop":"1","name":"PM800"}]`

Also the isa shows warning:
`warning : the expected version of node-opcua  is 0.0.56 - actual version is 0.0.60
`
then it crashed:
```
2 Feb 09:48:28 - [red] Uncaught Exception:
2 Feb 09:48:28 - Error: Cannot find module 'underscore'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/home/pi/.node-red/node_modules/node-red-contrib-isa/isa                                                                                                          /helpers/isa95_demo_address_space.js:4:9)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at post_initialize (/home/pi/.node-red/node_modules/node-red-contrib-isa/isa/ISA95-OPCUA-Server.js:158:51)
    at Immediate._onImmediate (/home/pi/.node-red/node_modules/node-red-contrib-isa/node_modules/node-opcua/lib/server/opcua_server.js:442:9)
    at processImmediate [as _immediateCallback] (timers.js:383:17)
```

biancode, 2017-02-02:
ah OK, then I've the underscore package in global install - I'll check that next week, but for now I did not see a difference - very interesting - please try new v1.0.2 with static 4.5.1 of modbus-serial package

mu45, 2017-02-13:
Hello biancode, 
we have the same problem like yonas124. We use the v1.0.1 and the modbus-serial package v4.5.1. When we readout only 1 input-register with modbus RTU then it works perfectly. But when we change to readout 2 input-registers, then we get the crc check error. 
The device we try to readout is DL-100T485-W/DL-100TM485-W from ICPDAS. 
Do you have an idea how we can solve this?

biancode, 2017-02-13:
please try the v1.0.5 package with modbus-serial 4.5.1 - v4.6.1 may has a problem or I've to change some for that version

biancode, 2017-02-13:
@yonas124 how about your tests?

biancode, 2017-03-03:
please test with node-red-contrib-modbus@1.0.6

mu45, 2017-03-06:
Hello Biancode, 

thanks for your answer I will test it and give you feedback.

biancode, 2017-03-23:
Now we have v1.0.8 does it work for you?

closed 2017-03-23 by biancode

kalyanTDS, 2018-08-01:
[image]
iam getting crc error. how to resolve this???

biancode, 2018-08-01:
@kalyanTDS  normally you have wrong settings in you config node. Try to connect via modbus-serial with a [tiny script](https://github.com/yaacov/node-modbus-serial/tree/master/examples). Does it work? and what are the parameters in the script to succeed?