BiancoRoyal/node-red-contrib-modbus #6: Modbus Reader does not recover from temporary loss of Modbus TCP Slave

sirockin, 2017-02-01:
With the modbus reader set to poll at 1s, we successfully read several bytes over several intervals. On physically disconnecting the slave from the network, the reader node shows its state change from 'active' to  "error: "FSM not ready to Read" then to "error{}", then alternates several times before staying in "error{}". 

Meanwhile the debug trace shows "Error: Port Not Open"

On reconnecting the slave, the node stays in the "error{}" state the "Error: Port Not Open" traces continue.

Proper operation can only be restored by redeploying. 

Sample trace below

`
[
    {
        "id": "mb_reader_mb_client_192.168.1.11_mr_0",
        "type": "modbus-read",
        "z": "tab_Main",
        "name": "mb_client_192.168.1.11_mr_",
        "showStatusActivities": false,
        "unitid": "",
        "dataType": "HoldingRegister",
        "adr": "12388",
        "quantity": "2",
        "rate": 1,
        "rateUnit": "s",
        "server": "mb_client_192.168.1.11",
        "x": 165,
        "y": 55,
        "wires": [
            [],
            []
        ]
    },
    {
        "id": "mb_client_192.168.1.11",
        "type": "modbus-client",
        "z": "tab_Main",
        "name": "AMCS1",
        "clienttype": "tcp",
        "tcpHost": "192.168.1.11",
        "tcpPort": "502",
        "serialPort": "/dev/ttyS0",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "unit_id": "1",
        "clientTimeout": "5000",
        "reconnectTimeout": "5000"
    }
]
`

sirockin, 2017-02-01:
This was not a problem with version 0.9.4.

biancode, 2017-02-01:
see https://youtu.be/r9Y5CxrkkQc it's not reproducible on my device - plug off and plug in and it reconnects

biancode, 2017-02-01:
[{"id":"4ee07071.79eb8","type":"modbus-read","z":"fa226315.bac79","name":"mb_client_192.168.1.11_mr_","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"20","quantity":"2","rate":1,"rateUnit":"s","server":"8c49b514.1411f8","x":300,"y":80,"wires":[[],["777b77a2.62a528"]]},{"id":"777b77a2.62a528","type":"modbus-response","z":"fa226315.bac79","name":"","registerShowMax":20,"x":570,"y":60,"wires":[]},{"id":"8c49b514.1411f8","type":"modbus-client","z":"fa226315.bac79","name":"AMCS1","clienttype":"tcp","bufferCommands":false,"stateLogEnabled":false,"tcpHost":"192.168.1.250","tcpPort":"502","serialPort":"/dev/ttyS0","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"","unit_id":"1","commandDelay":"","clientTimeout":"1000","reconnectTimeout":"5000"}]

sirockin, 2017-02-02:
Thanks for the feedback. 

I tried using your flow and got the same result on my setup. I'm uploading a video now and will provide the link once it's there.

biancode, 2017-02-02:
And please activate state logs on config and send me the verbose logs of node-red -v

sirockin, 2017-02-02:
Not sure what you mean by "activate state logs" but I can give you a verbose dump of the console.

sirockin, 2017-02-02:
Okay so here's a verbose console dump. At 11:09:02 I disconnect the cable then at 11:09:21 I reconnect. I then redeploy at 11:09:40, at which point the reader starts working again.
[node-red-log.txt](https://github.com/biancode/node-red-contrib-modbus/files/747404/node-red-log.txt)

sirockin, 2017-02-02:
And here's the video:
https://www.youtube.com/watch?v=IBsx8B1PxXw

biancode, 2017-02-04:
please uninstall and reinstall it and check that you got 4.5.1 for modbus-serial in the tree after installing

biancode, 2017-02-04:
and set the reconnecting time to 500ms, please - your windows looks very slow, is it from video?

biancode, 2017-03-03:
please test with node-red-contrib-modbus@1.0.6

biancode, 2017-03-05:
please test with node-red-contrib-modbus@1.0.7

saschaklick, 2017-03-15:
He's on Linux - `"serialPort": "/dev/ttyS0"` is a dead giveway - and that's the problem. More specifically it dwells right here:

https://github.com/biancode/node-red-contrib-modbus/blob/master/src/modbus-client.js#L461

You are only putting your state machine into "failure" when the `errno` is in the networkErrors-array. `errno` is only set on Windows and on Linux you only get err.message, which is always `'Timed out'` I think. Did not bother to investigate further.

I fixed it by setting `errno` if not set at https://github.com/biancode/node-red-contrib-modbus/blob/master/src/modbus-client.js#L459

`if(!err.errno) err.errno = 'ETIMEDOUT';`

Works for me on Node 6.10.0, Red 0.16.2, node-red-contrib-modbus v1.0.8 and I am one happy camper.

sirockin, 2017-03-15:
Hi @biancode , @saschaklick 

Sorry not to reply sooner - we've been busy with some other dev. We'll test on new version when we can.

@saschaklick your sleuthing has let you down - we're running on Windows 10 - the serialPort stuff is a default value set in by the node-red node.

saschaklick, 2017-03-15:
Aw shucks. Well I found my problem by figuring out why the state machine never went into the ONFAILED-callback, which if called should lead to a clean reconnection attempt.

If that one is called in your case, the problem is definitely a different one.

joshuatreenl, 2017-03-22:
More or less I have a similar issue. If I disconnect the network connection (I am using Modbus TCP) then there is a reconnection with no errors. I see a timeout on the Modbus read function. BUT if the network connection is correct AND I disabled the software on the server side, the initialisation is done, but after it is initialised it will stay in that mode even if I restart the application on the server side. There will be no connection/reading anymore. When I restart my node red app it works ok. 
I am using an Siemens S7 PLC with Modbus TCP (act as server) and an IOT2040 on the client side. Before someone react with why don't you use the S7 node red tool, I did. But I want to make a more general solution. Thanks in advance for helping me. 

Note: I don't know if this is the right place to place this question...

- Rene -

biancode, 2017-03-22:
@joshuatreenl - please start with DEBUG=node_red_contrib_client,modbus-serial node-red -v and send/post your log from that test case - or use wireshark to see which network error will be raised.

joshuatreenl, 2017-03-22:
Hi biancode, I will do that and I will upload the information.

biancode, 2017-03-23:
[Network Errno Codes](http://www.virtsync.com/c-error-codes-include-errno) - one of that list should be yours

biancode, 2017-03-23:
please try node-red-contrib-modbus@1.0.9-gamma

joshuatreenl, 2017-03-23:
Hi, some comments about the suggestions. 
1) Installing the DEBUG line: I am using the Yocto Linux version Siemens delivered. You can execute the setup command iot2000setup. There is one configuration option : startup node red at boot. The startup line is a little bit different than you mentioned DEBUG=... My startup looks like this: 
/usr/bin/node /usr/lib/node_modules/node-red/red >/dev/null &
If I analyse the 'red' part there is a red.js file which startup other programs (I think, my knowledge of Linux should be improved a little bit...). The option you mentioned: DEBUG= is different and would not work. I think somewhere before startup the 'packages which include the Modbus package' the DEBUG line should be placed... but where?
2) Version 1.0.9. I have tried to install this version, but it seems that version 1.08 is still running or didn't you update the version inside the package. In the package.json the  "version": "1.0.8" is mentioned. The file itself is from 19-03-2017 11:52:06. If that's the right one? then  the result is negative. The connection stays down...  

Thanks for all your help!
Kind regards,
René

biancode, 2017-03-23:
1. `DEBUG=node_red_contrib_modbus_client,modbus-serial node red.js` works for me - DEBUG is a variable to set see [Node.js Debug usage](https://strongloop.com/strongblog/the-node-js-debug-module-advanced-usage/)

2. You have to use the version - it is not released for now `npm install node-red-contrib-modbus@1.0.9-gamma`

joshuatreenl, 2017-03-24:
Hi Klaus, 
1) I've installed version 1.0.9 
2) I started node-red with the DEBUG statement

DEBUG output:

root@iot2000:/usr/lib/node_modules/node-red# DEBUG=node_red_contrib_modbus_client,modbus-serial node red.js -v

Welcome to Node-RED
===================

24 Mar 19:36:49 - [info] Node-RED version: v0.14.6
24 Mar 19:36:49 - [info] Node.js  version: v4.2.4
24 Mar 19:36:49 - [info] Linux 4.4.13-yocto-standard ia32 LE
24 Mar 19:36:49 - [info] Loading palette nodes
 Setting debug for  factories_schema_helpers        to  false
 Setting debug for  factory_code_generator          to  false
 Setting debug for  factories                       to  false
 Setting debug for  ua_variable                     to  false
 Setting debug for  monitored_item                  to  false
 Setting debug for  tools_event_filter              to  false
 Setting debug for  ua_variable_type                to  false
 Setting debug for  argument_list                   to  false
 Setting debug for  message_builder                 to  false
 Setting debug for  tcp_transport                   to  false
 Setting debug for  client_tcp_transport            to  false
 Setting debug for  client_secure_channel_layer     to  false
 Setting debug for  client_base                     to  false
 Setting debug for  client_session                  to  false
 Setting debug for  opcua_client                    to  false
 Setting debug for  node_crawler                    to  false
 Setting debug for  client_subscription             to  false
 Setting debug for  opcua_server                    to  false
 Setting debug for  condition                       to  false
 Setting debug for  subscription                    to  false
 Setting debug for  server_publish_engine           to  false
 Setting debug for  server_session                  to  false
24 Mar 19:38:13 - [info] Dashboard version 2.1.0 started at /ui
24 Mar 19:38:15 - [warn] ------------------------------------------------------
24 Mar 19:38:15 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
24 Mar 19:38:15 - [warn] [serialport] Error: Could not locate the bindings file. Tried:
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/build/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/build/Debug/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/build/Release/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/out/Debug/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/Debug/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/out/Release/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/Release/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/build/default/serialport.node
 ? /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/compiled/4.2.4/linux/ia32/serialport.node
24 Mar 19:38:15 - [warn] [OpcUa-Item] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
24 Mar 19:38:15 - [warn] [OpcUa-Client] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
24 Mar 19:38:15 - [warn] [OpcUa-Browser] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
24 Mar 19:38:15 - [warn] [OpcUa-Server] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
24 Mar 19:38:15 - [warn] [OpcUa-Event] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
24 Mar 19:38:15 - [warn] ------------------------------------------------------
24 Mar 19:38:15 - [info] Settings file  : /home/root/.node-red/settings.js
24 Mar 19:38:15 - [info] User directory : /home/root/.node-red
24 Mar 19:38:15 - [info] Flows file     : /home/root/.node-red/flows_iot2000.json
24 Mar 19:38:15 - [info] Creating new flow file
24 Mar 19:38:15 - [info] Starting flows
24 Mar 19:38:15 - [info] Started flows
24 Mar 19:38:15 - [info] Server now running at http://192.168.2.50:1880/
24 Mar 19:40:54 - [info] Stopping flows
24 Mar 19:40:54 - [info] Stopped flows
24 Mar 19:40:54 - [info] Starting flows
24 Mar 19:40:56 - [info] Started flows
24 Mar 19:40:59 - [warn] [modbus-client:b59492e.5ed4e7] Client -> try to reconnect by init in 2000 ms TCP@192.168.100.20:502 default Unit-Id: 1
  modbus-serial {} +0ms
  modbus-serial Port Not Open +22ms
[... 15 lines cut ...]
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,1,0,0,0,103,1,3,100,16,107,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0]},"buffer":{"type":"Buffer","data":[1,3,100,16,107,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,0,0,0,17,0,0,0,0,191,128,0,0,191,128,0,0,47,244]}} +11ms
.
.
.
[... cut ...]
  modbus-serial connect ECONNREFUSED 192.168.100.20:502 +2ms

timeout: 1 second
reconnection timeout: 2 seconds

In my program it's possible to 'refuse' incoming calls, node-red then tries to reconnect as you can see. BUT if I enable incoming calls within the reconnection time then there are no problems. IF I enable incoming calls > reconnection timeout (e.g. > 2 seconds) then there is no reconnection anymore as you can see. The last line is the final one. If I restart node-red then the connection is established again. 

-Rene-

biancode, 2017-03-24:
Please:

`npm install node-red-contrib-modbus@1.0.9-delta --unsafe-perm --build-from-source`

or if it is global installed

`npm install -g node-red-contrib-modbus@1.0.9-delta --unsafe-perm --build-from-source`

then start with all debug information

`DEBUG=node_red_contrib_modbus,node_red_contrib_modbus_client,modbus-serial node red.js -v > modbustcp.log`

and activate the  `Log states changes` in the TCP configuration of the Modbus node in Node-RED and send or upload the file.

Your TCP@192.168.100.20:502 sends much more as requested.

The timeout handle is on the 'modbus-serial' package and raises before reconnecting starts - the reconnect timeout is just to have a delay before reconnecting if that reconnect timeout is gone it reconnects and sends a request again. (see FSM on Wiki)

joshuatreenl, 2017-03-24:
The requested logfile:
[20170324 2350 modbustcp2_log.txt](https://github.com/biancode/node-red-contrib-modbus/files/869395/20170324.2350.modbustcp2_log.txt)
It looks like, maybe I am wrong, nothing has changed. 

>Your TCP@192.168.100.20:502 sends much more as requested.
The reason that more data is send is because different datatypes are involved, REAL4, INT2, INT4 etc. 

>The timeout handle is on the 'modbus-serial' package and raises before reconnecting starts - the >reconnect timeout is just to have a delay before reconnecting if that reconnect timeout is gone it >reconnects and sends a request again. (see FSM Wiki)
What I don't see in Node-Red is a timeout message. The timeout message only appears when I take out the physical networkconnection. With the software 'disable' the Modbus Read stays in its initialize state (see picture). I don't know what the software does if a connection is refused. Will it stay in the 'error' state or should the software perform a reconnection after the reconnection timeout e.g. open teh port again. 
 
[image]

biancode, 2017-03-25:
try this and set the right quantity:

[{"id":"5966e7a4.771fb","type":"modbus-read","z":"d7af511c.6d85f8","name":"","showStatusActivities":true,"showErrors":true,"unitid":"","dataType":"HoldingRegister","adr":"0","quantity":"8","rate":"1","rateUnit":"s","server":"e7f913fa.0fd6b","x":390,"y":220,"wires":[["64fcecb9.a05cbc"],["29c1a490.0c8074"]]},{"id":"29c1a490.0c8074","type":"modbus-response","z":"d7af511c.6d85f8","name":"","registerShowMax":20,"x":600,"y":240,"wires":[]},{"id":"64fcecb9.a05cbc","type":"debug","z":"d7af511c.6d85f8","name":"","active":true,"console":"false","complete":"false","x":580,"y":200,"wires":[]},{"id":"e7f913fa.0fd6b","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"92.168.100.20","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"100","clientTimeout":"1000","reconnectTimeout":"2000"}]

biancode, 2017-03-25:
For me, it looks like a problem with the modbus-serial. There are two ways to leave the Init-State: 1. Modbus is connected 2. Error occurs to catch, but nothing happens.

Please a last try with the settings of the flow before:

`DEBUG=node_red_contrib_modbus,node_red_contrib_modbus_client,node_red_contrib_modbus:client,modbus-serial node red.js -v > modbustcp.log`

and pot the log again

biancode, 2017-03-25:
Please:

npm install node-red-contrib-modbus@1.0.9-epsilon --unsafe-perm --build-from-source

or if it is global installed

npm install -g node-red-contrib-modbus@1.0.9-epsilon --unsafe-perm --build-from-source

`DEBUG=node_red_contrib_modbus,node_red_contrib_modbus:client,modbus-serial node red.js -v > modbustcp.log`

biancode, 2017-03-25:
it takes 80 sec. but it reconnecting

> 25 Mar 10:45:45 - [info] Starting flows
>   node_red_contrib_modbus:client Client -> initialized TCP@192.168.1.20:502 default Unit-Id: 1 +0ms
> 25 Mar 10:45:45 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> reconnect in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
>   node_red_contrib_modbus:client Client -> event: init old: NEW new: INIT TCP@192.168.1.20:502 default Unit-Id: 1 +10ms
> 25 Mar 10:45:45 - [info] Started flows
>   node_red_contrib_modbus:client Client -> TCP port TCP@192.168.1.20:502 default Unit-Id: 1 +2s
> 25 Mar 10:47:02 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> try to reconnect by init in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
> 25 Mar 10:47:04 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> reconnect in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
>   node_red_contrib_modbus:client Client -> event: init old: BROKEN new: INIT TCP@192.168.1.20:502 default Unit-Id: 1 +1m
>   node_red_contrib_modbus:client Client -> TCP port TCP@192.168.1.20:502 default Unit-Id: 1 +2s
> 25 Mar 10:48:22 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> try to reconnect by init in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
> 25 Mar 10:48:24 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> reconnect in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
>   node_red_contrib_modbus:client Client -> event: init old: BROKEN new: INIT TCP@192.168.1.20:502 default Unit-Id: 1 +1m
>   node_red_contrib_modbus:client Client -> TCP port TCP@192.168.1.20:502 default Unit-Id: 1 +2s
> 25 Mar 10:49:41 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> try to reconnect by init in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
> 25 Mar 10:49:43 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> reconnect in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
>   node_red_contrib_modbus:client Client -> event: init old: BROKEN new: INIT TCP@192.168.1.20:502 default Unit-Id: 1 +1m
>   node_red_contrib_modbus:client Client -> TCP port TCP@192.168.1.20:502 default Unit-Id: 1 +2s
> 25 Mar 10:51:00 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> try to reconnect by init in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
> 25 Mar 10:51:02 - [warn] [modbus-client:a9f457b.e26c1a8] Client -> reconnect in 2000 ms TCP@192.168.1.20:502 default Unit-Id: 1
>   node_red_contrib_modbus:client Client -> event: init old: BROKEN new: INIT TCP@192.168.1.20:502 default Unit-Id: 1 +1m
>   node_red_contrib_modbus:client Client -> TCP port TCP@192.168.1.20:502 default Unit-Id: 1 +2s

joshuatreenl, 2017-03-25:
Hi,

Ok, I've installed the node red case. Although 8 bytes are requested still more data is send. 
[20170325_1040_modbustcp_log.txt](https://github.com/biancode/node-red-contrib-modbus/files/869905/20170325_1040_modbustcp_log.txt)

Please look inside the logfile and see when I'll stop and start the software. 

On your last reply I'll check if after xx seconds the reconnection will start again. 
I did not yet install the last version.

biancode, 2017-03-25:
with the 1.0.9-eta version you should wait six or seven minutes after it is in BROKEN it takes the time to go to INIT and then it takes the time to connect - It uses the reconnect delay time twice on BROKEN.

biancode, 2017-03-25:
It is OK that the node stays in `initialize` - there is no change, if the connection does not work on connecting.

joshuatreenl, 2017-03-25:
The previous version befor eta gives: 
  node_red_contrib_modbus:client Client -> TCP port TCP@192.168.100.20:502 default Unit-Id: 1 +2s
  modbus-serial { action: 'port open error',
  modbus-serial   error:
  modbus-serial    { [Error: connect ECONNREFUSED 192.168.100.20:502]
  modbus-serial      code: 'ECONNREFUSED',
  modbus-serial      errno: 'ECONNREFUSED',
  modbus-serial      syscall: 'connect',
  modbus-serial      address: '192.168.100.20',
  modbus-serial      port: 502 } } +17ms
  modbus-serial {"code":"ECONNREFUSED","errno":"ECONNREFUSED","syscall":"connect","address":"192.168.100.20","port":502} +19ms
  modbus-serial connect ECONNREFUSED 192.168.100.20:502 +9ms
^C  node_red_contrib_modbus:client Client -> event: failure old: INIT new: FAILED TCP@192.168.100.20:502 default Unit-Id: 1 +8m
  node_red_contrib_modbus:client Client -> event: break old: FAILED new: BROKEN TCP@192.168.100.20:502 default Unit-Id: 1 +11ms
  node_red_contrib_modbus:client Client -> close node TCP@192.168.100.20:502 default Unit-Id: 1 +8ms

I waited 8 minutes. 
I will install the eta version you mentioned. I'll reply to you waht happens!

Thanks a lot again.

biancode, 2017-03-25:
`close node TCP@192.168.100.20:502 default Unit-Id: 1` was wrong - it should work for you

joshuatreenl, 2017-03-25:
You mean it was wrong in the previous version? 
Just curious why will the reconnection will wait for 6 or 7 minutes. Is this number somewhere set in a settings file?

joshuatreenl, 2017-03-25:
The 'close node' isexecuted after I hit CTLR-C...

joshuatreenl, 2017-03-25:
Still building eta...

biancode, 2017-03-25:
6 or 7 minutes is just to see multiple reconnects

joshuatreenl, 2017-03-25:
Unfortunately: 

@192.168.100.20:502 default Unit-Id: 1 +2s
  node_red_contrib_modbus:client Client -> TCP port TCP@192.168.100.20:502 default Unit-Id: 1 +2s
  modbus-serial { action: 'port open error',
  modbus-serial   error:
  modbus-serial    { [Error: connect ECONNREFUSED 192.168.100.20:502]
  modbus-serial      code: 'ECONNREFUSED',
  modbus-serial      errno: 'ECONNREFUSED',
  modbus-serial      syscall: 'connect',
  modbus-serial      address: '192.168.100.20',
  modbus-serial      port: 502 } } +17ms
  modbus-serial {"code":"ECONNREFUSED","errno":"ECONNREFUSED","syscall":"connect","address":"192.168.100.20","port":502} +15ms
  modbus-serial connect ECONNREFUSED 192.168.100.20:502 +2ms
  node_red_contrib_modbus:client Client -> event: failure old: INIT new: FAILED TCP@192.168.100.20:502 default Unit-Id: 1 **+15m**
  node_red_contrib_modbus:client Client -> event: break old: FAILED new: BROKEN TCP@192.168.100.20:502 default Unit-Id: 1 +22ms
  node_red_contrib_modbus:client Client -> close node TCP@192.168.100.20:502 default Unit-Id: 1 +18ms

After 15 minutes still no reconnection. 
What I observe is that, if the conn refused, still init I can DEPLOY new software and the modbus sofware is not 'hanging'. After a DEPLOY conenctions are correctly opened and works fine. Somewhere in the modbus software a reset after xx times is not executed. 
Another point, could I see if the correct Modbus (eta-version) is installed?

joshuatreenl, 2017-03-25:
This is the build info:

root@iot2000:/usr/lib/node_modules# npm install node-red-contrib-modbus@1.0.9-eta --unsafe-perm --build-from-source
WARN engine node-red-contrib-modbus@1.0.9-eta: wanted: {"node":">=6.9"} (current: {"node":"4.2.4","npm":"2.14.12"})
WARN engine node-modbus@3.2.6: wanted: {"node":">=4.7.0"} (current: {"node":"4.2.4","npm":"2.14.12"})
-
> serialport@4.0.7 install /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
npm WARN unmet dependency /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/tar-pack requires debug@'~2.2.0' but will load
npm WARN unmet dependency /usr/lib/node_modules/node-red/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/debug,
npm WARN unmet dependency which is version 2.3.3
node-red-contrib-modbus@1.0.9-eta node-red-contrib-modbus
├── stately.js@1.3.0
├── debug@2.6.3 (ms@0.7.2)
├── modbus-serial@5.2.3
├── node-modbus@3.2.6 (net@1.0.2, stampit-log@0.3.0, stampit-state-machine@0.2.1, stampit-event-bus@0.1.1, crc@3.4.4, bluebird@3.5.0)
├── stampit@2.1.2 (supermixer@1.0.3, lodash@3.10.1)
└── serialport@4.0.7 (bindings@1.2.1, commander@2.9.0, lie@3.1.1, nan@2.5.1, object.assign@4.0.4)

biancode, 2017-03-25:
Use `npm install modbus-serial` and follow the Wiki and README of that package to test without node-red what will happen.

joshuatreenl, 2017-03-25:
It seams that (you already mentioned) it's not the node red modbus implementation. Look at the logfile from the logger.js application. More or less the same....

[20170325_1932 logger_js.txt](https://github.com/biancode/node-red-contrib-modbus/files/870259/20170325_1932.logger_js.txt)

joshuatreenl, 2017-03-31:
Hi

You followed the discussion in modbus-serial. I came to the conclusion that some error handling should be installed in the Node-Red part. It could be automatically, then you might to have change the Modbus Read part (and other Modbus parts) or add new functions Modbus Close of Modbus Clear. If I encounter an error than I have to close the connection. 
Another option could be an extra input (asynchrone trigger) on the Modbus Read and other modules e.g. Reset Modbus trigger.  I can set this trigger if I encounter a timeout or connection refused. 
Maybe you have some other ideas as well?

biancode, 2017-04-08:
should work with v1.0.9+

closed 2017-04-08 by biancode