BiancoRoyal/node-red-contrib-modbus #11: Problem with installation contrib-modbus

chodki, 2017-02-02:
Hello,
I need help with installation node-red-contrib-modbus, on raspberry pi3
I do everything according to the instructions but in the dashboard I can't see elements modbus.

Here the installation process:
[image]

After this, when I start node-red i see error in loading modbus module:
[image]

Node-red is is up to date, on raspberry I use:
RASPBIAN JESSIE WITH PIXEL
Version:January 2017
Release date:2017-01-11
Kernel version:4.4 

Will someone tell me what' s going on?!

Thx Ł.

biancode, 2017-02-02:
see [Wiki](https://github.com/biancode/node-red-contrib-modbus/wiki/Raspberry-PI2-and-PI3-hints) - you node version should be wrong - please test with node -v

chodki, 2017-02-03:
Oh my God....
I didn't notice topic about pi in wiki :))
Sorry!
Now everything works perfectly, 
Goodwork Klaus!!
THX!

biancode, 2017-02-03:
cool - have fun!

closed 2017-02-03 by biancode