BiancoRoyal/node-red-contrib-modbus #9: Feature request - option for modbus-reader to nulls on failed polls

sirockin, 2017-02-02:
For our application, we'd like the receiving nodes to detect when a poll has failed so they can update their value to null. Currently the only way to 'know' that a poll has failed is to intercept the node's status information. We propose a 'send nulls on fail' option in the modbus-reader which will cause the reader to send an array of null values when the poll fails.

(We've created an implementation of this based on version 0.9.x but are currently unable to fork this properly or do a pull request since your git history doesn't go back to 0.9 and we have a problem with version 1.x (isssue #6). We'll create a pull request once that's resolved).

biancode, 2017-02-02:
use the node-red catch node to get errors and handle them - if there aren't data, why we should send data of null and which other packages work that way in node-red?

biancode, 2017-02-02:
and please think about that case, if you got values and for just some read errors you get null - that case is to handle by everyone - why, if you can work with errors in catch node?

sirockin, 2017-02-03:
While I accept that the catch node is one way to handle errors, there are also good reasons not to use it:

1. The flow of information is not so clear - the catch node has no obvious visual relation to the reader node.
2. Semantically, null values do make sense as part of information flow, rather than simply a gap in messaging which could simply be interpreted as an unchanged value.
3. Implementing the flow we require would need the output from the catch node to be multiplexed back into the output from the reader. Given that node red is aimed at providing simple readable flows, this seems to make things unnecessarily complicated.

We accept that our suggested approach is not for everyone which is why we suggested (and in our fork have implemented) a check box which allows users to choose this behavour or not.

biancode, 2017-03-23:
please reopen with your pull request later

closed 2017-03-23 by biancode