BiancoRoyal/node-red-contrib-modbus #12: Help in writing a function to work FC16

omelchuk890, 2017-02-10:
The function 16 Preset multiple registers must be submitted to the block input is an array of numbers. 
Have some input from the server mqtt convert from strings to numbers function:
var newMsg = { payload:parseFloat( msg.payload)*100 } ; 
return newMsg;
"27.85" -> 2785
Help to write a function with MULTIPLE inputs that creates an array for the block (modbus write).
When using FC6, everything works, but needs more blocks and passes to write data.
I apologize for the use of a software translator.
I have a problem how to make function several data.
Advance very grateful!
Beginner, not young, by the user.

biancode, 2017-02-10:
If I understand you correctly, you need to build an array of numbers or from numbers:

first translate or give names to MQTT variables by function nodes like:

msg.payload.myVariableNumber1 = parseFloat(msg.payload)*100;

msg.payload.myVariableNumber2 = parseFloat(msg.payload)*100;

msg.payload.myVariableNumber3 = parseFloat(msg.payload)*100;
....

later concatenate values in array of 16Bit chunks (max 16Bit values, otherwise you have to split 32Bit into two 16Bit for Modbus):
msg = { payload: [msg.payload.myVariableNumber1, msg.payload.myVariableNumber2, msg.payload.myVariableNumber3] } ;
return msg;

... was that your question? Otherwise use [Google Translation](https://translate.google.com/), please!

omelchuk890, 2017-02-11:
Thank you for the quick reply, Klaus.
Try more vividly and clearly explain your question.
It is in the attached picture.
I hope now You have clearer my problem is to create a Node-Red function with MULTIPLE INPUTS.
Sincerely, Alexander.

> If I understand you correctly, you need to build an array of numbers or from numbers:
> first translate or give names to MQTT variables by function nodes like:
> msg.payload.myVariableNumber1 = [...]

Александр Омельчук

omelchuk890, 2017-02-11:
Dear Klaus.
I figured out how to do it!
And it works!!!
The problem was the storage of information, to the generation of the array and the method of its installation in the desired position in the array.
Now I clearly indicate what and where should be.

-------- Пересылаемое сообщение --------
От кого: Омельчук Александр <omelchuk890@mail.ru>
Кому: biancode/node-red-contrib-modbus <reply@reply.github.com>
Дата: Суббота, 11 февраля 2017, 8:42 +03:00
Тема: Re[2]: [biancode/node-red-contrib-modbus] Help in writing a function to work FC16 (#12)

Thank you for the quick reply, Klaus.
Try more vividly and clearly explain your question.
It is in the attached picture.
I hope now You have clearer my problem is to create a Node-Red function with MULTIPLE INPUTS.
Sincerely, Alexander.

> If I understand you correctly, you need to build an array of numbers or from numbers:
> first translate or give names to MQTT variables by function nodes like:
> msg.payload.myVariableNumber1 = [...]

Александр Омельчук

----------------------------------------------------------------------

Александр Омельчук

closed 2017-02-11 by biancode