BiancoRoyal/node-red-contrib-modbus #15: RTU GPIO

markovip21, 2017-03-01:
Hi, 
It would be nice if you would post the GPIO for the RTU version you are using.

biancode, 2017-03-03:
Hi,
on PI or what do you mean exactly, please?

markovip21, 2017-03-04:
I mean, what pins (Rx, Tx) are used to transfer data to for example MAX485. Are they a standard Rpi serial ports? Do you have an option for additional pin for send/receive control on RS484?

biancode, 2017-03-05:
I'm using USB connectors and not the GPIO.

markovip21, 2017-03-05:
Me too. But as you let to chose /dev/ttyAMA0, I think you could tell what pins are used.

Thank you

biancode, 2017-03-05:
if you just want to connect to /dev/ttyAMA0, then just type it into that field, and it should work for you

biancode, 2017-03-09:
maybe @Kymbo59 can help ...

biancode, 2017-03-23:
I never used the pins - but try RPi GPIO TX and RX pins 8 and 10 or see [Infos](http://elinux.org/RPi_Serial_Connection) or [GPIO Settings](https://github.com/rewolff/bw_rpi_tools/blob/master/gpio/README)

[PI3 pins 14 and 15](http://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/)

closed 2017-03-23 by biancode

philesgilles, 2017-05-16:
A question again about GPIO. 
Did anybody managed to make it work? it's working fine with USB but using GPIO, I can see with an oscilloscope that the RX message comes back but the node says 'Empty' and then timeout.

on Raspberry PI3 using Stratopi board. and activated ttyAMA0