BiancoRoyal/node-red-contrib-modbus #16: failure after reboot

Kymbo59, 2017-03-06:
Klaus. I have the simple flow below....

[{"id":"8515c712.d09718","type":"modbus-read","z":"1df3eb3.add9115","name":"PLC Read","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"0","quantity":"5","rate":"2","rateUnit":"s","server":"fcde1705.2791e8","x":148.75,"y":199.75,"wires":[["27e42ae6.332f96"],[]]},{"id":"27e42ae6.332f96","type":"modbus-write","z":"1df3eb3.add9115","name":"Arduino Write","showStatusActivities":false,"unitid":"","dataType":"MHoldingRegisters","adr":"0","quantity":"5","server":"f268b976.c7d2","x":423.75,"y":194.75,"wires":[[],[]]},{"id":"fcde1705.2791e8","type":"modbus-client","z":"","name":"TCP","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.2.150","tcpPort":"502","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"f268b976.c7d2","type":"modbus-client","z":"","name":"buffered","clienttype":"simpleser","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","serialPort":"/dev/ttyAMA0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

If I build this from new, it works fine until I reboot the RPI, then the serial connection fails (I am guessing) and get the error message 'FSM not ready to write'.
If I delete everything associated with this flow, serial configurations, TCP configurations and start again as if it was a fresh install of node-red-contrib-modbus and rebuild all of the nodes, it works again until I reboot the RPI. I have 'upgraded and updated' and I still have the same issue.
Once working it will work for days without any problems....no timeouts...nothing. The reboot is causing the failure.

The RPI is reading 5 registers at 2 second intervals from a PLC and sending the result to a XBEE radio connected to the serial port (GPIO) on the RPI. When I have had it working, I have pushed the amount of data to much higher levels without any issues. I have reduced the amount of data here purely to remove 'volume of data' as being a possibility of cause of the failure.

Any clues????

biancode, 2017-03-09:
Do you use v1.0.8? Which node.js version is it? How about --unsafe-perm --build-from-source options on npm install?

biancode, 2017-03-09:
Did you try just to deploy a bit later instead of rebuilding the whole flow? Does it work, if you push Deploy a minute after reboot? Maybe it takes a bit time to connect the TTY.

closed 2017-04-08 by biancode