BiancoRoyal/node-red-contrib-modbus #17: Port Not Open when interrogating PLC with different IP on same port

bnjroos, 2017-03-07:
Hello,
When I try to query two different PLC via modbus tcp so with different IPs but both on port 502, I can query the first one and the second one once, and after it shows the Port Not Open error.
Here is my flow after deployment:
[image]
Both server configuration:
[image]
[image]
It is working once and after it shows the error: port not open
I have tried to add delay in between, it is not working either.
[image]
[image]

What is the work around for that ?
Thank you in advance,
Benjamin

biancode, 2017-03-07:
Hi Benjamin,

please, can you also post the flow in JSON?

A) Which version of package is it - looks old?
B) What is your version of node.js?
C) What is your host and operating system?
D) Please try, is it in latest version, too?

Some devices send a close for the port, if some is wrong in communication, that was fixed in v1.0.8.

bnjroos, 2017-03-07:
It just updated the package from 1.0.6 to 1.0.8 and it seems to work perfectly now..
I am using a docker container for RPi with nodered v0.16.2.
Thank you,

Benjamin

closed 2017-03-07 by bnjroos

agorkoli, 2018-12-28:
I have a similar issue but with serial RS485
here is my flow
The transaction keep timing out and I am getting "FSM Reset On State QUEUEING"
I am new to Node Red 
Any help will be much appreciated.

[{"id":"cb9888f1.fc8e88","type":"modbus-flex-getter","z":"6a9f636c.67183c","name":"","showStatusActivities":true,"showErrors":true,"logIOActivities":false,"server":"d7d2fb3d.dd73f8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":750,"y":300,"wires":[[],["9a9178e1.dae918"]]},{"id":"26f91b20.b5ee34","type":"function","z":"6a9f636c.67183c","name":"FC4 0x3003","func":"msg.payload = {\n    'fc': 4,\n    'unitid': 1,\n    'address' : 0x3003,\n    'quantity': 2\n};\nreturn msg;","outputs":1,"noerr":0,"x":450,"y":280,"wires":[["cb9888f1.fc8e88"]]},{"id":"3e26c778.979138","type":"inject","z":"6a9f636c.67183c","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":210,"y":260,"wires":[["26f91b20.b5ee34"]]},{"id":"9a9178e1.dae918","type":"modbus-response","z":"6a9f636c.67183c","name":"","registerShowMax":20,"x":950,"y":460,"wires":[]},{"id":"d7d2fb3d.dd73f8","type":"modbus-client","z":"6a9f636c.67183c","name":"","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyAMA0","serialType":"RTU","serialBaudrate":"115200","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":0,"commandDelay":1,"clientTimeout":3000,"reconnectTimeout":2000}]

hathemi, 2019-04-08:
@agorkoli  have you found a solution for that ? what was the problem ?

miniu89, 2019-07-26:
I have the same problem.

I am trying to conect to Universal Robot Modbus Server.

I am using:
Node_RED v0.20.7
Node.js v10.16.0
Everything is install on Windows 10 IoT system.
Modbus Node v4.1.3

[image]

Anybody can help?