BiancoRoyal/node-red-contrib-modbus #18: Node red restarting....

dedhdude, 2017-03-19:
I am using Raspberry Pi 3 as a hardware.
First when i tried "Modbus read" node with Serial ASCII , 7 data bit, 1 Stop Bit, EVEN parity it worked fine.
But when i tried after one month with same above configuration it restarting  node red continously.
When i deploy the above configuration in node red it is not getting connected and showing waiting.
But when i edit my node to RTU with 8Data bit, 1 stop bit, Even Parity it gets connected with attaching any device showing the error "Time Out", but in "ASCII" it is not getting connected to serial Port showing status "Waiting" with no error & getting restart continously.

[image]
[image]
[image]

I had tried flashing OS and reinstall node red & your package 6 to 7 times but still i am getting the same problem i don't know what mistake i am making. Please help Me with this Problem.

When I am Installing the package the following error is showing.
And at last it is also showing some warning of "package.json".

[image]

biancode, 2017-03-19:
Node.js 6.10 isn't tested, please use 6.9.x and check install hints in the wiki. Use the build problems options for npm see README.

biancode, 2017-03-19:
Otherwise install the last working version with node-red-contrib-modbus@vx.x.x  see npm info for the package.

biancode, 2017-03-23:
Hi, 
does it work for you?

closed 2017-04-08 by biancode

matthiaswinkeler, 2017-04-25:
Hi,

I have been using contrib-modbus for a couple of months now, I must say I really like the way it works.

but I seem to have a similar issue regarding nodered restarting after an error on modbus-serial.
WhenI delete all my flows and do a update nodejs and nodered and a rebuild the error just keeps showing up. The only thing I can do is start from scratch with a new sd card.

Now I have ran nodered with modbus read commands for months without issues but once the issue is there I cant get rid of it. 

"Type error: Cannot read property '_timeouthandle' of undefined."
[image]

The code for these modbus commands:
[{"id":"97b09704.0df068","type":"modbus-getter","z":"7ec49d8d.97a494","name":"STATUS HIJS","showStatusActivities":true,"showErrors":true,"unitid":"","dataType":"InputRegister","adr":"531","quantity":"48","server":"23fd0be5.348fb4","x":373.5,"y":747.5,"wires":[["5d387df1.084ff4"],[],[]]},{"id":"c77d91d.e1e197","type":"modbus-getter","z":"7ec49d8d.97a494","name":"RPM HIJS","showStatusActivities":true,"showErrors":true,"unitid":"","dataType":"InputRegister","adr":"97","quantity":"2","server":"23fd0be5.348fb4","x":368,"y":798,"wires":[["ea385827.713ff8"],["c4d8b46c.db6b78"],["c4d8b46c.db6b78"]]},{"id":"65558aa5.b83654","type":"inject","z":"7ec49d8d.97a494","name":"RPM POLL RATE","topic":"","payload":"true","payloadType":"bool","repeat":"1","crontab":"","once":true,"x":129,"y":908,"wires":[["c77d91d.e1e197"]]},{"id":"ee01293b.fa0148","type":"inject","z":"7ec49d8d.97a494","name":"STATUS POLL RATE","topic":"","payload":"true","payloadType":"bool","repeat":"1","crontab":"","once":true,"x":132,"y":585,"wires":[["97b09704.0df068"]]},{"id":"23fd0be5.348fb4","type":"modbus-client","z":"","name":"HIJSWERKEN N","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.3","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"100","clientTimeout":"2000","reconnectTimeout":"2000"}]  

I have the lates contrib-modbus installed, and did an upgrades etc...
Is there anyone who has had this too? Or are there any solutions regarding this issue?

Thanks!!

biancode, 2017-04-26:
@matthiaswinkeler new version of modbus-serial with v1.0.10

matthiaswinkeler, 2017-04-26:
@biancode & yaacov

Thanks for your work,
I installed your new code and checked using npm ls --depth the version numbers and I had the latest. 

I tested it once but the error returned. I will make a complete fresh install of everything. And import the flow code that caused the error again. 

Hopefully it will work then

matthiaswinkeler, 2017-04-26:
After a complete fresh install I ran into the error again. I checked I have the latest modbus-serial & contrib-modbus. 

But I imported the flow code into nodered that caused the error. So strange that once the error is there it doesnt want to go.

Do you guys have any other ideas?

biancode, 2017-04-26:
try to disable "Inject once at start?"

biancode, 2017-04-26:
`[{"id":"8ba872f3.0af41","type":"modbus-read","z":"2eaa3a51.aeaf96","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"InputRegister","adr":"531","quantity":"48","rate":"1","rateUnit":"s","server":"4751bca.ef7f844","x":330,"y":520,"wires":[[],[]]},{"id":"e306e26b.c3cc2","type":"modbus-read","z":"2eaa3a51.aeaf96","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"InputRegister","adr":"97","quantity":"2","rate":"1","rateUnit":"s","server":"4751bca.ef7f844","x":330,"y":580,"wires":[[],[]]},{"id":"4751bca.ef7f844","type":"modbus-client","z":"2eaa3a51.aeaf96","name":"HIJSWERKEN N","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.3","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":100,"clientTimeout":2000,"reconnectTimeout":2000}]`

matthiaswinkeler, 2017-04-26:
[image]

Slightly different error now, after update. 

I changed the modbus regquests with to code i recieved.

biancode, 2017-04-26:
a) The configuration stays if you delete just the flow.
b) The RTU settings and serial stuff are exported by node-red, because of they stay hidden in the config node, but they will never be used if TCP is active. It will just be used if Serial is active.