BiancoRoyal/node-red-contrib-modbus #19: New connection Modbus RTU over TCP

nokautai, 2017-03-21:
Hello. Great Job. But can You implement Modbus RTU over TCP connection - standard serial Modbus RTU queries/answers thru TCP/IP channel? I have RS485 to Ethernet gateway which can't convert Modbus TCP quries to Modbus RTU format, only waiting connection as RS485 TCP server and allows transparent flow.

http://www.simplymodbus.ca/TCP.htm
https://github.com/stephane/libmodbus/issues/192

biancode, 2017-03-21:
did you try the TCP RTU?

[image]

nokautai, 2017-03-21:
I can't see RTU and RTU-BUFFERED in TCP list, only TELNET and C701. Node version 1.0.8.
[image]

biancode, 2017-03-21:
you can see the more with `npm info` and package name
Please, try node-red-contrib-modbus@1.0.9-alpha - with my gateway it works a bit but not at all.
I'm working on it.

nokautai, 2017-03-21:
Thank you very much! On TCP->RTU-BUFFERED not respond, but combination TCP->TELNET gets response from MODBUS device.
[image]
[image]

closed 2017-03-23 by biancode

M0DDer, 2022-09-12:
Version node-red-contrib-modbus 5.23.1
ModBus over RTU still works ONLY with "Telnet" selected
p,s. in my moxa converter RTU is selected  (not telnet. telnet also avaliable here)