BiancoRoyal/node-red-contrib-modbus #20: All modbus nodes should only debug if RED.settings.verbose

Hugobox, 2017-03-23:
Hi!
Could all the modbus nodes act like the modbus-server and only print warnings in debug if NR is set to verbose?

I believe it would simply involve replacing in each node:
node.warn(failureMsg)
with:
if (RED.settings.verbose) {
	node.warn(failureMsg)
}

Let me know what you think!

biancode, 2017-03-23:
Hi,

Do you use v1.0.8?

Hugobox, 2017-03-23:
Yes!

biancode, 2017-03-23:
may that could be a request on node-red itself ;)

biancode, 2017-03-23:
you can set your node-red log level to error

Hugobox, 2017-03-23:
Its not the console logging that's problematic, its the debug panel in the editor. As I work on my dev machine which is not connected to all the modbus devices, the debug panel gets filled with modbus warnings and it gets harder to troubleshoot other parts of the flow. I'd like a way to let those nodes know that I know they aren't connected to the real system and RED.settings.verbose seems like a great solution, don't you agree?

biancode, 2017-03-23:
No, I do not agree in that case, but I'll give a new 'Show Errors' option to handle this.

Hugobox, 2017-03-23:
That would totally work! Thanks for your time!

biancode, 2017-03-23:
please try node-red-contrib-modbus@1.0.9-beta

Hugobox, 2017-03-23:
Thanks! Totally working for me!

closed 2017-03-23 by biancode