BiancoRoyal/node-red-contrib-modbus #21: Init via file input

joshuatreenl, 2017-04-11:
I don't know Node Red is the right language to use, but I would love to see that the ModbusRead function will have an input (actually two). On this input it would be great to 'feed' ModbusRead with configuration details (parameters). The advantage is that there is no need to fill in the parameters in Node Red manually, but simple feeding it via (for example) a configuration file. The disadvantage is that there should be another input where you can reset, init, and run the ModbusRead function. Just a thought... 

Kind Regards,
René

biancode, 2017-04-11:
For the most of your thoughts, it works with the ModbusFlexReader, which can be triggered with JSON to read what you want dynamically. In Node-RED concept, you just have 0..1 inputs and 0..n outputs. If you can describe some real Use Cases, then we could find a way to realize them.

joshuatreenl, 2017-04-12:
I presume that you might talk about the ModbusFlexGetter [MFG] (Node Red) instead of the ModbusFlexReader? I will look if MFG is suitable for my application.

closed 2017-09-23 by biancode