BiancoRoyal/node-red-contrib-modbus #23: Serial modbus goes to waiting after re-deploy

nygma2004, 2017-04-26:
Whenever I deploy my solution which deploys my Modbus Read nodes as well, the nodes go to waiting status and nothing happens. The have multiple modbus reads for the same modbus slave, but I also noticed the same if there is only 1. I have the nodes queued. Resetting the queue does not help. 

What I noticed that if I change the server settings, for example change the data bits from 8 to 7 and deploy the nodes start communicating again. Of course I need to set 7 back to 8 otherwise I am not getting any data. 

So it seems I need to change the server settings to make the nodes start communicating again.

biancode, 2017-04-26:
Please give some information about your environment (nodejs, node-red, contrib-modbus versions)

nygma2004, 2017-04-26:
node-red: 0.16.2
node-js: 6.10.1
node-red-contrib-modbus: 1.0.8

Looking at the palette, it seems I also have this:
node-red-contrib-serial-modbus: 0.0.10

Sorry, I don't remember if this is a dependency or an older version. I had a different modbus serial component before, but when I upgraded node-red it stopped working. That's when I installed your node.

biancode, 2017-04-26:
OK and 1.0.9 or 1.0.10 doesn't work for you?

biancode, 2017-04-26:
try `DEBUG=node_red_contrib_modbus_client,modbus-serial node-red -v > issue23.log`
and attach that log file here, please

nygma2004, 2017-04-26:
I added the debug statements, and everything appears to be working now. But I had this issue consistently throughout the day, and now it is gone. I am planning on playing around with modbus, so I will keep and eye on this. 
I will remember to come back next week as well, and if I am having the problem any more, I will close the issue. Sorry for the inconvenience caused.

biancode, 2017-04-26:
Does it work, if you switch the debug off again? Otherwise, it could be a time coupled problem.

nygma2004, 2017-04-27:
OK, have not thought of that. Sorry, but I have to ask for more assistance at this point. How do I turn off debug?

nygma2004, 2017-04-27:
OK, I am sure I have done something wrong. I was adding some error handling to the flow and made other changes and well, and I noticed the read going to wait state again. I also added a debug to the modbus queue (set the debug to complete message), but still nothing is showing up in the debug window.

Furthermore the debug is not showing much detail:
[issue23.txt](https://github.com/biancode/node-red-contrib-modbus/files/960793/issue23.txt)

I think I have to ask you a bit more detailed instructions on how to enable / disable debugging. Do I have to restart node-red after? I would also update to the latest version of contrib-nodered to make it easier for you. I want to work with the latest version too.

biancode, 2017-04-27:
a) How do I turn off debug? --->   just start node-red without the -v and DEBUG
b) it looks like node-red is started twice - are you on a device with a preinstalled Node-RED? Then you have to stop node-red before you call the command line

`DEBUG=node_red_contrib_modbus,node_red_contrib_modbus:client,modbus-serial node-red -v > issue23.log`

nygma2004, 2017-04-28:
Hi Klaus,

I managed to re-create the issue. I have contrib-modbus updated to 1.0.10 now. If I re-deploy the flow it is working. It was running with the debug on for about 30 min. 
I changed the timing on one of the modbus reads from 5 to 3 seconds. And after deploy all modbus reads gone to waiting. I changed the bits to 7 which resulted timeouts and back to 8 bits and now everything is working again. You should see all this towards the end of the document.

[issue23.txt](https://github.com/biancode/node-red-contrib-modbus/files/963684/issue23.txt)

biancode, 2017-04-28:
sounds like a reading problem, please give me the flows - there are many inside - exec starts and there are async starts:

> 28 Apr 09:25:52 - [info] Starting modified flows
> 28 Apr 09:25:52 - [warn] [modbus-client:SDM120] Client -> reconnect in 5000 ms Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 30
> 28 Apr 09:25:52 - [info] [inject:Check] repeat = 1000
> 28 Apr 09:25:52 - [info] Started modified flows
> 28 Apr 09:25:53 - [info] [exec:1 wire] /opt/openhab/onewiretemp.sh 28-00000433cda6 1493364353635
> 28 Apr 09:26:03 - [info] [exec:1 wire] /opt/openhab/onewiretemp.sh 28-00000433cda6 1493364363658
> 28 Apr 09:26:13 - [info] [exec:1 wire] /opt/openhab/onewiretemp.sh 28-00000433cda6 1493364373680

Two starts will never close the port and then your serial device will be blocked because it is open.

It directly reconnecting so the device doesn't work or got errno's - please activate Show Errors for the Modbus connection config - normally it should give you an error of Modbus

> 28 Apr 08:33:33 - [info] Starting flows
> 28 Apr 08:33:33 - [warn] [modbus-client:SDM120] Client -> reconnect in 5000 ms Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 30

nygma2004, 2017-04-28:
This is my flow. Since yesterday, I have connected another device on the same modbus line. SDM energy meter is 30 the temperature sensor is 50.
The nodes for the energy meter still works, but node for the temp is in waiting state. I did configure to server for them as the  unit number is in the service configuration. Maybe this was a problem. Still the original issue persist.

[{"id":"ac104c.0bcfefb8","type":"exec","z":"2934a51a.93393a","command":"/opt/openhab/onewiretemp.sh 28-00000433cda6","addpay":true,"append":"","useSpawn":"","timer":"","name":"1 wire","x":265,"y":63.5,"wires":[["26c38162.eb20ce","6c0d2f2.9ceaed"],[],[]]},{"id":"26c38162.eb20ce","type":"debug","z":"2934a51a.93393a","name":"","active":false,"console":"false","complete":"payload","x":450,"y":48,"wires":[]},{"id":"f1b57fcf.8f31","type":"inject","z":"2934a51a.93393a","name":"","topic":"","payload":"","payloadType":"date","repeat":"10","crontab":"","once":false,"x":117,"y":66,"wires":[["ac104c.0bcfefb8"]]},{"id":"8b486fd4.cdc7","type":"debug","z":"2934a51a.93393a","name":"","active":true,"console":"false","complete":"false","x":644.1666717529297,"y":1937.5833740234375,"wires":[]},{"id":"bda32a0f.183418","type":"function","z":"2934a51a.93393a","name":"Read buffer","func":"var regs = msg.payload;\nvar humid = (((regs[0] & 0x00FF) << 8) | (regs[1] & 0xFF))/10;\nvar temp = (((regs[2] & 0x00FF) << 8) | (regs[3] & 0xFF))/10;\n\n// msg.payload = \"{ \\\"temp\\\": \"+temp+\", \\\"humid\\\": \"+humid+\"}\";\n\nvar rsws = { temp:0, humid:0 };\nrsws.temp = temp;\nrsws.humid = humid;\n\nmsg.payload = rsws;\n\n// Update the status with current timestamp\nvar now = new Date();\nvar yyyy = now.getFullYear();\nvar mm = now.getMonth() < 9 ? \"0\" + (now.getMonth() + 1) : (now.getMonth() + 1); // getMonth() is zero-based\nvar dd  = now.getDate() < 10 ? \"0\" + now.getDate() : now.getDate();\nvar hh = now.getHours() < 10 ? \"0\" + now.getHours() : now.getHours();\nvar mmm  = now.getMinutes() < 10 ? \"0\" + now.getMinutes() : now.getMinutes();\nvar ss  = now.getSeconds() < 10 ? \"0\" + now.getSeconds() : now.getSeconds();\nnode.status({fill:\"blue\",shape:\"ring\",text:\"Last update: \"+dd + \".\" + mm + \".\" + yyyy + \" \" + hh + \":\" + mmm + \":\" + ss});\n\nreturn msg;","outputs":1,"noerr":0,"x":336.1666717529297,"y":1937.5833740234375,"wires":[["8b486fd4.cdc7","605870f1.4c06f","4b7f9241.5f9e3c"]]},{"id":"4b7f9241.5f9e3c","type":"ui_text","z":"2934a51a.93393a","group":"a82c54da.58e038","order":1,"width":0,"height":0,"name":"","label":"Temperature","format":"{{msg.payload.temp}} C","layout":"row-spread","x":634.1666717529297,"y":2004.5833740234375,"wires":[]},{"id":"605870f1.4c06f","type":"ui_text","z":"2934a51a.93393a","group":"a82c54da.58e038","order":2,"width":0,"height":0,"name":"","label":"Humidity","format":"{{msg.payload.humid}} %","layout":"row-spread","x":627.1666717529297,"y":2054.5833740234375,"wires":[]},{"id":"b3ae0448.822d08","type":"comment","z":"2934a51a.93393a","name":"SDM120 Energy Meter","info":"","x":141.5,"y":1142,"wires":[]},{"id":"53b4b356.bf124c","type":"modbus-read","z":"2934a51a.93393a","name":"SDM 120 Voltage","showStatusActivities":true,"showErrors":true,"unitid":"30","dataType":"InputRegister","adr":"0","quantity":"2","rate":"5","rateUnit":"s","server":"40f20c7f.13a934","x":113,"y":1235,"wires":[["e4d3fade.1132a8"],[]]},{"id":"e4d3fade.1132a8","type":"function","z":"2934a51a.93393a","name":"Voltage","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"voltage\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":342,"y":1228,"wires":[["94b7c2b7.b61ae","267d6c75.311c84"]]},{"id":"29fc9cac.869404","type":"modbus-queue-info","z":"2934a51a.93393a","name":"SDM120_queue","unitid":"30","lowLowLevel":"1","lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"40f20c7f.13a934","errorOnHighLevel":false,"x":484.6428527832031,"y":1602.107177734375,"wires":[["759e42e9.23762c"]]},{"id":"27ead677.91de0a","type":"inject","z":"2934a51a.93393a","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":113.64286041259766,"y":1603.1072025299072,"wires":[["1d0b599c.544786"]]},{"id":"1d0b599c.544786","type":"change","z":"2934a51a.93393a","name":"Reset queue","rules":[{"t":"set","p":"resetQueue","pt":"msg","to":"true","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":282.64286041259766,"y":1603.1071891784668,"wires":[["29fc9cac.869404"]]},{"id":"e3643b3e.be3a88","type":"function","z":"2934a51a.93393a","name":"Current","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"current\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":341,"y":1286,"wires":[["fa18b5f3.ba6be8","267d6c75.311c84"]]},{"id":"4dbf9277.487e4c","type":"function","z":"2934a51a.93393a","name":"Power","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nin [... cut ...] ":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"50","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"72b1a4dc.4f488c","type":"ui_tab","z":"","name":"Modbus","icon":"memory","order":15},{"id":"156af96f.f8fd27","type":"ui_tab","z":"","name":"Home","icon":"home","order":1}]

biancode, 2017-04-28:
node-red shows that it is not a valid flow

biancode, 2017-04-28:
But in the text it looks like you try to use the /dev/ttyUSB0 twice, which is not possible in serial - you have just on open in serial - just set the Unit-Id in the config to a default and use it on all nodes. The Unit-Id can be set by the reading to 30 or 50 or what you need and it will override the setting in the config node. Serial connections are one to one connections.

biancode, 2017-04-30:
Please, reopen if it doesn't fix your problem.

closed 2017-04-30 by biancode

nygma2004, 2017-04-30:
I did what you said. The Unit id on the server is default and there is now only one server for the entire modbus serial line.
The SDM120 modbus reads are working OK, but the new temperature sensor node is only reading data like very minute instead of the 5 seconds it is set to. I don't see any error raised, even though I have a catch on the modbus node. When I connect only the temperature sensor on the line, the values are being read every 5 seconds as usual. Do you have any clues?

[2_modbus.txt](https://github.com/biancode/node-red-contrib-modbus/files/966626/2_modbus.txt)