BiancoRoyal/node-red-contrib-modbus #25: how to detect that timeout happened

mohamad8, 2017-04-28:
I am using node-red and have trouble to detect that timeout happened when I am reading data through the Modbus-RTU Buffered. In flow page it just shows me timeout (show activity below the node). but I can not take advantage of it!!

biancode, 2017-04-29:
The timeout appears if your device doesn't answer to the request in the range of your timeout time (1 sec. per default). Try to give more time 2 sec. for example and tell me how it works for you. If it doesn't work, ask @yaacov how the timeout works in detail, please. Be sure that you are using 1.0.11 of the package and node.js >= 6.9

mohamad8, 2017-04-30:
@biancode Thank you for your reply
Maybe I ask my question in bad way.
I meant if timeout happened, how can I be informed. for example, send email notification when time out happened.

biancode, 2017-04-30:
Oh No, my mistake - sorry -  Yes you can watch the status by the 'Status' node of Node-RED.

ignored err.message:

* Timed out
* FSM Not Ready
* Port Not Open

These are 'soft' errors, which are handled by the client connection and they will fix automatically (ex: reconnect) by the Modbus-Client or their occurrences are too much to bring it to node.error (debug view). So the 'soft' errors got a state in the node, and you can watch that states.

[image]

mohamad8, 2017-05-01:
@biancode 
that is awesom, I will try it out.
Thank you for your time!

biancode, 2017-05-02:
Have fun!

closed 2017-05-02 by biancode