BiancoRoyal/node-red-contrib-modbus #26: TypeError: Cannot read property 'then' of undefined

Proone01, 2017-05-02:
Hi, 

I am using a Simatic IoT2040 and i want to communicate with a Adam-4015 (Rs485-PT100 converter) but i got this error when i deploy  with a modbus node :

Welcome to Node-RED
===================

2 May 12:55:57 - [info] Node-RED version: v0.15.2
2 May 12:55:57 - [info] Node.js  version: v6.9.2
2 May 12:55:57 - [info] Linux 4.4.18-yocto-standard ia32 LE
2 May 12:56:13 - [info] Loading palette nodes
2 May 12:57:15 - [info] Dashboard version 2.1.0 started at /ui
2 May 12:57:17 - [warn] ------------------------------------------------------
2 May 12:57:17 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
2 May 12:57:17 - [warn] ------------------------------------------------------
2 May 12:57:17 - [info] Settings file  : /home/root/.node-red/settings.js
2 May 12:57:17 - [info] User directory : /home/root/.node-red
2 May 12:57:17 - [info] Flows file     : /home/root/.node-red/flows_iot2000.json
2 May 12:57:18 - [info] Server now running at http://127.0.0.1:1880/
2 May 12:57:19 - [info] Starting flows
2 May 12:57:40 - [info] [inject:Receiving Simulation] repeat = 60000
2 May 12:57:51 - [info] [inject:Sending Simulation] repeat = 60000
2 May 12:57:53 - [info] [inject:32def808.29d7a8] repeat = 1000
2 May 12:57:53 - [info] Started flows
2 May 12:57:53 - [info] serial port /dev/ttyS2 opened at 9600 baud 8N1
2 May 13:03:26 - [info] Stopping modified nodes
2 May 13:03:27 - [info] Stopped modified nodes
2 May 13:03:27 - [info] Starting modified nodes
2 May 13:03:28 - [info] Started modified nodes
2 May 13:03:30 - [red] Uncaught Exception:
2 May 13:03:30 - TypeError: Cannot read property 'then' of undefined
    at Timeout.ModbusClientNode.node.connectClient [as _onTimeout] (/usr/lib/nod            e_modules/node-red-contrib-modbus/modbus/modbus-client.js:410:15)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

Thank !

biancode, 2017-05-02:
Hi,

with v1.0.11 of node-red-contrib-modbus? Please check you timeout config, it has to be a number in every field of the configuration.

biancode, 2017-05-02:
Please, use

`DEBUG=node_red_contrib_modbus:client,modbus-serial node-red -v > issue26connector.log`

and attach the resulting file here.

Proone01, 2017-05-03:
When i install the last version of conrtib-modbus, i got this :

root@iot2000:~# cd /usr/lib/node_modules
root@iot2000:/usr/lib/node_modules# npm install node-red-contrib-modbus

> serialport@4.0.7 install /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release/serialport.node" is installed via remote
/usr/lib
`-- node-red-contrib-modbus@1.0.11
  +-- debug@2.6.3
  | `-- ms@0.7.2
  +-- modbus-serial@5.2.9
  +-- node-modbus@3.2.6
  | +-- bluebird@3.5.0
  | +-- crc@3.4.4
  | +-- net@1.0.2
  | +-- stampit-event-bus@0.1.1
  | +-- stampit-log@0.3.0
  | `-- stampit-state-machine@0.2.1
  +-- serialport@4.0.7
  | +-- bindings@1.2.1
  | +-- commander@2.9.0
  | | `-- graceful-readlink@1.0.1
  | +-- lie@3.1.1
  | | `-- immediate@3.0.6
  | +-- nan@2.6.2
  | +-- node-pre-gyp@0.6.32
  | | +-- mkdirp@0.5.1
  | | | `-- minimist@0.0.8
  | | +-- nopt@3.0.6
  | | | `-- abbrev@1.0.9
  | | +-- npmlog@4.0.1
  | | | +-- are-we-there-yet@1.1.2
  | | | | +-- delegates@1.0.0
  | | | | `-- readable-stream@2.2.2
  | | | |   +-- buffer-shims@1.0.0
  | | | |   +-- core-util-is@1.0.2
  | | | |   +-- inherits@2.0.3
  | | | |   +-- isarray@1.0.0
  | | | |   +-- process-nextick-args@1.0.7
  | | | |   +-- string_decoder@0.10.31
  | | | |   `-- util-deprecate@1.0.2
  | | | +-- console-control-strings@1.1.0
  | | | +-- gauge@2.7.2
  | | | | +-- aproba@1.0.4
  | | | | +-- has-unicode@2.0.1
  | | | | +-- object-assign@4.1.0
  | | | | +-- signal-exit@3.0.2
  | | | | +-- string-width@1.0.2
  | | | | | +-- code-point-at@1.1.0
  | | | | | `-- is-fullwidth-code-point@1.0.0
  | | | | |   `-- number-is-nan@1.0.1
  | | | | +-- strip-ansi@3.0.1
  | | | | | `-- ansi-regex@2.0.0
  | | | | +-- supports-color@0.2.0
  | | | | `-- wide-align@1.1.0
  | | | `-- set-blocking@2.0.0
  | | +-- rc@1.1.6
  | | | +-- deep-extend@0.4.1
  | | | +-- ini@1.3.4
  | | | +-- minimist@1.2.0
  | | | `-- strip-json-comments@1.0.4
  | | +-- request@2.79.0
  | | | +-- aws-sign2@0.6.0
  | | | +-- aws4@1.5.0
  | | | +-- caseless@0.11.0
  | | | +-- combined-stream@1.0.5
  | | | | `-- delayed-stream@1.0.0
  | | | +-- extend@3.0.0
  | | | +-- forever-agent@0.6.1
  | | | +-- form-data@2.1.2
  | | | | `-- asynckit@0.4.0
  | | | +-- har-validator@2.0.6
  | | | | +-- chalk@1.1.3
  | | | | | +-- ansi-styles@2.2.1
  | | | | | +-- escape-string-regexp@1.0.5
  | | | | | +-- has-ansi@2.0.0
  | | | | | | `-- ansi-regex@2.0.0
  | | | | | +-- strip-ansi@3.0.1
  | | | | | | `-- ansi-regex@2.0.0
  | | | | | `-- supports-color@2.0.0
  | | | | +-- is-my-json-valid@2.15.0
  | | | | | +-- generate-function@2.0.0
  | | | | | +-- generate-object-property@1.2.0
  | | | | | | `-- is-property@1.0.2
  | | | | | +-- jsonpointer@4.0.0
  | | | | | `-- xtend@4.0.1
  | | | | `-- pinkie-promise@2.0.1
  | | | |   `-- pinkie@2.0.4
  | | | +-- hawk@3.1.3
  | | | | +-- boom@2.10.1
  | | | | +-- cryptiles@2.0.5
  | | | | +-- hoek@2.16.3
  | | | | `-- sntp@1.0.9
  | | | +-- http-signature@1.1.1
  | | | | +-- assert-plus@0.2.0
  | | | | +-- jsprim@1.3.1
  | | | | | +-- extsprintf@1.0.2
  | | | | | +-- json-schema@0.2.3
  | | | | | `-- verror@1.3.6
  | | | | `-- sshpk@1.10.1
  | | | |   +-- asn1@0.2.3
  | | | |   +-- assert-plus@1.0.0
  | | | |   +-- bcrypt-pbkdf@1.0.0
  | | | |   +-- dashdash@1.14.1
  | | | |   +-- ecc-jsbn@0.1.1
  | | | |   +-- getpass@0.1.6
  | | | |   +-- jodid25519@1.0.2
  | | | |   +-- jsbn@0.1.0
  | | | |   `-- tweetnacl@0.14.4
  | | | +-- is-typedarray@1.0.0
  | | | +-- isstream@0.1.2
  | | | +-- json-stringify-safe@5.0.1
  | | | +-- mime-types@2.1.13
  | | | | `-- mime-db@1.25.0
  | | | +-- oauth-sign@0.8.2
  | | | +-- qs@6.3.0
  | | | +-- stringstream@0.0.5
  | | | +-- tough-cookie@2.3.2
  | | | | `-- punycode@1.4.1
  | | | +-- tunnel-agent@0.4.3
  | | | `-- uuid@3.0.1
  | | +-- rimraf@2.5.4
  | | | `-- glob@7.1.1
  | | |   +-- fs.realpath@1.0.0
  | | |   +-- inflight@1.0.6
  | | |   | `-- wrappy@1.0.2
  | | |   +-- inherits@2.0.3
  | | |   +-- minimatch@3.0.3
  | | |   | `-- brace-expansion@1.1.6
  | | |   |   +-- balanced-match@0.4.2
  | | |   |   `-- concat-map@0.0.1
  | | |   +-- once@1.4.0
  | | |   | `-- wrappy@1.0.2
  | | |   `-- path-is-absolute@1.0.1
  | | +-- semver@5.3.0
  | | +-- tar@2.2.1
  | | | +-- block-stream@0.0.9
  | | | +-- fstream@1.0.10
  | | | | `-- graceful-fs@4.1.11
  | | | `-- inherits@2.0.3
  | | `-- tar-pack@3.3.0
  | |   +-- debug@2.2.0
  | |   | `-- ms@0.7.1
  | |   +-- fstream@1.0.10
  | |   | +-- graceful-fs@4.1.11
  | |   | `-- inherits@2.0.3
  | |   +-- fstream-ignore@1.0.5
  | |   | +-- inherits@2.0.3
  | |   | `-- minimatch@3.0.3
  | |   |   `-- brace-expansion@1.1.6
  | |   |     +-- balanced-match@0.4.2
  | |   |     `-- concat-map@0.0.1
  | |   +-- once@1.3.3
  | |   | `-- wrappy@1.0.2
  | |   +-- readable-stream@2.1.5
  | |   | +-- buffer-shims@1.0.0
  | |   | +-- core-util-is@1.0.2
  | |   | +-- inherits@2.0.3
  | |   | +-- isarray@1.0.0
  | |   | +-- process-nextick-args@1.0.7
  | |   | +-- string_decoder@0.10.31
  | |   | `-- util-deprecate@1.0.2
  | |   `-- uid-number@0.0.6
  | `-- object.assign@4.0.4
  |   +-- define-properties@1.1.2
  |   | `-- foreach@2.0.5
  |   +-- function-bind@1.1.0
  |   `-- object-keys@1.0.11
  +-- stampit@2.1.2
  | +-- lodash@3.10.1
  | `-- supermixer@1.0.3
  `-- stately.js@1.3.0

When i tried to use debug command i got this :

root@iot2000:~# DEBUG=node_red_contrib_modbus:client,modbus-serial node-red -v > issue26connector.log
-sh: node-red: command not found

Probably due to IoT2040.

Do you have an idea ?

biancode, 2017-05-07:
a) Sorry for the delay! Please try v1.0.12! There was a bug with the promise, which was fixed.

b) you have to find out where your node-red is installed on the device to do more - see [Link](https://www.rs-online.com/designspark/siemens-iot2020---installing-and-testing-node-red-mqtt-and-sqlite3)

biancode, 2017-05-07:
here is the [video for IoT2020](https://www.youtube.com/watch?v=5B_wJBD-LB8)

closed 2017-05-10 by biancode