BiancoRoyal/node-red-contrib-modbus #27: Reading multiple devices on the same line

nygma2004, 2017-05-03:
This is continuing on issue#23. I have two devices on the same modbus line, Unit 30 and 50. Both configured to the same comm settings. Unit 30 which is an SDM120 energy sensor has 5 modbus read boxes, and unit 50 which is a temp sensor 1 modbus read box. All linked to the same server, and all set to 5 second update interval.

The SDM120 modbus reads are working OK, but the new temperature sensor node is only reading data like very 20-30 minutes instead of the 5 seconds it is set to. I don't see any error raised (in the UI), even though I have a catch on the modbus node. When I connect only the temperature sensor on the serial line, the values are being read every 5 seconds as usual. Do you have any clues?

Below is some debug log:
[2_modbus.txt](https://github.com/biancode/node-red-contrib-modbus/files/973940/2_modbus.txt)

biancode, 2017-05-03:
Please export the flow from node-red and post it.

nygma2004, 2017-05-03:
The modbus part is about half way down in the flow. You can ignore the top.
[2_modbus.txt](https://github.com/biancode/node-red-contrib-modbus/files/974716/2_modbus.txt)

biancode, 2017-05-04:
I need a flow for node-red not the log data. ;)
Please give me logs and flow exports in separate files!

[image]

biancode, 2017-05-04:
Please, remember your last flow doesn't work in Node-RED Import. Test to import your flow after export.

nygma2004, 2017-05-04:
Sorry, I uploaded the wrong file. Find the entire flow below. It will import to a new 1W+Modbus tab.
[modbus_flow.txt](https://github.com/biancode/node-red-contrib-modbus/files/977637/modbus_flow.txt)

biancode, 2017-05-07:
flow looks good ...

sending","msg":{"topic":"polling","from":"RS-WS-N08","payload":{"unitid":"50","fc":3,"address":"0","quantity":"2"},"queueNumber":0,"queueUnit":50,"queueUnitId":50}} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 **+2s**
  modbus-serial {} +5ms
  modbus-serial **Timed out** +0ms

It looks like your Unit 50 doesn't answer every time

nygma2004, 2017-05-07:
If I disconnect device 30, and leave only 50 on the line, I will respond to each request, like every second.
What would you recommend? Reduce the baud rate? Have you seen anything similar in the past?

biancode, 2017-05-07:
No, for me it is a problem of serial. Now it works with parallel queues and it has to be only one queue. Your gateway will be confused by parallel requests, which is possible in TCP but not in serial. I'll have look to fix this issue and provide it to you for testing.

biancode, 2017-05-07:
It is RS485 right?

biancode, 2017-05-07:
try to set the queue delay to 10 or 20 and watch the queue with the queue node

biancode, 2017-05-07:
Can you set the baud rate to 115200 for all devices?

nygma2004, 2017-05-07:
Yes, I am using a USB-RS485 converted. I will change the delays and will let you know. 9600 baud is the fastest common speed between these two devices.

nygma2004, 2017-05-07:
Setting the queue delay to 10 or 20 does help. Result is the same.

biancode, 2017-05-07:
Try to go higher in intervals of 10. Your time is 6 requests per 5 sec. Every 5 sec. you produce 6 requests to the BUS. In that case you can go max to nearly 800 msec. for queue delay to bring all requests to the devices. The queue will send command by command with the delay of the queue settings.

biancode, 2017-05-07:
What about your [Flow](https://flows.nodered.org/flow/670ba131f74bd127d884aab2a580bffb) does it work with multiple units in the same flow?

nygma2004, 2017-05-07:
That is interesting. I just upped the queue delay to 30 ms, and it works now. I only have one of these temp sensors, so I could not test how it works with multiple device. 

But I have 3 more Modbus devices (16DO Adam unit, 8DO8DI Adam unit, and an HMI). I will start adding them to the modbus line and see the impact of having these many devices on the line.

Let me know if I should do any more testing.

biancode, 2017-05-07:
Please watch this [Video](https://www.youtube.com/watch?v=k993tAFRLSE)

closed 2017-05-07 by biancode