BiancoRoyal/node-red-contrib-modbus #29: Loosing connection to server when using more than 5 Modbus client

Kartheek1512, 2017-05-09:
Hi, I need to get data from 10 devices which are with different servers. I am using Modbus getter nodes. Pulling Qty: 50 registers from each device. Have 5 different modbus clients. When i run them individually, i am getting data. But, when i have them in one flow and hit Deploy,  the status of the nodes switch between "active" and "waiting" and later looses connection to server.  I have to remove the .json file and restart the whole process to come out of this error. Can anyone help me if there is any other way to get data from more than 5 modbus clients which are at different server addresses.

biancode, 2017-05-09:
a) Hi, do you have some log and version information? 
b) Is it possible to upload the flow to have a look? (file of clipboard export from node-red)
c) What does the queue info node tell you for each client when you add the devices one by one?
d) is the v1.0.12 installed?

@yaacov is there a restriction for multiple TCP connections? I mean no.

Kartheek1512, 2017-05-09:
Thanks for looking into it. 
a. Node-red version v0.16.2
   Node.js version v6.10.1
   i will try to record the log this time. 
b. I attached my flow in the txt file below. 
c. I am not using queue info. 
d. Does Modbus nodes have version? Sorry. I am new to this. Can you please let me know how to check. 

[Tracker combined data.txt](https://github.com/biancode/node-red-contrib-modbus/files/987915/Tracker.combined.data.txt)

closed 2017-05-09 by Kartheek1512

biancode, 2017-05-09:
Please, install again and you will get v1.0.13. Now it has some bug fixes from the core library for TCP transaction id handling.

Kartheek1512, 2017-05-09:
so do i have to run this command below in node-red directory? Will it upgrade itself?
npm install node-red-contrib-modbus

Kartheek1512, 2017-05-09:
[image]
I got this error.

yaacov, 2017-05-09:
> it has some bug fixes from the core library for TCP transaction id handling.

@biancode hi,
I do not know if this is related to this problem ... but I juse updated modbus-serial to v5.2.13, it should fix the `transactionId == NaN` you found.

biancode, 2017-05-09:
@yaacov v5.2.12 is now in v1.0.13 

@Kartheek1512 what is the command? You got v1.0.9, which is an old one.

yaacov, 2017-05-09:
> @yaacov v5.2.12 is now in v1.0.13

I added more fixes in v5.2.13 , you should update to get the fix for the `NaN` thing, but I do not know if it is related to this problem ...

Kartheek1512, 2017-05-09:
[image]

i am using npm install node-red-contrib-modbus

biancode, 2017-05-09:
Where or how was it installed before? Maybe you have to uninstall before and I mean you have to install in node-red folder, not in node_modules. It will use node_modules automatically.

Kartheek1512, 2017-05-09:
Oh.. I forgot. Thank you. It worked when i did it in node-red folder. i will run the flow again and let you know how it did.

biancode, 2017-05-09:
[{"id":"40c11ea2.6ac08","type":"modbus-queue-info","z":"129c512d.e72b7f","name":"Arcadia NCU1","unitid":"","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"f75b60f1.d5b12","errorOnHighLevel":false,"x":580,"y":900,"wires":[[]]},{"id":"f75b60f1.d5b12","type":"modbus-client","z":"","name":"Arcadia NCU1","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"166.156.46.207","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000}]

do that for all eight connections - see your configuration - some hint every Modbus connection is working in the background, it is used or not in nodes

biancode, 2017-05-09:
@yaacov v1.0.14 includes now v5.0.13 - Thank you!

Kartheek1512, 2017-05-09:
hi Biancode.. Thank you. But how to i send "Address" , "Quantity" and "FC" info into this node.

biancode, 2017-05-09:
never - just select the connection - see info on the right side of Node-RED

[image]

biancode, 2017-05-09:
watch [YouTube](https://www.youtube.com/watch?v=ND9ZYkMOwZs&index=6&list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH) to learn more

biancode, 2017-05-09:
You created mostly two connections to the same IP - hint: you can use only one per IP

Kartheek1512, 2017-05-09:
I am pulling registers 20600 to 20650 on one node and from 20800 to 20850 on another node. i dont need values from 20650 to 20800. So, i used two. I will look into your videos and see if I can use one node instead of using two. Thanks.

Kartheek1512, 2017-05-17:
Hi., The new modbus version is working without any issues. Thank you. I have another question. I want to pull registers 30100, 30200,30300,30400...39900.  As you see the registers are in increments of 100. Can you please guide me which modbus node to use and function program.