BiancoRoyal/node-red-contrib-modbus #30: Disable Nagle's algorithm for increased performance?

SkyTrix, 2017-05-15:
Could it be beneficial to disable Nagle's algorithm for faster throughput of Modbus messages? I was also wondering how the "queue commands" option in the Modbus Client node impacts performance. Should it be turned off for higher/faster throughput?

References:
https://nodejs.org/api/net.html#net_socket_setnodelay_nodelay
http://stackoverflow.com/questions/30232818/much-slower-response-in-node-js-http-server-with-smaller-files

biancode, 2017-05-15:
a) Nagle's algorithm - never tried - Let me know if you test!
b) all I can say is to read on the [Wiki](https://github.com/biancode/node-red-contrib-modbus/wiki/Queue-hints) - if it has performance issues with 1 msec., then you have a bad config or device - can you upload or send the flow?

biancode, 2017-05-15:
inspect the queue with the [queue info node](https://github.com/biancode/node-red-contrib-modbus/wiki/Queue-info-node)

biancode, 2017-05-17:
Some news?

SkyTrix, 2017-05-17:
I've only had a quick look. From what I saw I don't think it's possible to access the actual Socket that sends/receives the Modbus messages and modbus-serial package must be forked to set the `setNoDelay` flag on the socket?

Considering the queue: I asked because I'm trying to see how scalable it is on one host. I'm running anywhere from 20 to 100 [OpenPLC](https://github.com/thiagoralves/OpenPLC_v2) containers at a 10ms polling rate. I saw averages of around 12 ms between Modbus packets and a maximum of around 200 ms between packets. I was thinking if both Nagle or the queue impact this.

biancode, 2017-05-17:
Think about that you need a response for every request. Sometimes it makes sense to read more as you need like: address 1 with quantity 1000 and pick out later the data you need - it is one request in msec. - If you ask one by one, you have to calculate max = msec. per request * requests ;)

SkyTrix, 2017-05-17:
Of course :) But they are all different Modbus slaves so I need a request for each one. It's not a real world example of course. I'm just trying to see how many different Modbus slaves I can reliably communicate with.

biancode, 2017-05-17:
node-red IO's have also some impact I measured - close the browser and let it running and switch off all debug verbose stuff, and the CPU will go down

biancode, 2017-05-17:
take a look to the .node-red/settings.js to configure some values to powerful use

SkyTrix, 2017-05-20:
Thanks for the help, that does indeed improve performance. I found the issue was that there was just not enough CPU time available to process the queue, it just kept growing. So there was no reason to look into Nagle further.

I just released a package based on this wonderful node: https://www.npmjs.com/package/node-red-contrib-openplc. I have attributed this project in de README. You may close this issue if you want.

closed 2017-05-27 by biancode