BiancoRoyal/node-red-contrib-modbus #31: Reading different registers in the same device +help reading some datas

philesgilles, 2017-05-16:
Good day! 

Great tools for Modbus! It's working perfectly!

Is it possible to read different registers using the same function for reading in flex getter?
I'm trying to read different registers with a different address on the same device (here 50521 (2x16 U32), 50528 (2x 16b U32) and 50770 U32. (see link below)
I've tried 

`msg.payload = {'fc': 3, 'unitid': 69, 'address': [50521,50528,50770] , 'quantity': 3 }`
but seems to read 50521 and 50522 and 50523.

I have another issue with a counter maybe somebody can help me. Some data are parsed in 2x16 bit to 8x16b see: http://www.socomec.fr/files/live/sites/systemsite/files/SCP/6_gestion_energie/countis/countis_e/Countis_E03.html

How can I get the data in one string (Text) see register 50050 8x16bit, or one buffer instead of 2x16bit?

philesgilles, 2017-05-16:
found all infos for reading the buffers here => https://nodejs.org/docs/v0.10.48/api/buffer.html
No more problems

still need help for the first issue

Best,

biancode, 2017-05-16:
Hi,

great to see that it works well for you!

Please try to fire all three commands at the same time:
three function nodes going to one flex getter at the same time and queue will handle the read one by one for you
```
msg.payload = {'fc': 3, 'unitid': 69, 'address': 50521 , 'quantity': 3 }
msg.payload = {'fc': 3, 'unitid': 69, 'address': 50528 , 'quantity': 3 }
msg.payload = {'fc': 3, 'unitid': 69, 'address': 50770 , 'quantity': 3 }
```

Have fun!

biancode, 2017-05-16:
or if you just want to read the three registers:

```
msg.payload = {'fc': 3, 'unitid': 69, 'address': 50521 , 'quantity': 1 }
msg.payload = {'fc': 3, 'unitid': 69, 'address': 50528 , 'quantity': 1 }
msg.payload = {'fc': 3, 'unitid': 69, 'address': 50770 , 'quantity': 1 }
```

philesgilles, 2017-05-16:
Thanks a lot! will try it out tonight seems legit...

What about passing the msg.topic to the buffered output? Is there a way to do it now?
Or would it be possible to add that feature in a future update? 

I would need the counter ID passing to the end of the pipe.

biancode, 2017-05-16:
From v1.1.0+ releases, you'll get every time the same interface in messages.

`[{payload: values, responseBuffer: response, input: msg}, {payload: response, values: values, input: msg}]`

philesgilles, 2017-05-16:
ok that's the problem I have 1.0.16 will update

biancode, 2017-05-16:
it will be released soon

philesgilles, 2017-05-16:
great thank's.

biancode, 2017-05-16:
search closed issues for ttyAMA0

biancode, 2017-05-16:
a Wiki article would be cool about ttyAMA0 if you got some experiences

biancode, 2017-05-16:
OK try v1.1.0 it is released now

closed 2017-05-16 by biancode