BiancoRoyal/node-red-contrib-modbus #32: Getting Modbus register values in increments of 100

Kartheek1512, 2017-05-17:
Hi Klaus,

   I am trying to read holding registers at addresses 30100, 30200,.....39900 (99 registers in increments of 100). When i use Modbus getter, i can only read in increments of 1 when i have Quantity as 99.  So, I am getting 30100 to 30199. But I want values at every 100th register. Can you please help me what to do. I am really stuck here. Thank you for the help in advance.

biancode, 2017-05-17:
Hi,

Please, use the the Flex Getter and a function node before, which increments by 100 and sends requests for data to the Felx Getter. See example in the Info tab of Node-RED ;)

Have fun!

biancode, 2017-05-17:
A hint: quantity still 1, just address will grow in the command

biancode, 2017-05-17:
try that:

`[{"id":"bff03bfd.661dd8","type":"modbus-flex-getter","z":"85c09234.fa528","name":"","showStatusActivities":false,"showErrors":false,"server":"8e111151.ebb69","x":500,"y":660,"wires":[["c92bae89.03534"],["88a03643.5c41b8"]]},{"id":"6a037581.e2190c","type":"function","z":"85c09234.fa528","name":"Read every 100","func":"let address = 0\nlet incrementAddress = 1\n\nfor(let i = 0; i < 99; i++) {\n    address += incrementAddress;\n    msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}\naddress += incrementAddress; // last msg here to send a object at the end\nmsg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\nreturn msg;","outputs":1,"noerr":0,"x":280,"y":660,"wires":[["bff03bfd.661dd8"]]},{"id":"91ed128e.b76ac","type":"inject","z":"85c09234.fa528","name":"","topic":"","payload":"100","payloadType":"num","repeat":"10","crontab":"","once":false,"x":110,"y":660,"wires":[["6a037581.e2190c"]]},{"id":"88a03643.5c41b8","type":"modbus-response","z":"85c09234.fa528","name":"","registerShowMax":20,"x":730,"y":700,"wires":[]},{"id":"c92bae89.03534","type":"debug","z":"85c09234.fa528","name":"","active":true,"console":"false","complete":"true","x":690,"y":640,"wires":[]},{"id":"13a4ba0.c426c46","type":"modbus-server","z":"85c09234.fa528","name":"","logEnabled":false,"serverPort":10502,"responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":220,"y":740,"wires":[[],[],[]]},{"id":"31599e2e.9c6c02","type":"modbus-queue-info","z":"85c09234.fa528","name":"","unitid":"","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"8e111151.ebb69","errorOnHighLevel":false,"x":440,"y":740,"wires":[[]]},{"id":"8e111151.ebb69","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

Kartheek1512, 2017-05-17:
Hi Klaus, 

   Thank you for looking into it. Even after using the code, I get values from the same register 31200. It is not increment. Can you please check the code below. Thank you.

[{"id":"7d2fd983.3031d8","type":"modbus-flex-getter","z":"c53ab3da.86062","name":"","showStatusActivities":false,"showErrors":false,"server":"73b0f8ad.982fa8","x":818.5078125,"y":317.5234375,"wires":[["29929b26.f8fdc4"],["97073167.95028"]]},{"id":"b5ef5593.59cb38","type":"function","z":"c53ab3da.86062","name":"Read every 100","func":"let address = 30100;\nlet incrementAddress = 100;\n\nfor(let i = 0; i < 10; i++) {\n address =address+incrementAddress;\n msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n node.send(msg);\n}\naddress += incrementAddress; // last msg here to send a object at the end\nmsg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\nreturn msg;","outputs":1,"noerr":0,"x":598.5078125,"y":317.5234375,"wires":[["7d2fd983.3031d8"]]},{"id":"18332939.ee15d7","type":"inject","z":"c53ab3da.86062","name":"","topic":"","payload":"100","payloadType":"num","repeat":"","crontab":"","once":false,"x":428.5078125,"y":317.5234375,"wires":[["b5ef5593.59cb38"]]},{"id":"97073167.95028","type":"modbus-response","z":"c53ab3da.86062","name":"","registerShowMax":20,"x":1048.5078125,"y":357.5234375,"wires":[]},{"id":"29929b26.f8fdc4","type":"debug","z":"c53ab3da.86062","name":"","active":true,"console":"false","complete":"payload","x":1038.5078125,"y":297.5234375,"wires":[]},{"id":"2130af28.46a12","type":"modbus-server","z":"c53ab3da.86062","name":"","logEnabled":false,"serverPort":10502,"responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":538.5078125,"y":397.5234375,"wires":[[],[],[]]},{"id":"8041dda.17f172","type":"modbus-queue-info","z":"c53ab3da.86062","name":"","unitid":"","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"c94113.aeddeef","errorOnHighLevel":false,"x":758.5078125,"y":397.5234375,"wires":[[]]},{"id":"73b0f8ad.982fa8","type":"modbus-client","z":"","name":"Medford NCU1","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"166.156.46.208","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"c94113.aeddeef","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

biancode, 2017-05-17:
try that - works for me - change the increment value:

[{"id":"6a037581.e2190c","type":"function","z":"85c09234.fa528","name":"Read every 100","func":"var address = 0;\n\nfor(var i = 0; i < 99; i++) {\n    address = address + 1;\n    msg = {};\n    msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}\n","outputs":1,"noerr":0,"x":260,"y":640,"wires":[["bff03bfd.661dd8"]]}]

biancode, 2017-05-17:
`[{"id":"314eab73.f556f4","type":"function","z":"85c09234.fa528","name":"input msg","func":"msg.payload = msg.input.payload\nreturn msg;","outputs":1,"noerr":0,"x":690,"y":640,"wires":[["c92bae89.03534","bf047d91.5bfec"]]}]`

Kartheek1512, 2017-05-17:
Hi Klaus, 
   wow. Thank you very much..You are like god to me.

closed 2017-05-17 by biancode