BiancoRoyal/node-red-contrib-modbus #33: Modbus flex getter -Output 1- No Data Array

Kartheek1512, 2017-05-17:
Hi, 
 I am using Modbus flex getter and its first output is not a Data array. I am getting the entire object with fc,unit id,address and quantity. But I need only the read register values. How can i do that?

[image]

[{"id":"b9b2b34.43cf15","type":"function","z":"b8b529fa.105f98","name":"Read every 100","func":"var address = 30000;\n\nfor(var i = 0; i < 10; i++) {\n address = address + 100;\n msg = {};\n msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n node.send(msg);\n}\n","outputs":1,"noerr":0,"x":672.5078125,"y":481.5234375,"wires":[["6320aa97.3f3514"]]}]

biancode, 2017-05-17:
try out that flow:

`[{"id":"bff03bfd.661dd8","type":"modbus-flex-getter","z":"85c09234.fa528","name":"","showStatusActivities":false,"showErrors":false,"server":"8e111151.ebb69","x":500,"y":660,"wires":[["314eab73.f556f4","c92bae89.03534","1772f596.0e832a"],["88a03643.5c41b8"]]},{"id":"6a037581.e2190c","type":"function","z":"85c09234.fa528","name":"Read every 100","func":"var address = 0;\nvar value = msg.payload;\n\nfor(var i = 0; i < 99; i++) {\n    address = address + 1;\n    msg = {};\n    msg.payload = { 'value': value, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}\n","outputs":1,"noerr":0,"x":280,"y":660,"wires":[["bff03bfd.661dd8"]]},{"id":"91ed128e.b76ac","type":"inject","z":"85c09234.fa528","name":"","topic":"","payload":"100","payloadType":"num","repeat":"15","crontab":"","once":false,"x":110,"y":660,"wires":[["6a037581.e2190c"]]},{"id":"88a03643.5c41b8","type":"modbus-response","z":"85c09234.fa528","name":"","registerShowMax":20,"x":830,"y":780,"wires":[]},{"id":"c92bae89.03534","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"true","x":790,"y":700,"wires":[]},{"id":"13a4ba0.c426c46","type":"modbus-server","z":"85c09234.fa528","name":"","logEnabled":false,"serverPort":10502,"responseDelay":"50","delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":220,"y":740,"wires":[[],[],[]]},{"id":"31599e2e.9c6c02","type":"modbus-queue-info","z":"85c09234.fa528","name":"","unitid":"","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"8e111151.ebb69","errorOnHighLevel":false,"x":440,"y":740,"wires":[[]]},{"id":"bf047d91.5bfec","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"payload","x":970,"y":660,"wires":[]},{"id":"314eab73.f556f4","type":"function","z":"85c09234.fa528","name":"input msg","func":"msg.payload = msg.input.payload\nreturn msg;","outputs":1,"noerr":0,"x":800,"y":660,"wires":[["bf047d91.5bfec"]]},{"id":"1772f596.0e832a","type":"debug","z":"85c09234.fa528","name":"","active":true,"console":"false","complete":"payload","x":810,"y":740,"wires":[]},{"id":"d1be22af.142ae","type":"modbus-flex-write","z":"85c09234.fa528","name":"","showStatusActivities":false,"showErrors":false,"server":"8e111151.ebb69","x":490,"y":860,"wires":[["b37ffe2e.acb27","3a44d947.0b38a6","38849533.df41fa"],["e4a896e0.95b218"]]},{"id":"94d0bee6.502a7","type":"inject","z":"85c09234.fa528","name":"","topic":"","payload":"100","payloadType":"num","repeat":"25","crontab":"","once":false,"x":110,"y":860,"wires":[["134c51e1.3c507e"]]},{"id":"e4a896e0.95b218","type":"modbus-response","z":"85c09234.fa528","name":"","registerShowMax":20,"x":830,"y":980,"wires":[]},{"id":"3a44d947.0b38a6","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"true","x":790,"y":900,"wires":[]},{"id":"d2954735.d9a4e8","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"payload","x":970,"y":860,"wires":[]},{"id":"b37ffe2e.acb27","type":"function","z":"85c09234.fa528","name":"input msg","func":"msg.payload = msg.input.payload\nreturn msg;","outputs":1,"noerr":0,"x":800,"y":860,"wires":[["d2954735.d9a4e8"]]},{"id":"38849533.df41fa","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"payload","x":810,"y":940,"wires":[]},{"id":"134c51e1.3c507e","type":"function","z":"85c09234.fa528","name":"Read every 100","func":"var address = 0;\nvar value = msg.payload;\nfor(var i = 0; i < 99; i++) {\n    address = address + 1;\n    msg = {};\n    msg.payload = { 'value': value , 'fc': 6, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}\n","outputs":1,"noerr":0,"x":280,"y":860,"wires":[["d1be22af.142ae"]]},{"id":"8e111151.ebb69","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

closed 2017-05-17 by biancode

biancode, 2017-05-17:
in that example you have a server with very quick 50msec response time and watch - the queue will be clean.

`[{"id":"bff03bfd.661dd8","type":"modbus-flex-getter","z":"85c09234.fa528","name":"","showStatusActivities":false,"showErrors":false,"server":"8e111151.ebb69","x":500,"y":660,"wires":[["314eab73.f556f4","c92bae89.03534","1772f596.0e832a"],["88a03643.5c41b8"]]},{"id":"6a037581.e2190c","type":"function","z":"85c09234.fa528","name":"Read every 100","func":"var address = 0;\nvar value = msg.payload;\n\nfor(var i = 0; i < 99; i++) {\n    address = address + 1;\n    msg = {};\n    msg.payload = { 'value': value, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}\n","outputs":1,"noerr":0,"x":280,"y":660,"wires":[["bff03bfd.661dd8"]]},{"id":"91ed128e.b76ac","type":"inject","z":"85c09234.fa528","name":"","topic":"","payload":"100","payloadType":"num","repeat":"15","crontab":"","once":false,"x":110,"y":660,"wires":[["6a037581.e2190c"]]},{"id":"88a03643.5c41b8","type":"modbus-response","z":"85c09234.fa528","name":"","registerShowMax":20,"x":830,"y":780,"wires":[]},{"id":"c92bae89.03534","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"true","x":790,"y":700,"wires":[]},{"id":"13a4ba0.c426c46","type":"modbus-server","z":"85c09234.fa528","name":"","logEnabled":false,"serverPort":10502,"responseDelay":"50","delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":220,"y":740,"wires":[[],[],[]]},{"id":"31599e2e.9c6c02","type":"modbus-queue-info","z":"85c09234.fa528","name":"","unitid":"","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"8e111151.ebb69","errorOnHighLevel":false,"x":440,"y":740,"wires":[[]]},{"id":"bf047d91.5bfec","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"payload","x":970,"y":660,"wires":[]},{"id":"314eab73.f556f4","type":"function","z":"85c09234.fa528","name":"input msg","func":"msg.payload = msg.input.payload\nreturn msg;","outputs":1,"noerr":0,"x":800,"y":660,"wires":[["bf047d91.5bfec"]]},{"id":"1772f596.0e832a","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"payload","x":810,"y":740,"wires":[]},{"id":"d1be22af.142ae","type":"modbus-flex-write","z":"85c09234.fa528","name":"","showStatusActivities":false,"showErrors":false,"server":"8e111151.ebb69","x":490,"y":860,"wires":[["b37ffe2e.acb27","3a44d947.0b38a6","38849533.df41fa"],["e4a896e0.95b218"]]},{"id":"94d0bee6.502a7","type":"inject","z":"85c09234.fa528","name":"","topic":"","payload":"100","payloadType":"num","repeat":"15","crontab":"","once":false,"x":110,"y":860,"wires":[["134c51e1.3c507e"]]},{"id":"e4a896e0.95b218","type":"modbus-response","z":"85c09234.fa528","name":"","registerShowMax":20,"x":830,"y":980,"wires":[]},{"id":"3a44d947.0b38a6","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"true","x":790,"y":900,"wires":[]},{"id":"d2954735.d9a4e8","type":"debug","z":"85c09234.fa528","name":"","active":false,"console":"false","complete":"payload","x":970,"y":860,"wires":[]},{"id":"b37ffe2e.acb27","type":"function","z":"85c09234.fa528","name":"input msg","func":"msg.payload = msg.input.payload\nreturn msg;","outputs":1,"noerr":0,"x":800,"y":860,"wires":[["d2954735.d9a4e8"]]},{"id":"38849533.df41fa","type":"debug","z":"85c09234.fa528","name":"","active":true,"console":"false","complete":"payload","x":810,"y":940,"wires":[]},{"id":"134c51e1.3c507e","type":"function","z":"85c09234.fa528","name":"Read every 100","func":"var address = 0;\nvar value = msg.payload;\nfor(var i = 0; i < 99; i++) {\n    address = address + 1;\n    msg = {};\n    msg.payload = { 'value': value , 'fc': 6, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}\n","outputs":1,"noerr":0,"x":280,"y":860,"wires":[["d1be22af.142ae"]]},{"id":"8e111151.ebb69","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

Kartheek1512, 2017-06-01:
I tried this. It is working much faster now. Do you recommend any modbus node to which i can give the server address from function node?
As of now, I am using modbus flex getter and giving fc,unit id, address and quantity. I would like to give the IP address of the server and port too.