BiancoRoyal/node-red-contrib-modbus #34: Timeout problem with 1.1

philesgilles, 2017-05-18:
Hi ! 

I have a problem with the timeout since 1.1

When node is "ready"
First reads works well (but it doesnt come back to "ready", stays on yellow "reading done" and after 10-15sec of so, the getter is queueing and timeout.

It doesn't reconnect after Timeout.

Serial is configured like this :
9600 8N2
RTU-Buffered
connection delay 100ms
queue command yes
queue 2ms (Ive tried to put it up same problem)

It's the exact same code that was working before 1.1 (except format after the getter)

`[{"id":"2ea584ae.afc47c","type":"comment","z":"dfab6cca.74f9c","name":"Lire le Buffer 2x16 (U32 BE)","info":"","x":160,"y":60,"wires":[]},{"id":"92decbe2.624bb8","type":"inject","z":"dfab6cca.74f9c","name":"Compteur_1","topic":"Compteur_1","payload":"","payloadType":"date","repeat":"8","crontab":"","once":false,"x":140,"y":160,"wires":[["11c4e054.5cde6","b6290bc4.28a46","71fd4bd7.0890f4"]]},{"id":"11c4e054.5cde6","type":"function","z":"dfab6cca.74f9c","name":"Amps","func":"msg.payload = {'fc': 3, 'unitid': 69, 'address': 50528 , 'quantity': 2 };\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":160,"wires":[["27d84417.cc617c"]]},{"id":"b6290bc4.28a46","type":"function","z":"dfab6cca.74f9c","name":"Kw/h","func":"msg.payload = {'fc': 3, 'unitid': 69, 'address': 50770 , 'quantity': 2 };\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":200,"wires":[["27d84417.cc617c"]]},{"id":"71fd4bd7.0890f4","type":"function","z":"dfab6cca.74f9c","name":"Voltage","func":"msg.payload = {'fc': 3, 'unitid': 69, 'address': 50520 , 'quantity': 2 };\n\nreturn msg;","outputs":1,"noerr":0,"x":360,"y":120,"wires":[["27d84417.cc617c"]]},{"id":"9501b71e.5a3598","type":"debug","z":"dfab6cca.74f9c","name":"","active":true,"console":"false","complete":"true","x":930,"y":140,"wires":[]},{"id":"c831a01.df0ee6","type":"function","z":"dfab6cca.74f9c","name":"Format","func":"// setup topic to know wich counter it is (defined by the Inject node)\nmsg.topic = msg.input.topic;\n\n// setup adress to know wich register it is (defined by the getter function)\nmsg.address = msg.input.payload.address;\n\n// read and push the buffer in msg.payload\nmsg.payload = msg.responseBuffer.buffer.readUInt32BE(0);\n\nreturn msg;","outputs":1,"noerr":0,"x":720,"y":160,"wires":[["9501b71e.5a3598","34e2e20b.b2f15e"]]},{"id":"34e2e20b.b2f15e","type":"link out","z":"dfab6cca.74f9c","name":"Out Modbus buffer U32BE","links":["35f7dec5.58f1f2"],"x":895,"y":180,"wires":[]},{"id":"27d84417.cc617c","type":"modbus-flex-getter","z":"dfab6cca.74f9c","name":"Getter","showStatusActivities":true,"showErrors":false,"server":"54b98643.3d2b38","x":550,"y":160,"wires":[["c831a01.df0ee6","1c3314b3.8b7b8b"],[]]},{"id":"1c3314b3.8b7b8b","type":"modbus-queue-info","z":"dfab6cca.74f9c","name":"","unitid":"69","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"54b98643.3d2b38","errorOnHighLevel":false,"x":680,"y":40,"wires":[[]]},{"id":"54b98643.3d2b38","type":"modbus-client","z":"","name":"","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/Modbus-USB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"2","serialParity":"none","serialConnectionDelay":"100","unit_id":"","commandDelay":"2","clientTimeout":"1000","reconnectTimeout":"2000"}]`

philesgilles, 2017-05-18:
Might be my USB adapter will try later with an other

biancode, 2017-05-18:
As you can see on github, there is nothing changed in connecting and reconnecting - just output of nodes has changed

philesgilles, 2017-05-18:
very strange. I'm not at home will try with another USB adapter might be the cause then...

philesgilles, 2017-05-18:
Might be the USB adapter then. Will try later. Thanks

philesgilles, 2017-05-18:
I checked and my adapter was defect... sorry for that !

closed 2017-05-18 by philesgilles