BiancoRoyal/node-red-contrib-modbus #35: Read/write too many registers crashes red (buffer overrun)

varadi, 2017-05-22:
The Read/Write nodes do not respect the Modes PDU size limit.  Asking for to many registers (e.g., 130) creates a buffer overrun and crashes red. 

```bash
22 May 15:40:53 - [red] Uncaught Exception:
22 May 15:40:53 - TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1027:11)
    at Buffer.writeUInt8 (buffer.js:1075:5)
    at Object.<anonymous> (/var/node-red/node_modules/node-modbus/node-modbus/handler/server/ReadHoldingRegisters.js:58:14)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
22 May 15:40:54 - [info]
```

biancode, 2017-05-26:
It will take a bit time. We are working on a very new dynamically server and the updates for the old server.

varadi, 2017-05-26:
cool.  thank you.

biancode, 2017-09-23:
fixed from v2.x

closed 2017-09-23 by biancode