BiancoRoyal/node-red-contrib-modbus #36: Read 32 bit registers from PLC

ngservices, 2017-05-23:
Hello Klaus.

Firstly, i do sincerely apologize if this is the incorrect place to raise this issue. I would assume the issue is with me not your program. If you remove i will understand.
If not i was  hoping you could help me solve two remaining problems with the Node Red Modbus pack. ?
Firstly i'm a complete amateur and only just recently learnt  Modbus.  I can program in ladder and HMI's
Connect to any device as long as i have the address library,  i have learnt how to poll devices and curretly trying to understand the rest of what modbus is about.
I only discovered node red one week ago and was overwhelmed on the potential it can offer.

I have a PLC connected to Windows Node red. I have a small PLC connected to the computer
Coms is working OK. I'm able to read & write registers to an extend. I'm using Modbus serial- RS485
9600.8.1 

My fist issue is i'm unable to read 32bit registers from the PLC. 
For example i'm trying to read FC3 address 4 and address 5. value should be about 75400, i'm getting 49657.
There is no option to choose data type, so i appear to be only be able to read s-int.

Second issue is i cannot write values over 1 to the PLC. I presume some form of code would need to be written in the function node. I did manage to be be able to turn coils on off, this was achieved by using a change node and converting 1 to true.

I have no idea were to go from here, i have watched your videos and read an awful lot of material. Please could you help?
I could send over the flow if this is acceptable ?

Thanks in advance

Kind regards

biancode, 2017-05-23:
Hi,

have a look here at the [Wiki](https://github.com/biancode/node-red-contrib-modbus/wiki/Buffer-and-endianness---32Bit---64Bit)

You have to handle all in 16 Bit for now - so you have to split for writing and join to read. Search the closed issues for 32 to find more, please.

Find your way!

biancode, 2017-05-27:
If you need more help reopen, please.

closed 2017-05-27 by biancode