BiancoRoyal/node-red-contrib-modbus #37: why don't display the modbus-client nodes?

lcvcl, 2017-05-24:
i want have the function that i can config the ip or something else by myself,i think it's meaningful ,because the server can be change by oneself,so if have this function ,it will be conveniently

biancode, 2017-05-24:
Hi,

Sorry, but you have to explain more in detail. The config nodes are to find under configuration tab of Node-RED.

biancode, 2017-05-24:
Please htnk about that:

a) What is the problem?
b) In which case it will be conveniently?
c) Is there a real Use Case?
d) describe it in detail (Where, What, Why): "i want have the function that i can config the ip or something else by myself"

biancode, 2017-05-25:
Please, reopen if you need more.

closed 2017-05-25 by biancode

lcvcl, 2017-05-26:
Just as flex-getter can use msg to configure address, why not allow the slave's ip address to be configured using msg?
In the actual production, some companies do cloud platform convergence equipment need to set the ip from the station to read the information, but these IP is not necessarily static, the company may change the slave IP or add a new from Station, if it is written in HTML config, each time you need to change the configuration once again ip address, this is a very troublesome thing. But if the node can read information from the file, and then use its IP and other information for dynamic configuration of this is quite comfortable, then why not let IP address can be dynamically configured?
There is a problem in your project to see the existence of the client, but ask what is not the actual node?

biancode, 2017-07-02:
try v1.2.0