BiancoRoyal/node-red-contrib-modbus #38: Using FTDI serial to usb converter

mascapollo, 2017-05-25:
I've trying to connect to the Modbus using a FTDI chip USB to serial converter in Raspberry PI. The converter is recognized by Raspberry PI and I'm able to pick it from the  Modbus Flex Getter node. After deploy it, the node appears as connected, but when it receives the get request I get an error of the FSM not able to read. Is there any requirement to access this converters from this node?

biancode, 2017-05-25:
No, the connection client node is the same. If other nodes are working, then it should also work probably.

biancode, 2017-05-25:
log file and/or flow will help to find out more

mascapollo, 2017-05-25:
No modbus node works. I get the following message:
25 May 19:03:21 - [error] [modbus-flex-getter:MODBUS request] FSM Not Ready To Read

this is the device:

 new full-speed USB device number 8 using dwc_otg
[92771.291609] usb 1-1.3: New USB device found, idVendor=0403, idProduct=6001
[92771.291638] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[92771.291655] usb 1-1.3: Product: FT232R USB UART
[92771.291670] usb 1-1.3: Manufacturer: FTDI
[92771.291686] usb 1-1.3: SerialNumber: AI057JM0
[92771.300718] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[92771.301225] usb 1-1.3: Detected FT232RL
[92771.302475] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0

this is the node:

[{"id":"1ffa881e.d43378","type":"modbus-flex-getter","z":"3a3ed607.4d46fa","name":"MODBUS request","showStatusActivities":true,"showErrors":true,"server":"1fc73b5.5c448c5","x":228,"y":528,"wires":[["fe8a2a61.74e7c8"],["d8e9f60e.0f6b78"]]},{"id":"1fc73b5.5c448c5","type":"modbus-client","z":"","name":"USB converter","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU","serialBaudrate":"19200","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"500","unit_id":"","commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000}]

this is the payload passed to the node:

{ fc: 3, unitid: 2, address: 100, quantity: 10 }

There is not TX/RX activity on the wire.

mascapollo, 2017-05-25:
Just to mention that I have tested the adapter with minicon and connected to a windows with Putty and it works.

mascapollo, 2017-05-25:
Other test I've made with the serial node worked fine receiving ascii characters from a Putty.

[{"id":"fedb0cb7.3a3b38","type":"serial in","z":"3a3ed607.4d46fa","name":"prueba","serial":"e356c563.69623","x":184,"y":516,"wires":[["d8e9f60e.0f6b78"]]},{"id":"e356c563.69623","type":"serial-port","z":"","serialport":"/dev/ttyUSB0","serialbaud":"19200","databits":"8","parity":"none","stopbits":"1","newline":"20","bin":"false","out":"count","addchar":false}]

biancode, 2017-05-26:
Did you try the default of config: RTU-Buffered?

[{"id":"34b86216.257f6e","type":"modbus-flex-getter","z":"ce6fad20.88374","name":"MODBUS request","showStatusActivities":true,"showErrors":true,"server":"f52e4b7c.265c78","x":710,"y":740,"wires":[[],[]]},{"id":"f52e4b7c.265c78","type":"modbus-client","z":"","name":"USB converter","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU-BUFFERD","serialBaudrate":"19200","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"500","unit_id":"","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

Are you sure for the baud rate 19200 and not 9600?
Is it an ASCII or RTU device?

mascapollo, 2017-05-26:
Hello, I've tried both buffered and not buffered. My conection is 19200 and is RTU. 

I've tested other serial nodes and it works.  Both of these test work with 19200

Test 1 -  Serial node --- usb device ----- wire ----- usb device ---- windows with Putty

Test 2 -   other MODBUS serial node (other contrib) ----- usb device ----- wire ---- usb device ---- windows with diagslave

biancode, 2017-05-26:
OK, now I need a [detailed log](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG) to help.

mascapollo, 2017-05-26:
Can you help me with this detailed log? those DEBUG options go in the setting.js ? and the log generated goes to the console?

mascapollo, 2017-05-26:
Finally is working. After figuring out how to use the variable DEBUG and starting node red with the debug, I could see that there were several unused nodes (those deleted) that seemed to keep doing things and maybe one of them took over the USB port so the real node didn't have a chance. Now, I'm reading values from Modbus.

Thanks

biancode, 2017-05-27:
Yes, every client node (config) try to get a connection automatically. If you bind them to the flow instead of the global "all flows"

[image]

better is binding to the flow
[image]

closed 2017-05-27 by biancode

biancode, 2017-05-27:
watch [YouTube](https://youtu.be/TO88NryAT_c) for configurations in Node-RED