BiancoRoyal/node-red-contrib-modbus #39: Dynamic configuration of the slave IP address

lcvcl, 2017-05-26:
Just as flex-getter can use msg to configure address, why not allow the slave's ip address to be configured using msg?
In the actual production, some companies do cloud platform convergence equipment need to set the ip from the station to read the information, but these IP is not necessarily static, the company may change the slave IP or add a new from Station, if it is written in HTML config, each time you need to change the configuration once again ip address, this is a very troublesome thing. But if the node can read information from the file, and then use its IP and other information for dynamic configuration of this is quite comfortable, then why not let IP address can be dynamically configured?
There is a problem in your project to see the existence of the client, but ask what is not the actual node?

biancode, 2017-05-26:
> There is a problem in your project to see the existence of the client, but ask what is not the actual node?

No, this is how it works in Node-RED. It is not a problem.

lcvcl, 2017-05-26:
i don'think is a problem
i think if the ip can config dynamic ,it will be better

lcvcl, 2017-05-26:
two question 
1 why the client not dispaly
2 if the ip config dynamic ,will more adapt in the actual enviroment

biancode, 2017-05-26:
If you want a dynamically TCP connection, then it has to be a very new node to handle connection on Modbus.

1) this is Node-RED
2) think about multiple requests, just to have the IP dynamically is not the end

biancode, 2017-05-26:
if you want to discuss that contact me or let's have an online session

closed 2017-05-26 by biancode

biancode, 2017-05-27:
watch [YouTube](https://youtu.be/TO88NryAT_c) for configuration

biancode, 2017-07-02:
try v1.2.0