BiancoRoyal/node-red-contrib-modbus #41: Is there a modbus node which will take in modbus server ip and port number from a function node?

Kartheek1512, 2017-06-06:
I am presently using modbus flex getter and i am sending unit id, quantity from the function node. Is there a way to send Server address (Ip) and port (502) too from the function node?

biancode, 2017-06-06:
No.

biancode, 2017-06-06:
It is possible, but it is a lot of work, and there is more to think about as "give a dynamic IP and Port" for me. So it needs a clear Use Case and connection handling. Think about: you have to open and close for every request, and you have to wait for every answer. What if a connection fails? Error - try again, how often to try again. It easy to setup IP and Port by a function node, but the schmaltz is in handling connections. What if your device will close for seconds after 100 times of open and close in msec.?

biancode, 2017-06-06:
You can contact me directly if you need that feature to talk about how to realize it in a right way.

biancode, 2017-06-18:
We will see ...

icarogregorio, 2017-06-26:
I found a way to set up IP address by downloading a page and change the configuration page IP address, but that is not the correct way and takes a lot of time to process the information. Anyway, I am posting the flow maybe help anyone until the final version of this issue has been completed.

Example flow:

`[{"id":"8da5509d.95e5b","type":"tab","label":"Example"},{"id":"7de9cc28.ddc1d4","type":"inject","z":"8da5509d.95e5b","name":"","topic":"dd43f20a.97c81","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":178.75,"y":133.75,"wires":[["ec154b7d.ee9c48"]]},{"id":"ec154b7d.ee9c48","type":"http request","z":"8da5509d.95e5b","name":"Load Modbus Configuration","method":"GET","ret":"obj","url":"http://localhost:1880/flow/{{topic}}","tls":"","x":453.25,"y":133.75,"wires":[["8382063a.354418"]]},{"id":"8382063a.354418","type":"function","z":"8da5509d.95e5b","name":"Edit Configuration","func":"var fn = msg.payload.configs;\nvar sn = '28b9961b.8f24ba';\nmsg.headers = {\n    'Content-Type': 'application/json',\n    'Accept' : 'application/json'};\nmsg.url = \"http://localhost:1880/flow/\" + msg.topic;\nmsg.method =\"PUT\";\nfor(var i=0;i<fn.length;i++) {\n    if (fn[i].id == sn) {\n        fn[i].tcpPort = \"\";\n        \n        return msg;\n    }\n        \n}\n\n\nreturn msg;","outputs":1,"noerr":0,"x":697.75,"y":133.75,"wires":[["c750d1a1.002db"]]},{"id":"c750d1a1.002db","type":"http request","z":"8da5509d.95e5b","name":"Save configuration","method":"use","ret":"obj","url":"","tls":"","x":913.75,"y":133.75,"wires":[[]]}]`

In this example, I use the 'dd43f20a.97c81' serial number in the inject node, but you have to replace this number with a serial number of the flow where is a Modbus node. You can do this by clicking on the node where is a Modbus node and seeing on the browser's address bar after the '/#flow/' word.  In the edit configuration function node, I use the '28b9961b.8f24ba' serial number, for discover the configuration's serial number of your node you have to put a debug node after the "Load Modbus Configuration" Node and link with it. Then on the debug tab go to "Object > configs > [position of configuration with type modbus-client]" and copy the id parameter replacing the '28b9961b.8f24ba' serial number with your serial number.

With this flow, you can modify the config page using the following parameters just by typing the fn[i] array describled on the line 10 in "Edit configuration" function node:

tcpHost: "172.168.31.70"
tcpPort: "502"
tcpType: "DEFAULT"
serialPort: "/dev/ttyUSB"
serialType: "RTU-BUFFERD"
serialBaudrate: "9600"
serialDatabits: "8"
serialStopbits: "1"
serialParity: "none"
serialConnectionDelay: "100"
unit_id: ""
commandDelay: 1
clientTimeout: 1000
reconnectTimeout: 1000

This is an alternative way, but I enforce that is not the correct way! If you want performance settings like me, wait for this issue is complete.

@biancode, if I am wrong please remove this comment for me!

Thank you!

biancode, 2017-06-26:
OK, new idea - is it enough to use the dynamic given IP just if the connection is lost? That would be easy to handle.

icarogregorio, 2017-06-27:
Sorry, I don't undertood what did you mean!

Thanks!

biancode, 2017-06-27:
I did not search for a solution, how to change the IP address. That point is straightforward. The bigger question is, when should the connector use a new IP? I got no real Use Case for now, which describes a complete workflow about why and how often the IP will change. Is the connection broken if the connector gets a new IP or does it still working at that point and has to restart TCP connection? It would be a horrible solution to reconnect on every request.

icarogregorio, 2017-06-27:
Right, I understood! Is there a way for we talk instantly?
here is my skype name: icarogregorio
Please add me for we talk about an application that I am developing.

biancode, 2017-06-27:
OK, let's try on Thursday 08:00 pm

icarogregorio, 2017-06-27:
Right! I am in Brazil GMT -03:00
What's your location for world time?

biancode, 2017-06-27:
My local time is MESZ Germany

icarogregorio, 2017-06-29:
Our meeting in this time is ok?

biancode, 2017-07-02:
from v1.2.0 there is a new Modbus Flex Connector node - give it a try and let me know how it works for you!

`[{"id":"8960b1f1.739d5","type":"modbus-flex-connector","z":"37166668.ca483a","name":"","maxReconnectsPerMinute":4,"emptyQueue":false,"showStatusActivities":false,"showErrors":false,"server":"1f0aa890.fcb697","x":551.5,"y":485,"wires":[]},{"id":"659c8a87.3dcd04","type":"inject","z":"37166668.ca483a","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":172.5,"y":439,"wires":[["b59e3409.99d3e8"]]},{"id":"b59e3409.99d3e8","type":"function","z":"37166668.ca483a","name":"","func":"msg = {};\nmsg.payload = { 'connectorType': 'TCP', 'tcpHost': '192.168.1.18', 'unitId': 2 };\nreturn msg;","outputs":1,"noerr":0,"x":317.5,"y":439,"wires":[["8960b1f1.739d5","1f23f978.89c257"]]},{"id":"1f23f978.89c257","type":"debug","z":"37166668.ca483a","name":"","active":true,"console":"false","complete":"false","x":516.5,"y":443,"wires":[]},{"id":"f6da1d64.e251a","type":"inject","z":"37166668.ca483a","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":168,"y":487,"wires":[["a4c1ce88.c2e8"]]},{"id":"a4c1ce88.c2e8","type":"function","z":"37166668.ca483a","name":"","func":"msg.payload = { 'connectorType': 'TCP', 'tcpHost': 'localhost', 'unitId': 1 };\nreturn msg;","outputs":1,"noerr":0,"x":315,"y":486,"wires":[["8960b1f1.739d5","1f23f978.89c257"]]},{"id":"1f0aa890.fcb697","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

closed 2017-07-02 by biancode