BiancoRoyal/node-red-contrib-modbus #42: 16 bits to float register

Proone01, 2017-06-09:


biancode, 2017-06-10:
Is that [Wiki](https://github.com/biancode/node-red-contrib-modbus/wiki/Buffer-and-endianness---32Bit---64Bit) article helpful?

biancode, 2017-06-12:
reopen if you need more

closed 2017-06-12 by biancode

Proone01, 2017-06-13:
Your script only swap the 16 bits words. I already did it. I've 2 unsigned int (16bits each) and i need a 32 bits float.

Exemple : Register 256 : 16864 => unsigned 16 bits integer    | step 1
                Register 257 : 49552 => unsigned 16 bits integer    | step 1
                Result : 28.09451 => 32 bits float                              | step 2
*thoses values comes from ModbusDoctor.

I need to know what to do between step 1 and 2.

Any ideas ?

Thk

ggarza78, 2017-09-11:
I have the same issue, did you solve it?

Kartheek1512, 2017-10-18:
Use the Modbus flex getter. it outputs 16 bit unsigned by default. To converted to 32 bit array float, put a function node and convert using the code below.

var ui16 = new Uint16Array(msg.payload);
var fl32 = new Float32Array(ui16.buffer, ui16.byteOffset, ui16.byteLength / Float32Array.BYTES_PER_ELEMENT);

biancode, 2017-10-19:
see [Wiki](https://github.com/biancode/node-red-contrib-modbus/wiki/Buffer-8Bit,-16Bit,-32Bit,-64Bit,-128Bit-...-reading-with-SDM630-power-meter)

biancode, 2017-10-19:
feel free to [support us](https://bianco-royal.cloud/supporter/) for helping you

ciaowecia81, 2018-01-08:
Hi I try to use the function as describe below, but I can't perform the conversion.
I have an otuput from the modbus reader in Float - Big Endian (ABCD).
I need to convert it in float value.
For example: 43 66 00 00 is the read bytes - float 230 must be the final result

Can you help me please ?
Many Thanks 

This is my flow .
The connection is a real connection

[{"id":"dbc2b99b.06499","type":"modbus-read","z":"7a43f040.8108a8","name":"Automatic Read FC3","showStatusActivities":false,"showErrors":false,"unitid":"1","dataType":"HoldingRegister","adr":"44","quantity":"2","rate":"2","rateUnit":"s","server":"dbfe4a2c.827ec8","x":170,"y":100,"wires":[["586f0484.e8809c"],["1d9ea62.ca3485a","d65689d9.a73bd8"]]},{"id":"1d9ea62.ca3485a","type":"modbus-response","z":"7a43f040.8108a8","name":"","registerShowMax":"20","x":470,"y":140,"wires":[]},{"id":"586f0484.e8809c","type":"debug","z":"7a43f040.8108a8","name":"","active":false,"console":"false","complete":"false","x":450,"y":60,"wires":[]},{"id":"d65689d9.a73bd8","type":"debug","z":"7a43f040.8108a8","name":"","active":false,"console":"false","complete":"true","x":430,"y":100,"wires":[]},{"id":"dbfe4a2c.827ec8","type":"modbus-client","z":"","name":"MODBUS TCP RTU nwe OFFICE","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"jassoffice.dyndns.org","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

apanasara, 2018-04-29:
@Kartheek1512 

var ui16 = new Uint16Array(msg.payload);
var fl32 = new Float32Array(ui16.buffer, ui16.byteOffset, ui16.byteLength / Float32Array.BYTES_PER_ELEMENT);
msg.payload=fl32;
return fl32;

This works for <modbus reader> node, but not working for <modbus flex getter>...(Type error)

apanasara, 2018-04-29:
@Kartheek1512 

var ui16 = new Uint16Array(msg.payload);
var fl32 = new Float32Array(ui16.buffer, ui16.byteOffset, ui16.byteLength / Float32Array.BYTES_PER_ELEMENT);
msg.payload=fl32;
return msg;

This works for <modbus reader> node, but not working for <modbus flex getter>...(Type error)