BiancoRoyal/node-red-contrib-modbus #43: Help on Reading multiple non-contiguous modbus registers on same device

mascapollo, 2017-06-12:
What is the best way to read multiple strips (non contiguous) of modbus registers? 

I've done something like the capture attached and it works. But I'm not sure if this is the best way if you want to scale. Sending multiple requests to the same node make it difficult to keep a record of messages received.

[image]

biancode, 2017-06-12:
Did you try the [Flex Getter](https://youtu.be/G4p-Af1olo8?list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH)?

mascapollo, 2017-06-12:
That nodes are flex getter. My point is not sending multiple messages, but to arrange all the registers read from a particular device in an array so I can analyze them later. If I send multiple requests, how can I validate which messages are going to be joined in an array and in which order and messing up different requests for the same device?

mascapollo, 2017-06-12:
May be this is a request for an enhancement. A node that you can pass in one request all the registers needed.  for example: read register 100, 20 count, register 200, 15 count, etc. and getting one array with those readings in the order received.

biancode, 2017-06-12:
If you think about, how to program such a node, then you found the solution and it is possible to build it in a flow with node-red, maybe.

mascapollo, 2017-06-12:
I don't know what detail level you need, but I think of this:

along with the fc and unitid parameters, one would pass an array with [reg#1,qty#1,reg#2,qty#2,....,reg#n,qty#n]

var allValuesArray = []
var loopValuesArray =[]
validate input parameter array has even number of values
loop from x=1 to x=(inputArray.length)/2
read from modbus unitid regx,qtyx and place result in loopValuesArray[]
allValuesArray = allValuesArray.contact(loopValuesArray)
loopValuesArray = []
delay z msec
loop end
return allValuesArray and other parameters

mascapollo, 2017-06-12:
sorry I meant:
allValuesArray = allValuesArray._concat_(loopValuesArray)

closed 2017-09-23 by biancode