BiancoRoyal/node-red-contrib-modbus #46: Can't read over 127 holding registers

kevin100702, 2017-06-21:
Sorry！ My english is pool.

21 Jun 12:48:39 - TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1044:11)
    at Buffer.writeUInt8 (buffer.js:1092:5)
    at Object.<anonymous> (C:\Users\Kevin\.node-red\node_modules\node-red-contrib-modbus\node_modules\node-modbus\node-modbus\handler\server\ReadHoldingRegisters.js:58:14)
    at ontimeout (timers.js:380:14)
    at tryOnTimeout (timers.js:244:5)
    at Timer.listOnTimeout (timers.js:214:5)

biancode, 2017-06-21:
Please, post your flow to read or try translate.google.com! ;)

kevin100702, 2017-06-22:
I changed the  Quantity of [Automatic Read FC3] node in the example to 128.

This my flow.  

`[{"id":"fac3ce5e.ab6dc","type":"modbus-read","z":"ea701a17.43ed38","name":"Automatic Read FC3","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"HoldingRegister","adr":"200","quantity":"128","rate":"1","rateUnit":"s","server":"24f2a88d.c6acd8","x":265,"y":128,"wires":[["e209e1a3.a8351"],["2b8c015f.aafa1e","68632849.14f568","f99a6a7b.f67478"]]},{"id":"24f2a88d.c6acd8","type":"modbus-client","z":"","name":"Local Write VServer","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000}]`

Thank you！

kevin100702, 2017-06-22:
When I use the raspberry to read my HMI device when there is a similar problem, when I read the number is greater than 127, node output the number of the value will be wrong.
When I set the number to 128, the number of output values ​​is 0. When set to 130, the number of output values ​​is 2.
`[{"id":"20b5ffca.4eb1","type":"modbus-read","z":"40cb00ad.b82bb","name":"","showStatusActivities":false,"showErrors":false,"unitid":"1","dataType":"HoldingRegister","adr":"1101","quantity":"130","rate":"5","rateUnit":"s","server":"94460b98.86a5e8","x":127.10000610351562,"y":149,"wires":[["e83d2013.827d6"],[]]},{"id":"94460b98.86a5e8","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.11.80","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000}]`

Thank you very much.

JesseTheSE, 2017-08-30:
Quite simply, the Modbus protocol itself doesn't allow for reading more than 125 registers in a single function code 3 request.

In a Modbus request, only 1 byte is allocated to specify the number of Modbus registers requested, which means we don't have a way to request more than 127. A modbus message is limited to 256 bytes, which, when accounting for the 5 bytes that comprise the header and CRC , leaves us with 251 bytes for register values, or 125 2-byte register values.

closed 2017-09-03 by biancode

joshuatreenl, 2018-12-26:
Although this item is closed. The solution is plausible but I think the modbus code should be written in such a manner that the request will not quit node red... Suppose someone would love to write a 'virus' to quit all the modbus devices with this code than it's an easy job. I'll hope you could consider to change the code.