BiancoRoyal/node-red-contrib-modbus #47: http request msg parameters (msg.res and msg.req) do not pass through

bsense-rius, 2017-06-21:
The modbus writer does not pass through msg's with the .res and .req defined by the http request node.

I have been debugging your code. I've seen that the queueLOG performs a JSON.stringify over the overall msg object, ending in a Circular error. Despite I have overriden it (by modifying modbus-client.js calls to queueLOG to just stringify msg.payload) I have found another error I don't know how to correct. When the writing has been done (correctly) and the modbus-write,js is about to propagate the new built message to the next node:
node.send(buildMessage(msg.payload, resp, msg))
I get an awkward error "Cannot set property listening of #<Server> which has only a getter"

It is a pitty because not letting this through limits the capability to integrate this good node in webservices.

Thanks

biancode, 2017-06-21:
That sounds as your msg object is corrupt by references. Please post your input example to check.

bsense-rius, 2017-06-23:
Please find hereinafter attached a couple of flows about the issue I mentioned. I've seen in the source code the reasons for the lack keeping the msg. properties set by other previous nodes connected to the same flow. In fact I have patched them to properly work. I am able to run the following samples with minor changes in modbus-client.js modbus-getter.js and modbus-write.js. For example modbus-getter deifnes a new JSON structure for the msg object, overriding the previously defined properties of it. This is the reason why modbus-getter is not propagating the much needed .req and .res set by http in to finally prepare the http response. The circular refernce in JSON is caused by the queueLog function that tries to stringify the complete msg obj. In the case of modbus-write the object is not defined as new (good) but you know that some complex properties are not stringify friendly (a recurrent topic in node.js). I just overrid it by logging msg.payload and not the entire msg. I could send the patched files, but I guess that you perfectly got the point.

Thanks!!

[{"id":"20905624.e3232a","type":"http in","z":"43d3aef7.3752","name":"","url":"/mbread","method":"get","swaggerDoc":"","x":110,"y":2700,"wires":[["2c572382.60a92c"]]},{"id":"2c572382.60a92c","type":"modbus-getter","z":"43d3aef7.3752","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"HoldingRegister","adr":"100","quantity":"1","server":"6c8f5125.f43f","x":520,"y":2700,"wires":[["4f1f9658.9a9e78"],[]]},{"id":"4f1f9658.9a9e78","type":"http response","z":"43d3aef7.3752","name":"","x":690,"y":2700,"wires":[]},{"id":"53c45dc0.d21534","type":"http in","z":"43d3aef7.3752","name":"","url":"/mbwrite","method":"get","swaggerDoc":"","x":110,"y":2780,"wires":[["1f9f6e45.8e7a02"]]},{"id":"c20e1f2d.5ae43","type":"modbus-write","z":"43d3aef7.3752","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"HoldingRegister","adr":"100","quantity":"1","server":"6c8f5125.f43f","x":520,"y":2780,"wires":[["9025b16f.37538"],[]]},{"id":"1f9f6e45.8e7a02","type":"change","z":"43d3aef7.3752","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"123","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":310,"y":2780,"wires":[["c20e1f2d.5ae43"]]},{"id":"9025b16f.37538","type":"http response","z":"43d3aef7.3752","name":"","x":690,"y":2780,"wires":[]},{"id":"6c8f5125.f43f","type":"modbus-client","z":"","name":"modrssim","clienttype":"tcp","bufferCommands":false,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

closed 2017-09-03 by biancode