BiancoRoyal/node-red-contrib-modbus #51: Modbus flex write has no output

Kartheek1512, 2017-08-28:
Hi, 

 I am not seeing any output message in the Modbus flex wite. I am trying to control the power output of solar inverters by writing the power level onto one of its registers. Some times, the inverter takes it and some times it wouldnt. As I am not able to see the output of the node, I am not able to determine if the inverter took the write command or not. 

[{"id":"e014b77c.7f8228","type":"modbus-flex-write","z":"bad51642.a98048","name":"","showStatusActivities":false,"showErrors":false,"server":"66afd004.9dce2","x":884.8571090698242,"y":699.0067138671875,"wires":[[],["19b99901.257627"]]},{"id":"75bed00c.f9447","type":"debug","z":"bad51642.a98048","name":"","active":true,"console":"false","complete":"true","x":878.9998207092285,"y":633.2519416809082,"wires":[]},{"id":"19b99901.257627","type":"debug","z":"bad51642.a98048","name":"","active":true,"console":"false","complete":"true","x":1059.999849319458,"y":690.8212299346924,"wires":[]},{"id":"b9c69530.6d8238","type":"file in","z":"bad51642.a98048","name":"C","filename":"PPC_config/Inverter1/AP","format":"utf8","x":317.99756050109863,"y":703.8746671676636,"wires":[["c2776990.1726b8"]]},{"id":"c2776990.1726b8","type":"json","z":"bad51642.a98048","name":"","x":444.9998779296875,"y":703.3725423812866,"wires":[["b7329f4d.b25e7"]]},{"id":"b7329f4d.b25e7","type":"function","z":"bad51642.a98048","name":"","func":"var inv1sp=msg.payload;\nvar inv1= global.get(\"inv1\");\n\nif (inv1sp*10!=inv1[1]){\n   \nmsg={};\n msg.payload = {value:inv1sp*10, 'fc': 6, 'unitid': 1, 'address': 4097 , 'quantity': 1 };\n node.send(msg);\n}\n ","outputs":"1","noerr":0,"x":575.2287120819092,"y":704.5621147155762,"wires":[["e014b77c.7f8228","75bed00c.f9447","7803b8c8.dd2558"]]},{"id":"4b63f12d.b83fc","type":"inject","z":"bad51642.a98048","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":176.00001525878906,"y":633.8033895492554,"wires":[["b9c69530.6d8238"]]},{"id":"7803b8c8.dd2558","type":"delay","z":"bad51642.a98048","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":404.0022430419922,"y":653.8926763534546,"wires":[["b9c69530.6d8238"]]},{"id":"66afd004.9dce2","type":"modbus-client","z":"","name":"IP 501","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.13.125","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

biancode, 2017-08-29:
Dear Kartheek,

please log some of the values to get an idea ... are there negativ values? or is it between 0 and 65000?

Kartheek1512, 2017-08-30:
There are no negative values. I observed today and what i found was, when I write using the modbus flex write, after sending a value to a "write register", it goes to timeout. i dont see an output. 

I have an issue with read too. I have to **read** twice to make the "modbus flex read" node to active. Very surprised with the behavior.  I did not have this issue when using a Power meter. The first read will always give an error and it goes back to active after 5 seconds. When i pull again immediately, I will get values.

biancode, 2017-08-30:
`[{"id":"116847be.0d0798","type":"modbus-flex-write","z":"fd6dbae6.dfcf68","name":"","showStatusActivities":false,"showErrors":false,"server":"2811f2ae.a87dee","x":744.8571166992188,"y":142.75477600097656,"wires":[["8a472231.5398f"],["8a472231.5398f"]]},{"id":"c8d25244.38ef2","type":"debug","z":"fd6dbae6.dfcf68","name":"","active":true,"console":"false","complete":"true","x":703.9998168945312,"y":107.00000381469727,"wires":[]},{"id":"8a472231.5398f","type":"debug","z":"fd6dbae6.dfcf68","name":"","active":true,"console":"false","complete":"true","x":925.9998779296875,"y":141.5692901611328,"wires":[]},{"id":"23a60abb.11a846","type":"file in","z":"fd6dbae6.dfcf68","name":"C","filename":"PPC_config/Inverter1/AP","format":"utf8","x":280.99756813049316,"y":142.62272930145264,"wires":[["be4f52cd.7f3b8"]]},{"id":"be4f52cd.7f3b8","type":"json","z":"fd6dbae6.dfcf68","name":"","x":407.99988555908203,"y":142.12060451507568,"wires":[["46d5cfa7.b8b96"]]},{"id":"46d5cfa7.b8b96","type":"function","z":"fd6dbae6.dfcf68","name":"","func":"var inv1sp = msg.payload;\nvar globalInv1 = global.get(\"inv1\");\nvar valueToSend = inv1sp * 10;\n\nif (valueToSend !== globalInv1[1]) {\n        \n    msg = {};\n    msg.payload = { \n        value: valueToSend, \n        'fc': 6, 'unitid': 1, \n        'address': 4097 , 'quantity': 1\n    };\n    return msg;\n}\n ","outputs":"1","noerr":0,"x":538.2287197113037,"y":143.31017684936523,"wires":[["116847be.0d0798","c8d25244.38ef2","899a4241.5d638"]]},{"id":"eed6a117.dfb81","type":"inject","z":"fd6dbae6.dfcf68","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":109.0000228881836,"y":142.55145263671875,"wires":[["23a60abb.11a846"]]},{"id":"899a4241.5d638","type":"delay","z":"fd6dbae6.dfcf68","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":410.00225830078125,"y":203.64073944091797,"wires":[["23a60abb.11a846"]]},{"id":"2811f2ae.a87dee","type":"modbus-client","z":"","name":"IP 501","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.13.125","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"200","clientTimeout":"2000","reconnectTimeout":"4000"}]`

please try that

Kartheek1512, 2017-08-30:
I tried it, i am getting output from the function node but i am not getting output from the "modbus flex getter". I spoke with the solar inverter tech. He said, upon writing successfully, the inverter will acknowledge by sending back the set value.  

I am able to write to inverter after a couple of attempts but I dont see the output anytime from the "modbus flex getter" node.   

[image]

biancode, 2017-09-03:
Is it the same behavior with the Reader Node? Did you try?

biancode, 2017-09-03:
Did you try to send faster (1 sec. delay max)?

Kartheek1512, 2017-09-03:
Read node is pretty quick. But I have to send twice to get data..It works like this. 

12:00:05 - Read--error
12:00:06- Read---pass (Will get output)
12:00:07  read--pass
I will get all pass when i request with in a second. But if I idle it for a while, i will have to send twice. 
Case 2:
12:00:05- Read- error
12:00:06- Read- pass
12:00:16-Read-error
12:00:17-Read-pass
12:00:18-Read-pass
12:00:28-Read-error
12:00:29-Read-pass

closed 2017-09-23 by biancode