BiancoRoyal/node-red-contrib-modbus #52: How to troubleshoot "time out" issue?

clickworkorange, 2017-09-02:
First of all, thank you for making this collection of nodes available! Although I'm a newcomer to Modbus, I can see that it is a well thought out and flexible collection of tools, with decent documentation. Nice! But I'm afraid I'm having a little trouble getting even the most basic set-up to work, and I'm unable to figure out why - or indeed even _what exactly_ is going wrong :) At first I had trouble getting any communication at all to appear on the serial port (UART connected to a Pi2); I tried adding my "nodered" user to the "dialout" group, which is most likely necessary in order to grant it access to `/dev/ttyAMA0` where the port is located, but that didn't help. Then I looked a bit closer at the Node-RED log output and saw that I had an error on start-up: `[warn] [serialport] Error: Module did not self-register.`, and after a bit of searching I did a re-install of the `node-red-node-serialport` with `--unsafe-perm`, and immediately I could see traffic starting to go out, and the "Rx" LED on the one RTU I have started blinking. "Great" I thought, "so it's working now", but alas I am not getting any response back. I have enabled debug by putting 
```
Environment="NODE_RED_OPTIONS=-v"
Environment="DEBUG=modbus-serial,node_red_contrib_modbus"
```
in `/lib/systemd/system/nodered.service` and doing `systemctl daemon-reload` followed by `systemctl restart nodered`, and I can now see these debug messages in the syslog:  

```
Sep  2 22:06:47 system Node-RED[3946]: Sat, 02 Sep 2017 21:06:47 GMT modbus-serial { action: 'send serial rtu buffered',
Sep  2 22:06:47 system Node-RED[3946]: data: <Buffer 01 04 00 f6 00 01 d1 f8>,
Sep  2 22:06:47 system Node-RED[3946]: unitid: 1,
Sep  2 22:06:47 system Node-RED[3946]: functionCode: 4 }
Sep  2 22:06:48 system Node-RED[3946]: Sat, 02 Sep 2017 21:06:48 GMT modbus-serial {}
Sep  2 22:06:48 system Node-RED[3946]: Sat, 02 Sep 2017 21:06:48 GMT modbus-serial Timed out
```
The RTU in question is a Novus DigiRail 4C four channel counter, and it has a button "RCom" which when pressed sets the address and serial port parameters to known values (address 246, baud rate 1200, parity even, 1 stop bit) for one minute - I have used the same values in the configuration node (baud rate, parity, stop bit) and in the JSON I pass to the one `modbus-flex-getter` node I've put into my flow (full JSON is {"value":11,"fc":4,"address":246,"quantity":1}), and of course I press this button before triggering the read. Where do I go from here?

biancode, 2017-09-03:
did you install the package with `node-red-contrib-modbus --unsafe-perm --build-from-source` ?

biancode, 2017-09-03:
if yes check these:

a) is it 246 or 245?
b) is the timeout time enough?
c) is there only one to open the serial port or is it blocked by other?
d) did you search in the closed issues for ttyAMA0 QA's?

have fun
Klaus

clickworkorange, 2017-09-05:
Thanks Klaus, really appreciate your reply. Unfortunately I ran `update-nodejs-and-nodered`, despite the warnings, and now there is no sign of life at all on the serial port; before at least the "Rx" LED on the RTU would blink when I tried to send it a query, but now not even that. There are no errors in the Node-RED debug console, and no errors in the syslog:

```
Sep  5 22:53:46 system systemd[1]: Starting Node-RED graphical event wiring tool....
Sep  5 22:53:46 system systemd[1]: Started Node-RED graphical event wiring tool..
Sep  5 22:53:52 system Node-RED[14258]: 5 Sep 22:53:52 - [info]
Sep  5 22:53:52 system Node-RED[14258]: Welcome to Node-RED
Sep  5 22:53:52 system Node-RED[14258]: ===================
Sep  5 22:53:52 system Node-RED[14258]: 5 Sep 22:53:52 - [info] Node-RED version: v0.17.5
Sep  5 22:53:52 system Node-RED[14258]: 5 Sep 22:53:52 - [info] Node.js  version: v8.4.0
Sep  5 22:53:52 system Node-RED[14258]: 5 Sep 22:53:52 - [info] Linux 4.9.35-v7+ arm LE
Sep  5 22:53:56 system Node-RED[14258]: 5 Sep 22:53:56 - [info] Loading palette nodes
Sep  5 22:54:02 system Node-RED[14258]: 5 Sep 22:54:02 - [info] Dashboard version 2.4.3 started at /ui
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [warn] ------------------------------------------------------
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [warn] [sensehat] Error: Can't find Sense HAT python libraries. Run sudo apt-get install sense-hat
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [warn] ------------------------------------------------------
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [info] Settings file  : /home/nodered/.node-red/settings.js
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [info] User directory : /home/nodered/.node-red
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [info] Flows file     : /home/nodered/.node-red/flows_system.mydomain.net.json
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [info] Server now running at https://127.0.0.1:1880/
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [info] Starting flows
Sep  5 22:54:05 system Node-RED[14258]: 5 Sep 22:54:05 - [warn] [modbus-client:system] Client -> reconnect in 2000 ms Serial@/dev/ttyAMA0:1200bit/s default Unit-Id: 1
Sep  5 22:54:12 system Node-RED[14258]: 5 Sep 22:54:12 - [info] Started flows
Sep  5 22:54:17 system Node-RED[14258]: Tue, 05 Sep 2017 21:54:17 GMT modbus-serial modbus connection opened
```

Apart of course from the modbus-serial timeouts (I've also added `serialport` to the debug list since my original post): 
```
Sep  5 22:46:01 system Node-RED[1621]: Tue, 05 Sep 2017 21:46:01 GMT serialport write 8 bytes of data
Sep  5 22:46:02 system Node-RED[1621]: Tue, 05 Sep 2017 21:46:02 GMT modbus-serial {}
Sep  5 22:46:02 system Node-RED[1621]: Tue, 05 Sep 2017 21:46:02 GMT modbus-serial Timed out
```
No other software on the system uses the UART; the serial console had already been removed from the boot cmdline. I'm afraid I am completely at a loss here. The flow couldn't be simpler: 

[image]

The "inject" node sends the following JSON: 

`{"unitid":246,"fc":2,"address":0,"quantity":1}`

And yes, it is supposed to be 246, according to the DigiRail manual: 

> If the equipment has already been used and you do not know its address and/or other communication parameters, press the RCom button on your front panel. It will enter the diagnostics mode during 1 minute, assuming the following communication parameters: Address=246, Baud rate=1200, parity even, 1 stop bit.

I've checked, and triple checked, every thing I can think of - including swapping the A/B wires, just in case the standard differs between my interface and the RTU (which is marked D/~~D). Whichever way, I *did* previously see the RTU detecting traffic on the bus (by blinking its "Rx" LED), even if it ignored it, whereas now it remains dark. Something must have gone wrong when I updated Node-RED, but what?

biancode, 2017-09-06:
Please, try to work with Node.js 6 and not 8 and check your unitid is now 246 and address 0

clickworkorange, 2017-09-06:
Thank you. I'm not sure where Node 8 came from, must have been pulled in with an `apt-get dist-upgrade` since I haven't built from source or anything. In general, I always try to stay with the distro packages and the default settings. Since this is for a production server, automatic updates are enabled. Regarding address/unitid, I was under the impression that `unitid` is the Modbus RTU address (1-247) and `address` is the register I want to get/set (0-65535)? I've looked around for "other Modbus tools" but there doesn't seem to be a lot out there?

biancode, 2017-09-23:
if there is more - please reopen - did Node@6 fix it?

closed 2017-09-23 by biancode

clickworkorange, 2017-09-23:
Ah. Sorry, forgot to keep this updated - it seems the issue has to do with my RS-485 interface, a [Monarco Hat](https://www.monarco.io/), and the 4.9 Linux kernel. The Monarco devs are looking into it. And yes, this issue should probably be closed - many thanks for your help though!