BiancoRoyal/node-red-contrib-modbus #53: Error being thrown repeatedly: 'FSM Not Ready To Write'

Hugobox, 2017-09-05:
Hi! 
I have a Node Red v.0.16.2 running in a docker instance with node-red-contrib-modbus v 1.0.16
There are 2 Wago TCP modbus modules that we read the state of 30 relays every 10 seconds (with Modbus-Getter) and we rarely write a new state on some of those relays for maintenance (with Modbus Flex Write). 

The problem is that the current config can work for a few days but suddenly the Modbus Flex Write displays the "Initialize" message until I try to write a new state which displays 'FSM Not Ready To Write'. Changing the status config (ex: timeout times) and redeploying is the only way I found to reset the modbus communication back to "Active". I'm currently using a Timeout of 4 seconds, a reconnect timeout of 5 seconds and a queue delay of 100ms. Playing with these values seems to change the stability time but might be a coincidence. 

Is there anything that I can try to resolve this? Thanks for the advices!

biancode, 2017-09-05:
Hi, if it never comes back to run I need a [detailed log](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG), please.

Hugobox, 2017-09-05:
Ok will do! Is "DEBUG=node_red_contrib_modbus" an environment variable?

Hugobox, 2017-09-07:
Looking at nodejs doc, shouldn't the environment variable be NODE_DEBUG instead of DEBUG?

biancode, 2017-09-07:
a) DEBUG -> find your way ... ;)
b) version 1.2.2 is much better ... for me - give it a try

closed 2017-09-23 by biancode