BiancoRoyal/node-red-contrib-modbus #54: Cannot read modbus using node-red but able to read from Win-tech Modscan

Kartheek1512, 2017-09-12:
Hi Klaus, 

 I am trying to retrieve data from an Energy Tracking Power meter. i am able to get values using "Win Tech Modscan" application but not able to pull data using node-red. It shows active at first but when I pull, it throws an error.  Can you please check and let me know. Below is a screenshot of data from Modscan. I used the same port number, IP and Modbus registers. I am not understanding why node-red is not able to pull the data. 

[image]

[{"id":"eba8c392.626cb","type":"modbus-read","z":"a60483c2.2f813","name":"","showStatusActivities":false,"showErrors":false,"unitid":"255","dataType":"HoldingRegister","adr":"40002","quantity":"1","rate":"1","rateUnit":"s","server":"2614142b.42d80c","x":213.2142791748047,"y":465.71429443359375,"wires":[["d62b64e4.483188"],[]]},{"id":"d62b64e4.483188","type":"debug","z":"a60483c2.2f813","name":"","active":true,"console":"false","complete":"false","x":405.2142791748047,"y":455.4285888671875,"wires":[]},{"id":"2614142b.42d80c","type":"modbus-client","z":"","name":"Paul's ET","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"66.222.105.224","tcpPort":"9502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"255","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

Kartheek1512, 2017-09-12:
I am not able to get values form Modbus flex getter but getting a response when i use "modbus tcp" node connected to "modbus response"

[{"id":"8957c3a.f57c34","type":"modbustcp-read","z":"a60483c2.2f813","name":"","dataType":"HoldingRegister","adr":"40000","quantity":"4","rate":"1","rateUnit":"s","server":"36e854c5.42fa6c","x":200.4999942779541,"y":457.57147216796875,"wires":[["42d76bf3.6fbc34"]]},{"id":"42d76bf3.6fbc34","type":"modbus-response","z":"a60483c2.2f813","name":"","registerShowMax":20,"x":450.5000305175781,"y":451.2857942581177,"wires":[]},{"id":"36e854c5.42fa6c","type":"modbustcp-server","z":"","host":"66.222.105.224","port":"9502","unit_id":"255","reconnecttimeout":""}]

biancode, 2017-09-12:
[{"id":"11b96eae.77a7b1","type":"modbus-read","z":"f0d509a9.e4f638","name":"","showStatusActivities":false,"showErrors":false,"unitid":"255","dataType":"HoldingRegister","adr":"40000","quantity":"10","rate":"1","rateUnit":"s","server":"481cabf2.c3c194","x":450,"y":840,"wires":[["64bd94b3.1c13dc"],[]]},{"id":"64bd94b3.1c13dc","type":"debug","z":"f0d509a9.e4f638","name":"","active":true,"console":"false","complete":"false","x":642,"y":829.7142944335938,"wires":[]},{"id":"481cabf2.c3c194","type":"modbus-client","z":"","name":"Paul's ET","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"66.222.105.224","tcpPort":"9502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"255","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

biancode, 2017-09-12:
your second flow with the modbus-tcp is a mix of packages

biancode, 2017-09-12:
[{"id":"11b96eae.77a7b1","type":"modbus-read","z":"f0d509a9.e4f638","name":"","showStatusActivities":false,"showErrors":false,"unitid":"255","dataType":"HoldingRegister","adr":"40001","quantity":"10","rate":"1","rateUnit":"s","server":"481cabf2.c3c194","x":450,"y":840,"wires":[["64bd94b3.1c13dc"],[]]},{"id":"64bd94b3.1c13dc","type":"debug","z":"f0d509a9.e4f638","name":"","active":true,"console":"false","complete":"false","x":642,"y":829.7142944335938,"wires":[]},{"id":"481cabf2.c3c194","type":"modbus-client","z":"","name":"Paul's ET","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"66.222.105.224","tcpPort":"9502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"255","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

Kartheek1512, 2017-09-12:
It did not work either.. I downloaded a node "Modbus TCP"  which worked..

https://flows.nodered.org/node/node-red-contrib-modbustcp

closed 2017-09-23 by biancode