BiancoRoyal/node-red-contrib-modbus #56: How to read Coil status

Kartheek1512, 2017-09-23:
Hi Klaus, 

  Can you please let me know how to pull a coil status. I have a SEL grid protection relay whose breaker status is in a coil as shown in the below picture. When I pull data from address 84 and 85 using Modbus read node, i get "False, False, False...False" (8 times). Below is my code. Can you please verify if I am doing it correct. 

[image]

biancode, 2017-09-23:
... divde address by 8 and you know where you are ... address 0 with quantity 1 is Bit 0 to 7

biancode, 2017-09-23:
it should be addressed by 10 with quantity 1 and then Bit 5 and 6 is what you want

biancode, 2017-09-23:
the info before was wrong - sorry - test with that flow and see how it works

`[{"id":"2e057079.a750e","type":"modbus-server","z":"f0d509a9.e4f638","name":"","logEnabled":false,"serverPort":10502,"responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":620,"y":980,"wires":[[],[],[]]},{"id":"72dfe0f4.464cd","type":"modbus-read","z":"f0d509a9.e4f638","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"Coil","adr":"85","quantity":"2","rate":"3","rateUnit":"s","server":"dd236096.80abf","x":150,"y":1000,"wires":[["247bf5bf.2e16fa"],["241bc643.28d52a"]]},{"id":"247bf5bf.2e16fa","type":"debug","z":"f0d509a9.e4f638","name":"","active":true,"console":"false","complete":"false","x":330,"y":980,"wires":[]},{"id":"241bc643.28d52a","type":"modbus-response","z":"f0d509a9.e4f638","name":"","registerShowMax":20,"x":350,"y":1040,"wires":[]},{"id":"80b0233d.0cf9c","type":"modbus-write","z":"f0d509a9.e4f638","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"Coil","adr":"85","quantity":"1","server":"dd236096.80abf","x":300,"y":820,"wires":[[],["2197c69a.cc27da"]]},{"id":"e4847cbd.0bef1","type":"inject","z":"f0d509a9.e4f638","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"x":130,"y":780,"wires":[["80b0233d.0cf9c"]]},{"id":"8068a03e.432db","type":"inject","z":"f0d509a9.e4f638","name":"","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"x":130,"y":820,"wires":[["80b0233d.0cf9c"]]},{"id":"2197c69a.cc27da","type":"modbus-response","z":"f0d509a9.e4f638","name":"","registerShowMax":20,"x":510,"y":820,"wires":[]},{"id":"890ae498.80fe48","type":"modbus-write","z":"f0d509a9.e4f638","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"Coil","adr":"86","quantity":"1","server":"dd236096.80abf","x":300,"y":900,"wires":[[],["53fb5d94.728b14"]]},{"id":"67760da7.eb9e34","type":"inject","z":"f0d509a9.e4f638","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"x":130,"y":860,"wires":[["890ae498.80fe48"]]},{"id":"ec003a02.775418","type":"inject","z":"f0d509a9.e4f638","name":"","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"x":130,"y":900,"wires":[["890ae498.80fe48"]]},{"id":"53fb5d94.728b14","type":"modbus-response","z":"f0d509a9.e4f638","name":"","registerShowMax":20,"x":510,"y":900,"wires":[]},{"id":"dd236096.80abf","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

closed 2017-09-23 by biancode