BiancoRoyal/node-red-contrib-modbus #57: node-red-contrib-modbus wont start

radar50, 2017-09-27:
after instaling node-red-contrib-modbus the web gui of node red wont start.
DEBUG=node_red_contrib_modbus node-red -v
27 Sep 17:13:25 - [info] 

Welcome to Node-RED
===================

27 Sep 17:13:25 - [info] Node-RED version: v0.17.5
27 Sep 17:13:25 - [info] Node.js  version: v6.11.3
27 Sep 17:13:25 - [info] Linux 4.11.1-sun50iw2 arm64 LE
27 Sep 17:13:27 - [info] Loading palette nodes
27 Sep 17:13:32 - [info] Dashboard version 2.5.0 started at /ui
27 Sep 17:13:34 - [warn] ------------------------------------------------------
27 Sep 17:13:34 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
27 Sep 17:13:34 - [warn] ------------------------------------------------------
27 Sep 17:13:34 - [info] Settings file  : /root/.node-red/settings.js
27 Sep 17:13:34 - [info] User directory : /root/.node-red
27 Sep 17:13:34 - [info] Flows file     : /root/.node-red/flows_orangepipc2.json
27 Sep 17:13:34 - [info] Server now running at http://127.0.0.1:1880/
27 Sep 17:13:34 - [info] Starting flows
27 Sep 17:13:34 - [error] [modbus-read:8ca540ee.454db8] TypeError: Cannot read property 'on' of null
    at new ModbusRead (/root/.node-red/node_modules/node-red-contrib-modbus/modbus/modbus-read.js:78:17)
    at createNode (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:305:18)
    at Flow.start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:89:35)
    at start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:300:29)
    at tryCatchReject (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)
    at runContinuation1 (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)
    at Fulfilled.when (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)
    at Pending.run (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:483:13)
    at Scheduler._drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:62:19)
    at Scheduler.drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:27:9)
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickCallback (internal/process/next_tick.js:104:9)
27 Sep 17:13:34 - [info] Started flows

but the web gui http://192.168.12.188:1880 keep loading for ever.

if I uninstall the module everything works.

biancode, 2017-09-27:
The error tells you that there is no client connector configuration. 

a) Was it a fresh install or upgrade? 
b) Is it version 1.2.2? 
c) Did you wire some nodes or do you use old incompatible flows?

radar50, 2017-09-27:
fresh install  and the vnode_red_contrib_modbuersion is 1.2.2. I was tryying to uninstall all and reinstal. 
But clean instalation = rm  .node-red -R
aftre instaling node_red_contrib_modbus it started to stuck

biancode, 2017-09-27:
sounds like a conflict with other package or error in flow _flows_orangepipc2.json_

biancode, 2017-09-27:
if the modbus-read was added by Node-RED flow engine it is not possible to get null for the config of the node

biancode, 2017-09-27:
[image]

radar50, 2017-09-27:
npm remove node-red-contrib-modbus
npm remove -g node-red-contrib-modbus
rm .npm/ -R
rm .node-* -R
npm install -g npm-check-updates
node-red  
---now it works I can acces web gui

npm install -g  node-red-contrib-modbus
> serialport@4.0.7 install /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-linux-arm64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.11.3 (node-v48 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.11.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.11.3/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Linux 4.11.1-sun50iw2
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
gyp ERR! node -v v6.11.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:920:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:230:5)
node-pre-gyp ERR! System Linux 4.11.1-sun50iw2
node-pre-gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
node-pre-gyp ERR! node -v v6.11.3
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node /usr/lib/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build/Release' (1)
/usr/lib
└── (empty)

npm ERR! Linux 4.11.1-sun50iw2
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "-g" "node-red-contrib-modbus"
npm ERR! node v6.11.3
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log
npm ERR! code 1

biancode, 2017-09-27:
check: `npm -v`

try the README hints, please:
`
npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source`

radar50, 2017-09-27:
I was able to install node-red-contrib-modbus through the web gui, but than the reading mode did not have the pencil on the right side of Server == tere was no pop-up menu. And when I restarted node-red it was gagin not accesable

biancode, 2017-09-27:
try `node-red -v` or `DEBUG=* node-red -v` to see more

radar50, 2017-09-27:
npm -v
3.10.10

radar50, 2017-09-27:
~/.node-red# npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source

> serialport@4.0.7 install /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::__cxx11::string)’:
../src/serialport.cpp:90:52: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
In file included from /root/.node-gyp/6.11.3/include/node/node.h:42:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/6.11.3/include/node/v8.h:8214:14: note: declared here
 Local<Int32> Value::ToInt32() const {
              ^
../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Local<v8::Object>, std::__cxx11::string)’:
../src/serialport.cpp:102:53: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
                                                     ^
In file included from /root/.node-gyp/6.11.3/include/node/node.h:42:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/6.11.3/include/node/v8.h:8178:15: note: declared here
 Local<Number> Value::ToNumber() const {
               ^
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:174:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
     int fd = argv[1]->ToInt32()->Int32Value();
                               ^
In file included from /root/.node-gyp/6.11.3/include/node/node.h:42:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/6.11.3/include/node/v8.h:8214:14: note: declared here
 Local<Int32> Value::ToInt32() const {
              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:193:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
In file included from /root/.node-gyp/6.11.3/include/node/node.h:42:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/6.11.3/include/node/v8.h:8214:14: note: declared here
 Local<Int32> Value::ToInt32() const {
              ^
../src/serialport.cpp:217:116: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   baton->baudRate = Nan::Get(options, Nan::New<v8::String>("baudRate").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                                    ^
In file included from /root/.node-gyp/6.11.3/include/node/node.h:42:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/6.11.3/include/node/v8.h:8214:14: note: declared here
 Local<Int32> Value::ToInt32() const {
              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Write(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:252:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
In file included from /root/.node-gyp/6.11.3/include/node/node.h:42:0,
                 from ../../nan/nan.h:49,
                 from ../src/./serialport.h:7,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/6.11.3/include/node/v8.h:8214:14: note: declared here
[... 222 lines cut ...]
  │ │   │ ├── inherits@2.0.3 
  │ │   │ └─┬ minimatch@3.0.3 
  │ │   │   └─┬ brace-expansion@1.1.6 
  │ │   │     ├── balanced-match@0.4.2 
  │ │   │     └── concat-map@0.0.1 
  │ │   ├─┬ once@1.3.3 
  │ │   │ └── wrappy@1.0.2 
  │ │   ├─┬ readable-stream@2.1.5 
  │ │   │ ├── buffer-shims@1.0.0 
  │ │   │ ├── core-util-is@1.0.2 
  │ │   │ ├── inherits@2.0.3 
  │ │   │ ├── isarray@1.0.0 
  │ │   │ ├── process-nextick-args@1.0.7 
  │ │   │ ├── string_decoder@0.10.31 
  │ │   │ └── util-deprecate@1.0.2 
  │ │   └── uid-number@0.0.6 
  │ └─┬ object.assign@4.0.4 
  │   ├─┬ define-properties@1.1.2 
  │   │ └── foreach@2.0.5 
  │   ├── function-bind@1.1.1 
  │   └── object-keys@1.0.11 
  ├─┬ stampit@2.1.2 
  │ ├── lodash@3.10.1 
  │ └── supermixer@1.0.3 
  └── stately.js@1.3.0

biancode, 2017-09-27:
but now it is installed in global and Node-RED, right? Be sure to have only one (global now) installed.

radar50, 2017-09-27:
DEBUG=* node-red -v &> debug.txt

biancode, 2017-09-27:
next is :

`npm install -g node-red --unsafe-perm --build-from-source`

radar50, 2017-09-27:
to be sure I run
npm remove node-red-contrib-modbusls
to uninstal potential local coy
but still nothing

Triing to attache  DEBUG=* node-red -v &> debug.txt but I§ not sure if it is atached to the message

radar50, 2017-09-27:
~/.node-red# npm install -g node-red --unsafe-perm --build-from-source
npm WARN deprecated i18next-client@1.10.3: you can use npm install i18next from version 2.0.0
npm WARN deprecated nodemailer@1.11.0: All versions below 4.0.1 of Nodemailer are deprecated. See https://nodemailer.com/status/
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
/usr/bin/node-red -> /usr/lib/node_modules/node-red/red.js
/usr/bin/node-red-pi -> /usr/lib/node_modules/node-red/bin/node-red-pi
/usr/lib
└── node-red@0.17.5 

root@orangepipc2:~/.node-red# DEBUG=* node-red -v 
  oauth2orize register exchanger password password +0ms
  express:application set "x-powered-by" to true +0ms
  express:application set "etag" to 'weak' +18ms
  express:application set "etag fn" to [Function: wetag] +9ms
  express:application set "env" to 'development' +3ms
  express:application set "query parser" to 'extended' +2ms
  express:application set "query parser fn" to [Function: parseExtendedQueryString] +2ms
  express:application set "subdomain offset" to 2 +2ms
  express:application set "trust proxy" to false +3ms
  express:application set "trust proxy fn" to [Function: trustNone] +3ms
  express:application booting in development mode +1ms
  express:application set "view" to [Function: View] +2ms
  express:application set "views" to '/root/.node-red/views' +1ms
  express:application set "jsonp callback name" to 'callback' +1ms
  express:application set "x-powered-by" to true +82ms
  express:application set "etag" to 'weak' +1ms
  express:application set "etag fn" to [Function: wetag] +1ms
  express:application set "env" to 'development' +1ms
  express:application set "query parser" to 'extended' +0ms
  express:application set "query parser fn" to [Function: parseExtendedQueryString] +1ms
  express:application set "subdomain offset" to 2 +0ms
  express:application set "trust proxy" to false +5ms
  express:application set "trust proxy fn" to [Function: trustNone] +2ms
  express:application booting in development mode +1ms
  express:application set "view" to [Function: View] +2ms
  express:application set "views" to '/root/.node-red/views' +1ms
  express:application set "jsonp callback name" to 'callback' +1ms
  express:application set "x-powered-by" to true +16ms
  express:application set "etag" to 'weak' +0ms
  express:application set "etag fn" to [Function: wetag] +1ms
  express:application set "env" to 'development' +0ms
  express:application set "query parser" to 'extended' +1ms
  express:application set "query parser fn" to [Function: parseExtendedQueryString] +0ms
  express:application set "subdomain offset" to 2 +1ms
  express:application set "trust proxy" to false +0ms
  express:application set "trust proxy fn" to [Function: trustNone] +1ms
  express:application booting in development mode +0ms
  express:application set "view" to [Function: View] +1ms
  express:application set "views" to '/root/.node-red/views' +0ms
  express:application set "jsonp callback name" to 'callback' +1ms
  express:application set "x-powered-by" to true +189ms
  express:application set "etag" to 'weak' +2ms
  express:application set "etag fn" to [Function: wetag] +0ms
  express:application set "env" to 'development' +1ms
  express:application set "query parser" to 'extended' +0ms
  express:application set "query parser fn" to [Function: parseExtendedQueryString] +1ms
  express:application set "subdomain offset" to 2 +0ms
  express:application set "trust proxy" to false +1ms
  express:application set "trust proxy fn" to [Function: trustNone] +0ms
  express:application booting in development mode +1ms
  express:application set "view" to [Function: View] +0ms
  express:application set "views" to '/root/.node-red/views' +2ms
  express:application set "jsonp callback name" to 'callback' +1ms
  express:router use '/' query +7ms
  express:router:layer new '/' +9ms
  express:router use '/' expressInit +10ms
  express:router:layer new '/' +1ms
  express:router:route new '/' +3ms
  express:router:layer new '/' +2ms
  express:router:route get '/' +3ms
[... 268 lines cut ...]
  express:router:layer new '/' +1ms
  express:router:route new '/twitter-credentials/:id/auth' +4s
  express:router:layer new '/twitter-credentials/:id/auth' +0ms
  express:router:route get '/twitter-credentials/:id/auth' +1ms
  express:router:layer new '/' +1ms
  express:router:route new '/twitter-credentials/:id/auth/callback' +1ms
  express:router:layer new '/twitter-credentials/:id/auth/callback' +0ms
  express:router:route get '/twitter-credentials/:id/auth/callback' +2ms
  express:router:layer new '/' +1ms
  express:router:route new '/serialports' +55ms
  express:router:layer new '/serialports' +1ms
  express:router:route get '/serialports' +1ms
  express:router:layer new '/' +1ms
  express:router:route get '/serialports' +0ms
  express:router:layer new '/' +1ms
27 Sep 20:16:48 - [warn] ------------------------------------------------------
27 Sep 20:16:48 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
27 Sep 20:16:48 - [warn] ------------------------------------------------------
27 Sep 20:16:48 - [info] Settings file  : /root/.node-red/settings.js
27 Sep 20:16:48 - [info] User directory : /root/.node-red
27 Sep 20:16:48 - [info] Flows file     : /root/.node-red/flows_orangepipc2.json
27 Sep 20:16:48 - [info] Creating new flow file
27 Sep 20:16:48 - [info] Starting flows
27 Sep 20:16:48 - [info] Started flows
27 Sep 20:16:48 - [info] Server now running at http://127.0.0.1:1880/

radar50, 2017-09-27:
but still no gui ...

biancode, 2017-09-27:
looks good - now it should work 

next you cloud try :

npm install -g node-red --unsafe-perm --build-from-source

I'm sorry no other idea for now ...

radar50, 2017-09-27:
[image]

biancode, 2017-09-27:
and if you uninstall or remove Modbus package it works well?

biancode, 2017-09-27:
you could try:

`npm install -g node-red-contrib-modbus@2.0.1-beta.1 --unsafe-perm --build-from-source`

radar50, 2017-09-27:
It is realy stange if I do uninstall it keep saing that it is there ...
npm remove node-red-contrib-modbusls
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.
root@orangepipc2:~/.node-red# npm remove -g node-red-contrib-modbusls
root@orangepipc2:~/.node-red#
~/.node-red# node-red
27 Sep 20:27:34 - [info] 

Welcome to Node-RED
===================

27 Sep 20:27:34 - [info] Node-RED version: v0.17.5
27 Sep 20:27:34 - [info] Node.js  version: v6.11.3
27 Sep 20:27:34 - [info] Linux 4.11.1-sun50iw2 arm64 LE
27 Sep 20:27:36 - [info] Loading palette nodes
27 Sep 20:27:42 - [warn] ------------------------------------------------------
27 Sep 20:27:42 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
27 Sep 20:27:42 - [warn] ------------------------------------------------------
27 Sep 20:27:42 - [info] Settings file  : /root/.node-red/settings.js
27 Sep 20:27:42 - [info] User directory : /root/.node-red
27 Sep 20:27:42 - [info] Flows file     : /root/.node-red/flows_orangepipc2.json
27 Sep 20:27:42 - [info] Creating new flow file
27 Sep 20:27:42 - [info] Starting flows
27 Sep 20:27:42 - [info] Started flows
27 Sep 20:27:42 - [info] Server now running at http://127.0.0.1:1880/

radar50, 2017-09-27:
[image]

biancode, 2017-09-27:
use `find` to search for and delete it or try to disable all

`npm uninstall -g node-red-contrib-modbus`

radar50, 2017-09-27:
kliking on disable in web gui gives mi after restatr of node-red working acces
[image]

radar50, 2017-09-27:
but no modbus :-(

biancode, 2017-09-27:
yes, you have to remove it from inside node-red and install it as we did global with the build from source and then it should work

biancode, 2017-09-27:
search inside node-red folder node_modules/

biancode, 2017-09-27:
you will find a way ;)

radar50, 2017-09-27:
~/.node-red# ls -al /root/.node-red/node_modules/
total 8
drwxr-xr-x 2 root root 4096 Sep 27 20:35 .
drwxr-xr-x 4 root root 4096 Sep 27 19:59 ..

radar50, 2017-09-27:
but in /usr/lib/node_modules/node-red-contrib-modbus are some files so I delete them

radar50, 2017-09-27:
deleting /usr/lib/node_modules/node-red-contrib-modbus  gives me web gui acces
npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source
-- again it gets stuck

radar50, 2017-09-27:
npm remove -g node-red-contrib-modbusls  ---does not remove the files have to do:
rm -R /usr/lib/node_modules/node-red-contrib-modbus

radar50, 2017-09-27:
~/.node-red# npm install -g node-red-contrib-modbus@2.0.1-beta.1 --unsafe-perm --build-from-source

> serialport@5.0.0 install /usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/usr/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
/usr/lib
└─┬ node-red-contrib-modbus@2.0.1-beta.1 
  ├─┬ debug@3.1.0 
  │ └── ms@2.0.0 
  ├── modbus-serial@6.0.0 
  ├─┬ node-modbus@4.0.1 
  │ ├── bluebird@3.5.0 
  │ ├── crc@3.5.0 
  │ ├── net@1.0.2 
  │ ├── stampit-event-bus@0.1.1 
  │ ├── stampit-log@0.3.0 
  │ └── stampit-state-machine@0.2.1 
  ├─┬ serialport@5.0.0 
  │ ├── bindings@1.3.0 
  │ ├── commander@2.11.0 
  │ ├─┬ debug@2.6.8 
  │ │ └── ms@2.0.0 
  │ ├── nan@2.7.0 
  │ ├─┬ node-pre-gyp@0.6.36 
  │ │ ├─┬ mkdirp@0.5.1 
  │ │ │ └── minimist@0.0.8 
  │ │ ├─┬ nopt@4.0.1 
  │ │ │ ├── abbrev@1.1.0 
  │ │ │ └─┬ osenv@0.1.4 
  │ │ │   ├── os-homedir@1.0.2 
  │ │ │   └── os-tmpdir@1.0.2 
  │ │ ├─┬ npmlog@4.1.2 
  │ │ │ ├─┬ are-we-there-yet@1.1.4 
  │ │ │ │ └── delegates@1.0.0 
  │ │ │ ├── console-control-strings@1.1.0 
  │ │ │ ├─┬ gauge@2.7.4 
  │ │ │ │ ├── aproba@1.1.2 
  │ │ │ │ ├── has-unicode@2.0.1 
  │ │ │ │ ├── object-assign@4.1.1 
  │ │ │ │ ├── signal-exit@3.0.2 
  │ │ │ │ ├─┬ string-width@1.0.2 
  │ │ │ │ │ ├── code-point-at@1.1.0 
  │ │ │ │ │ └─┬ is-fullwidth-code-point@1.0.0 
  │ │ │ │ │   └── number-is-nan@1.0.1 
  │ │ │ │ ├─┬ strip-ansi@3.0.1 
  │ │ │ │ │ └── ansi-regex@2.1.1 
  │ │ │ │ └── wide-align@1.1.2 
  │ │ │ └── set-blocking@2.0.0 
  │ │ ├─┬ rc@1.2.1 
  │ │ │ ├── deep-extend@0.4.2 
  │ │ │ ├── ini@1.3.4 
  │ │ │ ├── minimist@1.2.0 
  │ │ │ └── strip-json-comments@2.0.1 
  │ │ ├─┬ request@2.81.0 
  │ │ │ ├── aws-sign2@0.6.0 
  │ │ │ ├── aws4@1.6.0 
  │ │ │ ├── caseless@0.12.0 
  │ │ │ ├─┬ combined-stream@1.0.5 
  │ │ │ │ └── delayed-stream@1.0.0 
  │ │ │ ├── extend@3.0.1 
  │ │ │ ├── forever-agent@0.6.1 
  │ │ │ ├─┬ form-data@2.1.4 
  │ │ │ │ └── asynckit@0.4.0 
  │ │ │ ├─┬ har-validator@4.2.1 
  │ │ │ │ ├─┬ ajv@4.11.8 
  │ │ │ │ │ ├── co@4.6.0 
  │ │ │ │ │ └─┬ json-stable-stringify@1.0.1 
  │ │ │ │ │   └── jsonify@0.0.0 
  │ │ │ │ └── har-schema@1.0.5 
  │ │ │ ├─┬ hawk@3.1.3 
  │ │ │ │ ├── boom@2.10.1 
  │ │ │ │ ├── cryptiles@2.0.5 
  │ │ │ │ ├── hoek@2.16.3 
  │ │ │ │ └── sntp@1.0.9 
  │ │ │ ├─┬ http-signature@1.1.1 
  │ │ │ │ ├── assert-plus@0.2.0 
  │ │ │ │ ├─┬ jsprim@1.4.0 
  │ │ │ │ │ ├── assert-plus@1.0.0 
  │ │ │ │ │ ├── extsprintf@1.0.2 
  │ │ │ │ │ ├── json-schema@0.2.3 
  │ │ │ │ │ └── verror@1.3.6 
  │ │ │ │ └─┬ sshpk@1.13.1 
  │ │ │ │   ├── asn1@0.2.3 
  │ │ │ │   ├── assert-plus@1.0.0 
  │ │ │ │   ├── bcrypt-pbkdf@1.0.1 
  │ │ │ │   ├─┬ dashdash@1.14.1 
  │ │ │ │   │ └── assert-plus@1.0.0 
  │ │ │ │   ├── ecc-jsbn@0.1.1 
  │ │ │ │   ├─┬ getpass@0.1.7 
  │ │ │ │   │ └── assert-plus@1.0.0 
  │ │ │ │   ├── jsbn@0.1.1 
  │ │ │ │   └── tweetnacl@0.14.5 
  │ │ │ ├── is-typedarray@1.0.0 
  │ │ │ ├── isstream@0.1.2 
  │ │ │ ├── json-stringify-safe@5.0.1 
  │ │ │ ├─┬ mime-types@2.1.16 
  │ │ │ │ └── mime-db@1.29.0 
  │ │ │ ├── oauth-sign@0.8.2 
  │ │ │ ├── performance-now@0.2.0 
  │ │ │ ├── qs@6.4.0 
  │ │ │ ├── stringstream@0.0.5 
  │ │ │ ├─┬ tough-cookie@2.3.2 
  │ │ │ │ └── punycode@1.4.1 
  │ │ │ ├── tunnel-agent@0.6.0 
  │ │ │ └── uuid@3.1.0 
  │ │ ├─┬ rimraf@2.6.1 
  │ │ │ └─┬ glob@7.1.2 
  │ │ │   ├── fs.realpath@1.0.0 
  │ │ │   ├── inflight@1.0.6 
  │ │ │   ├─┬ minimatch@3.0.4 
  │ │ │   │ └─┬ brace-expansion@1.1.8 
  │ │ │   │   ├── balanced-match@1.0.0 
  │ │ │   │   └── concat-map@0.0.1 
  │ │ │   └── path-is-absolute@1.0.1 
  │ │ ├── semver@5.4.1 
  │ │ ├─┬ tar@2.2.1 
  │ │ │ ├── block-stream@0.0.9 
  │ │ │ ├─┬ fstream@1.0.11 
  │ │ │ │ └── graceful-fs@4.1.11 
  │ │ │ └── inherits@2.0.3 
  │ │ └─┬ tar-pack@3.4.0 
  │ │   ├── fstream-ignore@1.0.5 
  │ │   ├─┬ once@1.4.0 
  │ │   │ └── wrappy@1.0.2 
  │ │   ├─┬ readable-stream@2.3.3 
  │ │   │ ├── core-util-is@1.0.2 
  │ │   │ ├── isarray@1.0.0 
  │ │   │ ├── process-nextick-args@1.0.7 
  │ │   │ ├── string_decoder@1.0.3 
  │ │   │ └── util-deprecate@1.0.2 
  │ │   └── uid-number@0.0.6 
  │ ├── promirepl@1.0.1 
  │ └── safe-buffer@5.1.1 
  ├─┬ stampit@2.1.2 
  │ ├── lodash@3.10.1 
  │ └── supermixer@1.0.3 
  ├── stately.js@1.3.0 
  └── vm2@3.5.0 

the same results

radar50, 2017-09-27:
Debug shows nothing
~/.node-red# DEBUG=node_red_contrib_modbus node-red -v
27 Sep 20:55:08 - [info] 

Welcome to Node-RED
===================

27 Sep 20:55:08 - [info] Node-RED version: v0.17.5
27 Sep 20:55:08 - [info] Node.js  version: v6.11.3
27 Sep 20:55:08 - [info] Linux 4.11.1-sun50iw2 arm64 LE
27 Sep 20:55:10 - [info] Loading palette nodes
27 Sep 20:55:17 - [warn] ------------------------------------------------------
27 Sep 20:55:17 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
27 Sep 20:55:17 - [warn] ------------------------------------------------------
27 Sep 20:55:17 - [info] Settings file  : /root/.node-red/settings.js
27 Sep 20:55:17 - [info] User directory : /root/.node-red
27 Sep 20:55:17 - [info] Flows file     : /root/.node-red/flows_orangepipc2.json
27 Sep 20:55:17 - [info] Creating new flow file
27 Sep 20:55:17 - [info] Starting flows
27 Sep 20:55:17 - [info] Started flows
27 Sep 20:55:17 - [info] Server now running at http://127.0.0.1:1880/

radar50, 2017-09-27:
here is output of 
DEBUG=* node-red -v &> debug.txt
file:///tmp/debug.txt

radar50, 2017-09-27:
[debug.txt](https://github.com/biancode/node-red-contrib-modbus/files/1338915/debug.txt)

biancode, 2017-10-09:
try the new version 2.0.x

closed 2017-10-09 by biancode