BiancoRoyal/node-red-contrib-modbus #58: unhandled exception

oict18, 2017-10-09:
so I am new to this, so please excuse me if this is supposed to go somewhere else.  getting started with node-red and modbus library, connected to Ctc 5300 Server on TCP and first test was a simple read of 1 holding register.  But I goofed and left the IP pointer to 0.0.0.0 in node red.  didn't like that.  kicked out of web programming environment.

looking at node-red server and log, it was attempting to restart and pitching up the same error repeatedly.  error ---> changed to failed state --->exit code --->scheduled restart.   and in this condition, the connection request from chrome was always denied.

not sure if this was the best solution, but what worked was closing the node-red server then
cd /home/pi/.node-red
ls
rm flows_raspberrypi.json
then restart webserver.  looking at the log, it re-created the missing flows_raspberrypi.json file.  it then accepted the chrome login.

biancode, 2017-10-09:
0.0.0.0 is for the Modbus nodes 127.0.0.1 or localhost or pcname.local or what did you try exactly?

closed 2017-10-15 by biancode

oict18, 2017-10-15:
Hello Mr Landsdorf,

Sorry for the delayed response.

I am trying to do is determine if the modbus library is a good means for a
particular project.  The project requires 8 raspberry pi's in a sort of
masterless network.  each pi is on modbus tcp.  On a local event, each pi
needs to write out 1 register to the other 7 pi's.  it must also read a
register from each of the other 7 pi's on a timed interrupt which is fairly
slow, likely 1 time a second or so.

It is possible that 1 or more of the pi boards will be off line (powered
down).  I will have to figure that out.

this is for a commercial project with pretty high volume and no budget for
a PLC.

Think this is a good fit?  I will obviously continue
experimenting/learning, but am interested in your feedback.

By the way- great idea and very nice work on this library.

Lee Clore
Onyx Industries
651.260.7115

biancode, 2017-10-15:
These are not enough information to give feedback. check node-red-contrib-openplc and every PI could work like a PLC ...

oict18, 2017-10-15:
will check it out- thanks

Lee Clore
Onyx Industries
651.260.7115