BiancoRoyal/node-red-contrib-modbus #59: Doesn't play nice with HTTP input/response nodes

TheOriginalMrWolf, 2017-10-10:
Hi,

First, thanks so much for this fabulous library!  Just getting started with node-red and modbus, and have found it very useful!

Unfortunately I've now run into a problem.

I have a serial modbus power meter.  I can read this perfectly well every time.  I now want to expose the power reading on an http endpoint, so I tried using the http input/response nodes (as per below).

Unfortunately, every time I call the endpoint I get the error "no response object" in debug.

Looking further, I see that modbus-getter seems to destroy the msg.req & msg.res objects, which appear to be mandatory for http response to work.

Could you please have a look at this?

Many thanks!!

[{"id":"fb4debae.d8df08","type":"debug","z":"16e78b38.7c1145","name":"","active":true,"console":"false","complete":"true","x":939.0000114440918,"y":379.00000381469727,"wires":[]},{"id":"c8f38f1d.4c53e","type":"function","z":"16e78b38.7c1145","name":"Parse modbus response","func":"var total_energy = ((msg.payload[0] << 16) + msg.payload[1]) / 100;\n\nmsg.tot_kwh = total_energy;\nmsg.volts = msg.payload[12] / 10;\nmsg.amps = msg.payload[13] / 100;\nmsg.kw = msg.payload[14] / 1000;\nmsg.var = msg.payload[15] / 1000; \nmsg.pf = msg.payload[16] / 1000;\nmsg.hz = msg.payload[17] / 100;\n\nmsg.payload = \"Total Energy:\" + total_energy;\n\nreturn msg;","outputs":1,"noerr":0,"x":713.0000076293945,"y":258.99999713897705,"wires":[["fb4debae.d8df08","7303ca56.0adb84"]]},{"id":"922619f4.c40838","type":"modbus-getter","z":"16e78b38.7c1145","name":"Read Modbus Power Meter 1","showStatusActivities":true,"showErrors":true,"unitid":"1","dataType":"HoldingRegister","adr":"0","quantity":"18","server":"60392c55.d885b4","x":419,"y":265,"wires":[["c8f38f1d.4c53e","3f403555.8d974a"],[]]},{"id":"4fdd52c7.f6c0ec","type":"http in","z":"16e78b38.7c1145","name":"total_energy request","url":"/total_energy","method":"get","upload":false,"swaggerDoc":"","x":135.38890075683594,"y":265.5555896759033,"wires":[["488aadd6.09eb64","922619f4.c40838"]]},{"id":"488aadd6.09eb64","type":"debug","z":"16e78b38.7c1145","name":"","active":true,"console":"false","complete":"true","x":353.3888931274414,"y":373.1111717224121,"wires":[]},{"id":"3f403555.8d974a","type":"debug","z":"16e78b38.7c1145","name":"","active":true,"console":"false","complete":"true","x":661.3888969421387,"y":379.11111640930176,"wires":[]},{"id":"7303ca56.0adb84","type":"http response","z":"16e78b38.7c1145","name":"total_energy response","statusCode":"","headers":{},"x":984.3889007568359,"y":259.1111812591553,"wires":[]},{"id":"60392c55.d885b4","type":"modbus-client","z":"","name":"/dev/modbus","clienttype":"serial","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/modbus","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"0","unit_id":"1","commandDelay":"100","clientTimeout":"1000","reconnectTimeout":"2000"}]

biancode, 2017-10-14:
Yes, I know, but it is much work to handle it. Node-RED does not allow to copy such messages for now, but I have to copy to reach the Modbus layer and with that, I've to find a way.

TheOriginalMrWolf, 2017-10-14:
Sorry, I must be missing something (easy to do, as I have just started teaching myself js & I've only been using Node Red for just over a week) but...

I had a quick look at modbus-getter.js for example.  I can see the input listener function just deletes the incoming message (so you lose topic, res, req, and any/all other objects attached to the message) and replaces it with a new one which you create:

```
if (msg.payload) {
        msg = {
          topic: node.id,
          payload: {
            unitid: node.unitid,
            fc: node.functionCodeModbus(node.dataType),
            address: node.adr,
            quantity: node.quantity
          }
        }

        ...etc
```

Surely, with one simple change, you can instead just replace the payload itself, and then problem goes away?

```
if (msg.payload) {
        msg.payload = {
            unitid: node.unitid,
            fc: node.functionCodeModbus(node.dataType),
            address: node.adr,
            quantity: node.quantity
          }
        ... etc
```

(I checked, and you don't seem to use topic anywhere else either, so I guess safe to stop that from corrupting incoming topic as well)

Sorry if I've misunderstood something fundamental...

biancode, 2017-10-15:
give it a try - you have the code ...

http://localhost:1880/modbus?value=0&unitid=1&fc=3&address=0&quantity=1

[image]

biancode, 2017-10-15:
that error occurs if msg.res and/or msg.req are in msg object there is an [issue](https://github.com/node-red/node-red/issues/1367) for node-red

biancode, 2017-10-15:
if it would be so easy, then I had done it before ;)

biancode, 2017-10-15:
with a _msg object it is the same:

[image]

biancode, 2017-10-15:
the communication is async and with that, we need to know the msg input to give the right one on at the async event

biancode, 2017-10-15:
works from coming soon v2.1.1 and above - please test that release, but first I've to test it before release ...

[image]

biancode, 2017-10-15:
I'm using the _msgid and a Map to identify them and to bring origin and response together. That works without copies and I got a response via HTTP.

biancode, 2017-10-15:
[image]

biancode, 2017-10-15:
install the new v2.1.0 and test it with the [HTTP-Test-Flow](https://github.com/biancode/node-red-contrib-modbus/blob/master/examples/httpModbusFlow.json)

closed 2017-10-15 by biancode

TheOriginalMrWolf, 2017-10-16:
Thanks, will do - really appreciate the fast & comprehensive response!

One question - when you say that "the communication is async", do you mean that the modbus-getter response is generated asynchronously (ie you don't just query modbus when the msg arrives at modbus-getter, and respond to that)?

Cheers!

biancode, 2017-10-16:
The order of messages is sync with Modbus, but inside the Modbus Getter node, msg is not in sync.

Example:
Msg comes into Getter node goes to Modbus client queue for reading, and next msg input appears, at that point, msg has changed in Getter node when the response comes back after reading from Modbus client. The new msg goes to Modbus client queue and so on ... With that, I had to use an msg queue inside Modbus Getter node by _msgid to resync msg with its response.

TheOriginalMrWolf, 2017-10-16:
Oh!

Obviously I have more to learn about NR because - according to my basic understanding - the race condition you have described shouldn't be possible.  Would be really interesting to put a counter on the map to see the max number of elements it stores!

Thanks for fixing the issue and taking the time to explain to a n00b like me! :)

TheOriginalMrWolf, 2017-10-17:
...and new version works well, thanks!