BiancoRoyal/node-red-contrib-modbus #60: Reading modbus data which is in 32 bit sint format

Kartheek1512, 2017-10-19:
Hi klaus, 

  I have been struggling since past few days to read data from a Power meter. Data sheet link below. Its registers are in 32 bit signed. 
http://www.elkor.net/pdfs/WattsOn-Mark_II_Manual.pdf

 When the actual value that i have to see is -273 in registers 257 ad 278. But instead i see values 62717 and 65535. can you please help me. Now a days many industrial power meters are saving values in 32 bits. 
[image]

biancode, 2017-10-19:
see [Wiki](https://github.com/biancode/node-red-contrib-modbus/wiki/Buffer-8Bit,-16Bit,-32Bit,-64Bit,-128Bit-...-reading-with-SDM630-power-meter)

biancode, 2017-10-19:
feel free to [support](https://bianco-royal.cloud/supporter/) us for our help

closed 2017-10-19 by biancode

Kartheek1512, 2017-10-21:
Thanks Klaus.. The link helped me.

I have another device which does DNP3 communication. Do you have an idea
how to get data from those devices. I believe there is no node for DNP3 in
node-red.