BiancoRoyal/node-red-contrib-modbus #61: Modbus serial use high CPU after connect to serial port

amnart22, 2017-10-19:
Hi klaus,

I use Modbus-Read and Flex-Getter in serial mode then I notice after connected to serial port, Node.js use CPU 20% without data transfer ( before connect was 0%) but everything work well.
I tested with [node-red-node-serialport](https://www.npmjs.com/package/node-red-node-serialport), it is use 0% of CPU after connect.
I use Intel i7-3630QM + windows10 and tested with driver of CH340 and CP210x, it's same high CPU usage.

And after I deleted Modbus-Read node from flow, it is still use high CPU until I changed to TCP mode in config tab, CPU usage will be 0%.

[image]

Could you suggest me, how to solve this problem?

Thanks,
Amnart Kassanook

biancode, 2017-10-31:
Dear Amnart, 

the Modbus-Read node has a cron-service copy of the inject node inside. That will use some CPU.
The next is the polling timer of the node.js interval to poll the Modbus device. Please, try the inject of Node-RED.

Happy testing!
Klaus

biancode, 2017-10-31:
more information about [event loops and timers](https://nodejs.org/en/docs/guides/event-loop-timers-and-nexttick/)

biancode, 2017-10-31:
I am very interested if you find more or if you have any idea to make that using less CPU ...

amnart22, 2017-10-31:
Hi Claus

I checked it and found :
- High CPU required when I use 'node-red-contrib-modbus version 2.0 and newer' with Windows 10 64 bit (not test with 32 bit yet)
- It's require 20% of overall CPU after connect 'modbus read' = 100% in 1 core of CPU.

So I found 2 solutions to use less CPU
1. 'node-red-contrib-modbus' V1.3.0 work on Windows10 64 bit require 0% - 1% of CPU ( Core i7-3630QM )
2. 'node-red-contrib-modbus' all version work on Ubuntu 16.04 64 bit required 0% -1% of CPU ( Celeron 1037u CPU )

I think may be this issue depend on version of serial port that changed from 4.x to 5.x.
And now it's worked very well for me with those 2 solutions 

Regards,
Amnart

biancode, 2017-10-31:
Hi Amnart,

then it depends more on serialport 6.x, which is used in our modbus-serial library. The package serialport 5.x is just for the Modbus server node, which is in use by the node-modbus library.
Did you try `npm update` or to reinstall my package? With that, you should get the 6.0.4 of serialport.

Regards,
Klaus

amnart22, 2017-10-31:
Thanks Klaus,
I will check it and let you know.

Amnart

amnart22, 2017-11-01:
Hi Klaus

Upgrading serialport in modbus-serial to 6.x.x can solve this problem, it's reduce using CPU to 1%.
Quite sure this problem occur when using serialport 5.x.x on Windows (not on Ubuntu).

But when I install node-red-contrib-modbus 2.1.1, I will get modbus-serial 6.0.1 with serialport 5.x.x.
Need manual updrade serialport to 6.x.x to solve the problem. And I found modbus-serial 7.0.0 on Github but latest on npm is v6.0.1, maybe they will upgrade on npm soon.

Many thanks for your kind help

Regard,
Amnart

closed 2017-11-01 by amnart22

biancode, 2017-11-01:
@yaacov please see the last comment here

biancode, 2017-11-01:
[library issue](https://github.com/yaacov/node-modbus-serial/issues/159)

amnart22, 2017-11-02:
Hi Klaus

After @yaacov upgrade modbus-serial 7.0.1, it's work, CPU load 0% and No need manual install anymore.

Thanks to you and Yaacov very much for kind help
Amnart

biancode, 2017-11-02:
Thanks for that issue.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>