BiancoRoyal/node-red-contrib-modbus #62: Modbus connection between my PLc and node-red

aekanshgupta, 2017-10-27:
So, I am trying to establish a connection between my Schneider PLC and my raspberry pi using node-red but my modbus node on node red shows the error "FSM reset on state empty".I have established the connection between them for 2-3 second it shows connected, then timeout and the error mentioned above pops up. Please help me I have been stuck here for a week.

biancode, 2017-10-30:
Please, follow the README and send me your debug.log file with all Debug information.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

aekanshgupta, 2017-10-31:
Sir, I can't locate your README file.

biancode, 2017-10-31:
oh, here - see [Debug](https://github.com/biancode/node-red-contrib-modbus#debug)

biancode, 2017-10-31:
The messages means that your Modbus device has a failure state while the queue is empty (no messages for the devices) - also try to switch on the options inside the Modbus node - Show Errors

Please, configure a catch node with debug output in your flow!

`[{"id":"d48e8407.d12038","type":"catch","z":"dfac3efe.3a37d","name":"","scope":null,"x":650,"y":80,"wires":[["68f817a3.2ea308"]]},{"id":"68f817a3.2ea308","type":"debug","z":"dfac3efe.3a37d","name":"","active":true,"console":"false","complete":"true","x":800,"y":80,"wires":[]}]`

aekanshgupta, 2017-10-31:
Sir, I did as you asked me to but there is no change. What I am trying to do is, I have connected a push button( digital input) to my PLC and downloaded a LD diagram in it, where the digital input is connected to digital output and I am trying to read that digital Output via modbus. So in the modbus read node I wrote the address of the digital output where I am supposed to enter the address. I am expecting that I would get output as **TRUE** as soon as I press the button and **FALSE** when not pressed.
Kindly refer to the flow-:

[{"id":"13d9b374.efa9bd","type":"tab","label":"Flow 4"},{"id":"cd30ff66.452e1","type":"debug","z":"13d9b374.efa9bd","name":"","active":true,"console":"false","complete":"true","x":755,"y":305,"wires":[]},{"id":"5c18e6a5.776b68","type":"modbus-read","z":"13d9b374.efa9bd","name":"Digital O/p","showStatusActivities":false,"showErrors":true,"unitid":"","dataType":"HoldingRegister","adr":"8528","quantity":"1","rate":"3","rateUnit":"s","server":"4d0181ba.9128d","x":448,"y":249,"wires":[["cd30ff66.452e1"],[]]},{"id":"35d688cc.4f8e98","type":"catch","z":"13d9b374.efa9bd","name":"","scope":null,"x":916,"y":98,"wires":[["341b8040.fff4b"]]},{"id":"341b8040.fff4b","type":"debug","z":"13d9b374.efa9bd","name":"","active":true,"console":"false","complete":"true","x":1066,"y":98,"wires":[]},{"id":"4d0181ba.9128d","type":"modbus-client","z":"","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"192.168.137.102","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

aekanshgupta, 2017-10-31:
The debug node connected to the catch node just responded saying **" Error: Modbus exception 10"**.

biancode, 2017-10-31:
please, then check [exception hints](http://www.simplymodbus.ca/exceptions.htm?nocache=1509469348)

biancode, 2017-10-31:
> Specialized use in conjunction with gateways, indicates that the gateway was unable to allocate an internal communication path from the input port to the output port for processing the request. Usually means the gateway is misconfigured or overloaded.

aekanshgupta, 2017-11-02:
So, can you tell me what am i doing wrong. Am I providing the wrong address?

biancode, 2017-11-02:
I am sorry, this is out of package support. Please follow [become a supporter](https://bianco-royal.cloud/supporter/) and then we can give more support. 

Have a nice day! 
Klaus

closed 2017-11-02 by biancode