BiancoRoyal/node-red-contrib-modbus #63: Modbus Flex Getter doesn't wait for response in BUFFERED-RTU mode

candronikos, 2017-10-30:
My colleague tested this for use and found that if querying multiple devices it doesn't wait for the response before sending the next message.

Is this a known issue? Or is there a configuration setting I'm missing?

biancode, 2017-10-31:
Hi,

are you working with the queue? Or is it disabled? Please, send me the flow or configuration of your Modbus Client node and some log about that! (see [Debug](https://github.com/biancode/node-red-contrib-modbus#debug))

That behaviour should just work without the queue if your gateway manage a message queue.

biancode, 2017-10-31:
It sends very fast (see settings queue delay - default 1 msec.) to different devices if there are messages in the queue for multiple devices, but it waits every time for the response of each device, and it knows its callback to address the right sender to route the message.

biancode, 2017-10-31:
watch [queue reset](https://youtu.be/ND9ZYkMOwZs).  and [queue or not](https://youtu.be/Eg8qCAF8suo)

candronikos, 2017-10-31:
Thanks for the references. I am going through them now. My flow is at the bottom of this post.

The desired outcome is to iterate over all devices and return the values in the registers. Also have to retry a read if I don't see a response which is why I have a delay and a function block loop back to the flex getter. What's the best way to do this in your mind?

```
[
    {
        "id": "f28d534f.23c85",
        "type": "function",
        "z": "c8ca9dbd.c1c94",
        "name": "Read Loop",
        "func": "/*\npayload = {\n    //value: msg.payload,\n    'fc': 3,\n    'unitid': 1,\n    'address': 0,\n    'quantity':10, \n};\n\nreturn {\n    payload: payload\n};\n*/\n\ndevices = context.flow.get('config').devices;\n\nif(devices === undefined) {\n    node.error(\"No devices in config\");\n}\n\nfor(var i in devices) {\n    d = devices[i];\n    d.unitid = parseInt(i);\n    d.fc = 3;\n\n    node.send({\n        payload: d\n    });\n}\n\nreturn null;",
        "outputs": 1,
        "noerr": 0,
        "x": 290,
        "y": 520,
        "wires": [
            [
                "a3adc60.6c36c38",
                "7ddc4b4.4be55b4",
                "e641741f.6ea798"
            ]
        ]
    },
    {
        "id": "a3adc60.6c36c38",
        "type": "modbus-flex-getter",
        "z": "c8ca9dbd.c1c94",
        "name": "Flex Getter",
        "showStatusActivities": true,
        "showErrors": true,
        "server": "60b70824.f661f8",
        "x": 550,
        "y": 520,
        "wires": [
            [
                "1006e4c6.feff2b"
            ],
            [
                "5dac35bf.78dc7c",
                "1b3a4f.720565b2"
            ]
        ]
    },
    {
        "id": "7ddc4b4.4be55b4",
        "type": "debug",
        "z": "c8ca9dbd.c1c94",
        "name": "",
        "active": false,
        "console": "false",
        "complete": "true",
        "x": 690,
        "y": 600,
        "wires": []
    },
    {
        "id": "e641741f.6ea798",
        "type": "delay",
        "z": "c8ca9dbd.c1c94",
        "name": "",
        "pauseType": "delay",
        "timeout": "10",
        "timeoutUnits": "seconds",
        "rate": "1",
        "nbRateUnits": "1",
        "rateUnits": "second",
        "randomFirst": "1",
        "randomLast": "5",
        "randomUnits": "seconds",
[... 117 lines cut ...]
    {
        "id": "60b70824.f661f8",
        "type": "modbus-client",
        "z": "",
        "name": "Test",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyS0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "500",
        "unit_id": 1,
        "commandDelay": 750,
        "clientTimeout": 2000,
        "reconnectTimeout": 2000
    }
]
```

biancode, 2017-11-02:
`[{"id":"cf0fceb3.b99a","type":"function","z":"a959be51.7ab6f","name":"Read One","func":"payload = {\n    //value: msg.payload,\n    'fc': 3,\n    //'unitid': 1,\n    'address': 0,\n    'quantity':13, \n};\n\nreturn {\n    payload: payload\n};\n","outputs":1,"noerr":0,"x":230,"y":320,"wires":[["db226302.db8dd","5223b345.d1731c","cfefc252.202be"]]},{"id":"2a1211ab.1e349e","type":"inject","z":"a959be51.7ab6f","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":120,"y":270,"wires":[["cf0fceb3.b99a"]]},{"id":"db226302.db8dd","type":"modbus-flex-getter","z":"a959be51.7ab6f","name":"Flex Getter","showStatusActivities":true,"showErrors":true,"server":"ee858c42.1aa82","x":560,"y":230,"wires":[["6ca5ed71.86c094"],["24340c9.f0110f4","ae0bfbfb.055868"]]},{"id":"5223b345.d1731c","type":"debug","z":"a959be51.7ab6f","name":"","active":false,"console":"false","complete":"true","x":530,"y":310,"wires":[]},{"id":"cfefc252.202be","type":"delay","z":"a959be51.7ab6f","name":"","pauseType":"delay","timeout":"10","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":410,"y":130,"wires":[["6ca5ed71.86c094"]]},{"id":"6ca5ed71.86c094","type":"function","z":"a959be51.7ab6f","name":"Retry","func":"ctx = context;\n\nvar msg_ids = ctx.get('msg_ids') || {};\n\nif('retry' in msg)\n    return null;\n\nif('responseBuffer' in msg) {\n    // Then this is a response object from the modbus slave\n\n    msg_ids[msg._msgid] = msg;\n    ctx.set('msg_ids', msg_ids);\n} else {\n    // Then this is an input object to a modbus slave\n\n    if(msg._msgid in msg_ids) {\n        // Then this message was handled. Nothing to do here\n    } else {\n        // Then this message didn't prompt a response from the slave.\n        // Need to resend\n        delete msg_ids[msg._msgid];\n        ctx.set('msg_ids', msg_ids);\n        msg.retry = true;\n        //node.warn(\"Message _msgid: '\" + msg._msgid + \"' not found. Resending...\");\n        node.send(msg);\n    }\n    \n}\n\n//return [ [ null ], [ msg_ids ] ];","outputs":"2","noerr":0,"x":570,"y":130,"wires":[["fe3815fd.b34728","db226302.db8dd"],[]]},{"id":"24340c9.f0110f4","type":"debug","z":"a959be51.7ab6f","name":"","active":false,"console":"false","complete":"true","x":770,"y":250,"wires":[]},{"id":"ae6b63c7.3d56d","type":"function","z":"a959be51.7ab6f","name":"Read Loop","func":"/*\npayload = {\n    //value: msg.payload,\n    'fc': 3,\n    'unitid': 1,\n    'address': 0,\n    'quantity':10, \n};\n\nreturn {\n    payload: payload\n};\n*/\n\n/* count the time for all devices and check queue */\ndevices = context.flow.get('config').devices;\n\nif(devices === undefined) {\n    node.error(\"No devices in config\");\n}\n\nfor(var i in devices) {\n    d = devices[i];\n    d.unitid = parseInt(i);\n    d.fc = 3;\n\n    node.send({\n        payload: d\n    });\n}\n\nreturn null;","outputs":1,"noerr":0,"x":220,"y":220,"wires":[["db226302.db8dd","5223b345.d1731c","cfefc252.202be"]]},{"id":"ae0bfbfb.055868","type":"modbus-response","z":"a959be51.7ab6f","name":"","registerShowMax":20,"x":810,"y":200,"wires":[]},{"id":"fe3815fd.b34728","type":"debug","z":"a959be51.7ab6f","name":"","active":false,"console":"false","complete":"true","x":720,"y":130,"wires":[]},{"id":"beaa2f24.8f739","type":"inject","z":"a959be51.7ab6f","name":"","topic":"","payload":"","payloadType":"date","repeat":"10","crontab":"","once":false,"x":150,"y":170,"wires":[["ae6b63c7.3d56d"]]},{"id":"d41b0843.41b3f8","type":"modbus-queue-info","z":"a959be51.7ab6f","name":"","unitid":"","lowLowLevel":25,"lowLevel":75,"highLevel":150,"highHighLevel":300,"server":"ee858c42.1aa82","errorOnHighLevel":false,"x":540,"y":50,"wires":[["c94c82d0.d2025"]]},{"id":"c94c82d0.d2025","type":"debug","z":"a959be51.7ab6f","name":"","active":false,"console":"false","complete":"true","x":740,"y":50,"wires":[]},{"id":"ee858c42.1aa82","type":"modbus-client","z":"","name":"Test","clienttype":"serial","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyS0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"500","unit_id":1,"commandDelay":750,"clientTimeout":2000,"reconnectTimeout":2000}]`

biancode, 2017-11-02:
the queue delay is for every message (750msec. in your flow) - set the minimum of time between every message you want

biancode, 2017-11-02:
and please, try the v2.2.0 update

candronikos, 2017-11-02:
Thanks, I'll give that a try

biancode, 2017-11-06:
Please, reopen or open a new issue if you find a way to reproduce that behavior.

closed 2017-11-06 by biancode

candronikos, 2017-11-07:
Tested again with my colleague and could not reproduce the issue