BiancoRoyal/node-red-contrib-modbus #64: Catching timeouts

mascapollo, 2017-11-01:
Hello There, 

It is possible to catch read or write timeouts from the flex nodes?

biancode, 2017-11-02:
Hi,

please, switch on the option "Show Errors" and "Show Activities" and insert a catch node with debug msg.

Happy testing!
Klaus

biancode, 2017-11-02:
`[{"id":"89631453.62a938","type":"catch","z":"f5ce9336.3dec9","name":"","scope":null,"x":530,"y":70,"wires":[["a80c480b.97e4b8"]]},{"id":"a80c480b.97e4b8","type":"debug","z":"f5ce9336.3dec9","name":"","active":true,"console":"false","complete":"true","x":680,"y":70,"wires":[]}]`

closed 2017-11-02 by biancode

mascapollo, 2017-11-03:
Hello Klaus.

I have this setup but the catch node does no send any timeout message. 

[image]