BiancoRoyal/node-red-contrib-modbus #65: Suddenly all reads have CRC errors

TheOriginalMrWolf, 2017-11-02:
Hi,

Running on a Raspberry/DietPi installation.

Have been successfully reading a power meter for several weeks.

Disabled the modbus flow for about a week and now, once I re-enabled it, every read fails with a CRC error (I can see the interface send/receive lights blinking, so I know it's getting some sort of response from the slave).

Nothing on the modbus side has changed - same meter / wiring / flow, however did a Pi update/upgrade a few days ago.

Any suggestions on where to start debugging this?

Thanks!!

biancode, 2017-11-02:
Hi,

you can send me some [debug log](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG) including modbus-serial flaq, please.

Happy logging!
Klaus

biancode, 2017-11-02:
and after log with the actual version, try to update or reinstall, please to v2.2.0

biancode, 2017-11-04:
Hi, did you found the problem?

biancode, 2017-11-06:
reopen if the problem is not fixed ... for you

closed 2017-11-06 by biancode

TheOriginalMrWolf, 2017-11-07:
Hi,

Sorry I didn't respond earlier - troubleshooting took longer than expected.

Issue in the end was that the power meter went faulty. :(

Replaced the unit and all came good.

Thanks again!