BiancoRoyal/node-red-contrib-modbus #66: Setting up RaspBerry Pi 1 as Slave/Server

Roklo, 2017-11-02:
Hi,

I have been trying for the last month to get my RaspBerry Pi 1 with Node Red to work as a slave/server in my Modbus TCP network with no success. My current network consists of an Wago PLC 8100-750 (Slave/Server), a robotic arm (Master) and the Raspberry Pi (Slave/Master). The idea is when the robot arm is done performing a given task, it will send a signal to the PLC, then the PLC will perform some calculations and send a signal to the Raspberry telling it to play a specific sound.

The communication between the robot and the PLC works. The Raspberry is set up like the contrib-modbus Wiki says. 

The ideal setup would be to use Coils and Boolean values (as the robot also uses it). 
I do believe it is a simple mistake, probably I have misunderstood how some of the nodes work. Any help setting this up would be much appreciated.

biancode, 2017-11-02:
Dear Rollo,

This is out of package support. Please follow [become a supporter](https://bianco-royal.cloud/supporter/) and then we can give more support. Have a nice day! klaus

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2017-11-02:
You can contact me via E-Mail and please attach the flow.

closed 2017-11-02 by biancode